moveit/moveit2 #16: Continuous integration

vmayoral, 2019-02-19:
@LanderU, I see you fixed CI in https://github.com/AcutronicRobotics/moveit2/tree/master.

I started a branch at https://github.com/AcutronicRobotics/moveit2/tree/ci for making a PR once it was working (now is!). Would you like me to go ahead and cherry-pick all your changes and then submit the PR myself or would you rather like to do it yourself?

davetcoleman, 2019-02-21:
Much needed testing!

vmayoral, 2019-02-24:
We will also take care of this ticket, work in progress for now.

vmayoral, 2019-03-05:
@LanderU, after a few days of testing, I'd say our CI prototype is ready for a PR. We need to submit only the CI-related changes. Would you like to submit the PR yourself or should I go ahead an do it?

LanderU, 2019-03-05:
> Would you like to submit the PR yourself or should I go ahead an do it?

Feel free to go ahead.

davetcoleman, 2019-03-08:
@mlautman is going to help convert https://github.com/ros-planning/moveit_ci/ to be ros2 compatible. its been a many years effort to build a very comprehensive set of tests for the code base, between myself and @rhaschke

vmayoral, 2019-03-09:
That is great to hear, thanks @davetcoleman. We will monitor the progress and contribute by ensuring we pick this in our hardware tests. To do so more easily, https://github.com/ros-planning/moveit_resources/pull/24 should probably be considered. It would also help very much getting a unified infrastructure altogether with the hardware CI we are planning.

@ahcorde and @LanderU this will be interesting for you.

davetcoleman, 2019-12-03:
@henningkayser can this issue be closed?

closed 2019-12-07 by henningkayser