moveit/moveit2 #21: Convert to ROS2 Logging

davetcoleman, 2019-02-22:
My understanding of the situation: logging in ROS2 is different than ROS1 due to the lack of a singleton (the assumption that there is one ROS node per process). In ROS2 each call to the logger requires you pass in a logger handle, or the node handle. 

In order to achieve the MVP of moveit2, I recommend we create the ROS2 node as a global variable, or at least the logger handle as a global variable. This will limit moveit2 from having more than one instance in the same process, but we can fix that in the future.

So, in MoveGroup ``main`` function we'll need to create this singleton. 

This issue is to track the modification of logging in moveit2. @wjwwood is the lead contact for us who is expert on this issue.

wjwwood, 2019-02-23:
That sounds right to me. I imagine something like `moveit_core::get_logger() -> rclcpp::Logger` which can be called from anywhere to get a logger to be used locally and a way to set that from something ros specific in main function or core constructor, e.g. `moveit_ros::set_logger(my_node->get_logger())` which internally might call `moveit_core::set_logger()`.

vmayoral, 2019-02-24:
Thanks for the input @wjwwood.

@davetcoleman as pointed out at https://github.com/ros-planning/moveit2/pull/7#issuecomment-466764787, I think we should address this issue in a separate PR since it'll touch submodules of several of the packages. I presume that changes will need to be made in `moveit_ros` package, right? 

It'll be really helpful to get the logging somewhat addressed before we port the rest of the code. @davetcoleman since you know best the code, can you make an initial prototype extending MoveGroup `main` function while following the guidelines of @wjwwood? That'll speed things up significantly on our side.

davetcoleman, 2019-02-24:
I don't think I'll be able to get to this in the next week due to travel, and I don't think I'm the most qualified to prototype this for ROS2.

vmayoral, 2019-02-26:
@davetcoleman, we finished a prototype in the kinematic_contraints submodule available at https://github.com/AcutronicRobotics/moveit2/blob/master/moveit_core/kinematic_constraints/src/kinematic_constraint.cpp#L52 and used across the code as https://github.com/AcutronicRobotics/moveit2/blob/master/moveit_core/kinematic_constraints/src/kinematic_constraint.cpp#L131.

We are systematically porting packages while verifying that things compile (as opposed to porting changes across all moveit2 packages, we don't know the code well enough to be doing so confidently and such ). 

Given the discussion above where you'd like to have a global variable (I guess in MoveGroup from what's been discussed in other threads), this approach won't the final however it'll be easily changed in the future and makes **now** use of RCLPP APIs which will hopefully unblock several PRs which are pending such as https://github.com/ros-planning/moveit2/pull/7 or https://github.com/ros-planning/moveit2/pull/9.

Let me know if that's acceptable and we'll get it done.

davetcoleman, 2019-02-27:
Thanks for putting together this prototype! I'd like some level of buy in from @wjwwood but it seems good to me.

To keep our [style guideline](https://moveit.ros.org/documentation/contributing/code/) roughly the same in moveit2 (see the ROS section of that page), I'd like to keep the namespace variable the same name of ``LOGNAME``. Its also more space efficient. 

In this new approach you're proposing, I assume you'll also delete your [logging.h](https://github.com/ros-planning/moveit2/blob/master/moveit_core/logging/include/moveit/logging/logging.h#L41) ? I was actually a fan of that approach as it would minimize the diff, for now, between ROS1 & 2. Though that isn't super important to me.

Rather than links to lines of code, it's easiest to discuss the proposed change in a PR that targets logging, could you all do that please?

wjwwood, 2019-02-28:
Looks fine to me, only thing I'd say is that the global logger may also be `static` and `const`.

mlautman, 2019-03-04:
(copied form #12 )

According to the ROS/Google style guides it seems less than ideal to have the logger initialized here. (see: http://wiki.ros.org/CppStyleGuide#Globals, 
https://google.github.io/styleguide/cppguide.html#Static_and_Global_Variables)

It would be better have a string as a global variable much like [robot_state.cpp](https://github.com/ros-planning/moveit/blob/211158d00a6938c819f5d51301662b9d993b0c19/moveit_core/robot_state/src/robot_state.cpp#L58) where we declare the LOGNAME string at the top of the source. (`const std::string LOGNAME = "transforms";`) 

When possible, we should use class variables for the loggers (eg: `rclcpp::Logger logger_transforms_`) which get set on initialization and used throughout the class.

When it is impossible to use a logger class variable, `RCLCPP_ERROR` can still create it's own logger to avoid having non-trivial global variables floating around. 

```cpp
RCLCPP_ERROR(rclcpp::get_logger(LOGNAME), "...");
```

vmayoral, 2019-03-05:
I finished changing all the submodules of moveit_core in our [master branch](https://github.com/AcutronicRobotics/moveit2/tree/master) to the logging primitives recommended by @wjwwood. Changes follow from the prototype presented before at https://github.com/ros-planning/moveit2/issues/21#issuecomment-467385110.

This took quite a bit of time and effort.

> To keep our style guideline roughly the same in moveit2 (see the ROS section of that page), I'd like to keep the namespace variable the same name of LOGNAME. Its also more space efficient.

I went through the style guide and will keep it in mind. Do you intent do have a const variable only used once to define the logger? Even if that complies with the current style, I'd argue that can probably change given the fact that the whole logging structure (and how it's used) has also changed. Maybe I'm missing something and I don't quite see how you'd like to have this so I leave it up to you.

I can see @mlautman has further ideas and change suggestions however I don't think we'll be able to put more time into this in the short term since we want to move forward. I let @mlautman, @davetcoleman and others submit further improvements if you consider them necessary. Upcoming submodules ported will include the changes discussed above.

wjwwood, 2019-03-05:
From the style guide:

> As a rule of thumb: a global variable satisfies these requirements if its declaration, considered in isolation, could be constexpr.

This implies to me that `std::string` and therefore `rclcpp::Logger` should be ok to have static storage. It would be possible to make it `constexpr` if you were allowed to allocate memory for `constexpr` expressions (something that is being worked on, see: https://github.com/ldionne/wg21/blob/master/generated/p1004r1.pdf / https://www.youtube.com/watch?v=CRDNPwXDVp0&feature=youtu.be&t=3080).

At any rate, I think it's safe and more efficient to statically create the logger object rather than statically store a `char *` and create a logger each time you need it (each time you create a logger from a `char *` a memory allocation may need to occur). However, in practice the SBO for std::string may save you most of the performance cost.

It's up to you guys.

davetcoleman, 2019-03-06:
I'm frustrated by the lack of tangible PRs to have these discussions on - links to code is much more difficult to iterate on and discuss inline. I re-reviewed all the discussions so far on logging am proposing a simple example that I believe scales easily to the whole code base:

https://github.com/ros-planning/moveit2/pull/33

Feel free to comment inline there if you see issues with it. Or make a new PR with an alternative suggestion.

mlautman, 2019-03-08:
> I can see @mlautman has further ideas and change suggestions however I don't think we'll be able to put more time into this in the short term since we want to move forward. I let @mlautman, [...]

@vmayoral I may have been being overly critical. I think the prototypes at https://github.com/ros-planning/moveit2/issues/21#issuecomment-467385110 and #33 offer a good path forward and if the logging infrastructure changes in the future, we can reassess.

mlautman, 2019-03-20:
Copied from: https://github.com/ros-planning/moveit2/pull/43#discussion_r266140408

I believe we settled on `static const  rclcpp:Logger LOGGER` in general. 

For class specific loggers in files with multiple loggers `static const  rclcpp:Logger ROBOT_STATE_LOGGER` would work as well. 

For logger namespacing, we should keep the `"moveit.` prefix. So the logger in RobotState might have a logger:
```c++
const rclcpp:Logger LOGGER = rclcpp::get_logger("moveit.robot_state")
```
And if there is a reason to use a child logger:
```c++
// within the method
rclcpp:Logger some_method_logger = LOGGER.get_child("some_method_name")
// outside the method 
const rclcpp:Logger SOME_METHOD_LOGGER = LOGGER.get_child("some_method_name")
```

wjwwood, 2019-03-20:
> `const rclcpp:Logger LOGGER = rclcpp::get_logger("moveit.robot_state")`

I'd suggest this small alteration:

```
const rclcpp:Logger LOGGER = rclcpp::get_logger("moveit").get_child("robot_state");
```

In case we ever change the separator.

vmayoral, 2019-03-30:
Added @wjwwood's feedback https://github.com/ros-planning/moveit2/pull/59/commits/4caa9472828d0cd57a95ef9a21a0fdcf16a03737 however note that I've omitted the `const` at the beginning.

Including it causes the following  error which I haven't spent time debugging:
```
/Users/victor/ros2_moveit_ws/src/moveit2/moveit_core/robot_state/src/conversions.cpp:62:5: error: static_assert failed due to requirement '::std::is_same<typename std::remove_reference<decltype(LOGGER)>::type, typename ::rclcpp::Logger>::value' "First argument to logging macros must be an rclcpp::Logger"
    RCLCPP_ERROR(LOGGER, "Different number of names and positions in JointState message: %zu, %zu",
    ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
/Users/victor/ros2_ws/install/rclcpp/include/rclcpp/logging.hpp:411:3: note: expanded from macro 'RCLCPP_ERROR'
  static_assert( \
  ^
/Users/victor/ros2_moveit_ws/src/moveit2/moveit_core/robot_state/src/conversions.cpp:78:5: error: static_assert failed due to requirement '::std::is_same<typename std::remove_reference<decltype(LOGGER)>::type, typename ::rclcpp::Logger>::value' "First argument to logging macros must be an rclcpp::Logger"
    RCLCPP_ERROR(LOGGER, "Different number of names, values or frames in MultiDOFJointState message.");
    ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
/Users/victor/ros2_ws/install/rclcpp/include/rclcpp/logging.hpp:411:3: note: expanded from macro 'RCLCPP_ERROR'
  static_assert( \
  ^
/Users/victor/ros2_moveit_ws/src/moveit2/moveit_core/robot_state/src/conversions.cpp:100:9: error: static_assert failed due to requirement '::std::is_same<typename std::remove_reference<decltype(LOGGER)>::type, typename ::rclcpp::Logger>::value' "First argument to logging macros must be an rclcpp::Logger"
        RCLCPP_ERROR(LOGGER, "Caught %s", ex.what());
        ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
```

wjwwood, 2019-04-02:
> Including it causes the following error which I haven't spent time debugging:

I think that's a bug in our template magic that supposed to detect when you're passing something other than a logger and instead should give a better error.

It needs to be updated to remove const-ness before comparing with `std::is_same`.

I just made a new issue for it here: https://github.com/ros2/rclcpp/issues/679

It would be a good first contribution to `rclcpp` if anyone has time for it :)

vmayoral, 2019-04-02:
https://github.com/ros2/rclcpp/pull/680 fixes it. Thanks @wjwwood.

henningkayser, 2019-12-07:
Added migration instructions to https://github.com/ros-planning/moveit2/blob/master/doc/MIGRATION_GUIDE.md

closed 2019-12-07 by henningkayser