moveit/moveit2 #23: Convert to ROS2 rostest

davetcoleman, 2019-02-22:
MoveIt 1 uses rostest a fair amount. In ROS 2 the exact replacement is still [under development](https://discourse.ros.org/t/test-framework-in-ros2/2667) according to @wjwwood so there is not great documentation yet.

This issue will track the progress of us understanding how to do the conversion.

wjwwood, 2019-02-23:
This is probably a better link than that discourse post:

https://github.com/ros2/launch/issues/161 (found via the ROS 2 Dashing ticket: https://github.com/ros2/ros2/issues/607)

The above gets you the functionality of rostest, but not an XML frontend, if you're wanting to continue use that pattern like rostest from ROS 1, that's also being worked on, but tracker here:

https://github.com/ros2/launch/issues/163 and https://github.com/ros2/design/pull/207 (again via ROS 2 Dashing ticket: https://github.com/ros2/ros2/issues/607)

There was also the idea to support existing roslaunch and rostest files from ROS 1, but that's not being tracked anywhere, and I am not aware of anyone working on it or any of the technical risks for it.

shivaang12, 2020-06-01:
@davetcoleman I was looking for testing relate stuff for Moveit hackathon tomorrow. Let me know if there are any specific thing you want to get done.

davetcoleman, 2020-06-02:
@shivaang12 this is a better question for @henningkayser as he leads MoveIt 2 development

@tylerjw is there a place to get CodeCov reports outside of specific PRs? Like a badge we can put on the README? Would be good to point @shivaang12 there.

See also: [Integration/Unit Tests Tutorial](https://ros-planning.github.io/moveit_tutorials/doc/tests/tests_tutorial.html)

henningkayser, 2020-10-14:
Closing this, remaining efforts are discussed in https://github.com/ros-planning/moveit2/issues/216

closed 2020-10-14 by henningkayser