moveit/moveit2 #24: Add CircleCI Code Coverage to MoveIt2

davetcoleman, 2019-02-23:
We've been wanting this for a [long time](https://github.com/ros-planning/moveit/issues/1) and I just noticed that the navigation2 project has a [nice CircleCI setup](https://github.com/ros-planning/navigation2/blob/master/.circleci/config.yml) we could perhaps use. 

@ruffsl it looks like you were the one who spun this up for navigation, could you help us setup this up for moveit also?

ruffsl, 2019-02-23:
> @ruffsl it looks like you were the one who spun this up for navigation, could you help us setup this up for moveit also?

Sure, I'd be happy to help! I've got some deadlines this week, but I'll code up a PR after Thursday. In the meantime, could you complete these action items for moveit repo/org , like those I detailed in the original navigation PR? E.g linking the GitHub repo to circleci, a docker hub repo, also nailing down and external dependencies not shipped in the nightly image into a .repo file to streamline the build for current master. 

https://github.com/ros-planning/navigation2/pull/534

ruffsl, 2019-03-06:
Hey @davetcoleman , I'd like to start on this. I'd like to know what subset packages currently build with ROS2 Crystal that I could used to test with?

davetcoleman, 2019-03-08:
Within this moveit2 repo I don't believe any of the ROS2 packages build yet, unfortunately.

@mlautman can you help provide the info to @ruffsl

vmayoral, 2019-03-09:
> Within this moveit2 repo I don't believe any of the ROS2 packages build yet

That's not right, there're several submodules within `moveit_core` that should build. Several more will be once we approve and merge the [existing PRs](https://github.com/ros-planning/moveit2/pulls).

@ruffsl if you need a _build_-able branch for development purposes, feel free to pick `master` branch from http://github.com/acutronicrobotics/moveit2. We're maintaining a CI (which needs to be constantly updated due to the addition of new dependencies as we port new submodules) while ensuring things compile both for OS X and Linux .

davetcoleman, 2019-12-03:
@tylerjw can you add your codecov approach to this repo?

davetcoleman, 2021-04-26:
@tylerjw can this issue be closed after your recent efforts?

tylerjw, 2021-04-26:
Yes, I'm closing this because we are now in the process of migrating all of the moveit repos to use github actions.

closed 2021-04-26 by tylerjw