moveit/moveit2 #32: Keeping Isometry3d in ROS2

davetcoleman, 2019-03-01:
A change was made by @rhaschke in ros1/geometry2 that was not forward ported to ros2/geometry2, see https://github.com/ros/geometry2/pull/335.

This change urgently needs to be ported so that there isn't a big regression in the moveit2 codebase as demonstrated here and in the related conversation: https://github.com/ros-planning/moveit2/pull/12#discussion_r258148928

vmayoral, 2019-03-03:
https://github.com/ros2/geometry2/pull/93 addresses the issue in geometry2. Changes like https://github.com/ros-planning/moveit2/pull/12/commits/94ebbcc191db16d6e3fae760d2d8b8062517d53f should follow.

vmayoral, 2019-03-14:
Shall we close this @davetcoleman? @mlautman?

davetcoleman, 2019-03-15:
Let's wait until https://github.com/ros2/geometry2/pull/93 is closed. This should actually auto-close now that I've tagged its dependency on github.

anasarrak, 2019-06-03:
This issue can be closed, they've merged the changes on https://github.com/ros2/geometry2/pull/90

closed 2019-06-03 by rhaschke