moveit/moveit2 #45: How to sync moveit1 and moveit2

davetcoleman, 2019-03-15:
Moving @rhaschke 's documentation and proposal here for further debate.

@rhaschke :
----------------------------------
In contrast to the merge-strategy suggested in https://github.com/ros-planning/moveit2/pull/26#issue-255589277, I propose the following for the future:

```sh
git clone https://github.com/ros-planning/moveit2
cd moveit2
git fetch https://github.com/ros-planning/moveit master  # fetch MoveIt 1 branch
git merge FETCH_HEAD  # and merge into MoveIt 2 head
```
If approved, such a sync PR should be "merged" manually using **`fast-forward`**. github's web interface doesn't provide such an option!
Using github's merge commit, will create another, superfluous commit and squashing or rebasing doesn't make sense at all.
However, looks like @davetcoleman has forbidden to push to the MoveIt 2 repo from cmdline. This is a typical use case, where interaction from cmdline _is_ desirable :wink:

**When there are no merge conflicts**, the following procedure using github's web interface will work too:
```sh
git clone https://github.com/ros-planning/moveit2
cd moveit2
git fetch https://github.com/ros-planning/moveit master  # fetch MoveIt 1 branch
git checkout -b melodic-sync FETCH_HEAD  # check it out as branch melodic-sync
git push origin melodic-sync  # push to file a PR
```
Subsequently, file a PR for the created `melodic-sync` branch and merge using a standard merge-commit.
----------------------------------

davetcoleman, 2019-03-15:
I don't think we should push a large amount of code (a bunch of commits from moveit1) into a repo without first ensuring that the CI passes and the code changes are reviewed.

rhaschke, 2019-03-15:
> I don't think we should push a large amount of code (a bunch of commits from moveit1) into a repo without first ensuring that the CI passes and the code changes are reviewed.

I fully agree! How is your comment, @davetcoleman, related to my suggestion for a sync strategy?

davetcoleman, 2019-03-15:
i added the quote i was replying to in my above comment.

@rhaschke also do you mind doing another moveit1 / moveit2 sync? its been a week since the last one and you're certainly the most experienced to take it on.

rhaschke, 2019-03-16:
> @rhaschke also do you mind doing another moveit1 / moveit2 sync?

Done: #47.

davetcoleman, 2019-03-19:
@rhaschke this is what the current settings are for the ``moveit2`` repo:

[image]

I believe what you are asking for is to have no protection to the ``master`` branch, correct?

What if instead we temporarily disabled this protection just for the occasional moveit1/moveit2 sync. Would that work?

Still, I'm concerned we will then break the master build since we won't be able to run CI and there will surely be regressions during the sync. Do you have a suggestion for this?

Finally, what would your exact git commands be if you could push directly to ``master``?

rhaschke, 2019-03-20:
> I believe what you are asking for is to have no protection to the master branch, correct?

Yes.

> What if instead we temporarily disabled this protection just for the occasional moveit1/moveit2 sync.

That should be sufficient.

> Still, I'm concerned we will then break the master build since we won't be able to run CI and there will surely be regressions during the sync. Do you have a suggestion for this?

I need to emphasize once more, that I never suggested to push to master without running Travis or having reviews beforehand. Consider for example #47. This was a essentially a merge-commit that was prepared by me, resolving all conflicts from merging MoveIt 1 commits into MoveIt 2.

However, instead of using github's web gui for merging, which creates an extra merge commit, I suggest to fast-forward master to this prepared merge-commit _after_ successful review and Travis check.

> Finally, what would your exact git commands be if you could push directly to master?

This can be easily accomplished - from commandline - with
`git push origin master`

If you like, @davetcoleman, I can handle this is future.

This commit tree shows the extra merge-commits generated by github (the 2 selected commits):
[image]

closed 2021-11-08 by henningkayser