moveit/moveit2 #97: Failed to reproduce the first demonstrator of MoveIt2

RoboticsYY, 2019-08-06:
### Description

I followed [moveit-2-journey-first-demonstrator/](https://acutronicrobotics.com/news/moveit-2-journey-first-demonstrator/) to reproduce the first demonstrator of MoveIt2 with MARA robot. However, the efforts failed either with the pre-built docker container or with the source build. 

### Your environment
* ROS Distro: dashing
* OS Version: Ubuntu 18.04
* RMW: FastRTPS

### Problems

#### 1. Failure with the docker container

* https://github.com/AcutronicRobotics/MARA dashing

At first, I tried the instructions using docker container `acutronicrobotics/moveit2:dashing-alpha`.
 
- MARA gazebo launched successfully.  
- */root/moveit2_dashing_alpha/lib/moveit_ros_move_group/move_group* launched successfully.
- */root/moveit2_dashing_alpha/lib/mara_moveit_tutorials/follow_joint_trajectory_node* launched successfully.
- *ros2 launch test_app.launch.py* launched successfully.

It seemed that the path planning managed to succeed, which printed:
```
[INFO] [moveit_ros_planning.move_group_interface]: Ready to take commands for planning group manipulator .
[INFO] [move_group_interface_demo]: Listening to joint states on topic 'joint_states'
current state! ok!
[INFO] [move_group_interface_demo]: Reference frame: world
[INFO] [move_group_interface_demo]: End effector link: ee_link
current state! ok!
[INFO] [move_group_interface_demo]: Visualizing plan 2 (joint space goal) 
6 6 6
motor1 0.00000
motor2 0.00000
motor3 0.00001
motor4 0.00000
motor5 0.00000
motor6 -0.00000
0.00000	0.00000	0.00001	0.00000	0.00000	-0.00000	
time_from_start 0 0
-0.04774	0.00000	0.07495	-0.00000	0.00001	-0.00000	
time_from_start 0 387732368
-0.09547	0.00001	0.14989	-0.00000	0.00001	-0.00000	
time_from_start 0 549350976
-0.14321	0.00001	0.22483	-0.00001	0.00002	-0.00000	
time_from_start 0 673603846
-0.19094	0.00001	0.29977	-0.00001	0.00002	-0.00000	
time_from_start 0 779779955
-0.23584	0.00001	0.37025	-0.00001	0.00002	-0.00000	
time_from_start 0 868979284
-0.28074	0.00001	0.44074	-0.00001	0.00003	-0.00000	
time_from_start 0 949730274
-0.31535	0.00001	0.49507	-0.00001	0.00003	-0.00000	
time_from_start 1 8374859
-0.34996	0.00002	0.54941	-0.00002	0.00003	-0.00000	
time_from_start 1 68796433
-0.36400	0.00002	0.57145	-0.00002	0.00003	-0.00000	
time_from_start 1 94051680
-0.37804	0.00002	0.59350	-0.00002	0.00003	-0.00000	
time_from_start 1 119559479
-0.39386	0.00002	0.61833	-0.00002	0.00003	-0.00000	
time_from_start 1 148869276
-0.40967	0.00002	0.64316	-0.00002	0.00003	-0.00000	
time_from_start 1 178768200
-0.44961	0.00002	0.70586	-0.00002	0.00003	-0.00000	
time_from_start 1 258113284
-0.48954	0.00002	0.76855	-0.00002	0.00003	-0.00000	
time_from_start 1 344891834
-0.53395	0.00002	0.83828	-0.00002	0.00003	-0.00000	
time_from_start 1 453881888
[... 103 lines cut ...]
[INFO] [kinematics_plugin_loader]: Using kinematics solver 'kdl_kinematics_plugin/KDLKinematicsPlugin' for group 'manipulator'.
[INFO] [kinematics_plugin_loader]: 1 param is kdl_kinematics_plugin/KDLKinematicsPlugin 
[INFO] [kinematics_plugin_loader]: Looking for param endeffector.kinematics_solver 
[INFO] [kinematics_plugin_loader]: Looking for param robot_description_kinematics.endeffector.kinematics_solver 
[INFO] [kinematics_plugin_loader]: Using kinematics solver 'not' for group 'endeffector'.
[INFO] [kinematics_plugin_loader]: Using kinematics solver 'set' for group 'endeffector'.
[INFO] [kinematics_plugin_loader]: 1 param is not set 
[INFO] [kinematics_plugin_loader]: Trying to allocate kinematics solver for group 'manipulator'
[INFO] [kinematics_plugin_loader]: Choosing tip frame of kinematic solver for groupbased on last link in chain
[INFO] [kinematics_plugin_loader]: Planning group 'manipulator' has tip(s): ee_link, 
Link base_link had 1 children
Link base_robot had 1 children
Link motor1_link had 2 children
Link motor1_cover had 0 children
Link motor2_link had 1 children
Link motor3_link had 2 children
Link motor3_cover had 0 children
Link motor4_link had 1 children
Link motor5_link had 2 children
Link motor5_cover had 0 children
Link motor6_link had 2 children
Link ee_link had 0 children
Link tool0 had 0 children
```
Can anyone who succeed to reproduce the first demonstrator provide some suggestions? Did I miss something?

RoboticsYY, 2019-08-16:
Seems the segmentation fault happens [here](https://github.com/AcutronicRobotics/moveit2/blob/238996d3006c77faaa2f7ec8d21354b60691bf3b/moveit_kinematics/kdl_kinematics_plugin/src/kdl_kinematics_plugin.cpp#L199), `node_` which is a shared pointer to a ROS2 node is not initialized anywhere.

henningkayser, 2019-10-18:
@RoboticsYY thanks for reporting this issue! You're right about `node_` not being initialized. As we are right now starting to continue development things like these will be fixed soon and running on the official `ros-planning/moveit` repository.

closed 2020-05-13 by henningkayser