moveit/moveit2 #116: Porting from AcutronicRobotics repo progress table

JafarAbdi, 2019-10-28:
| Package name | Package name | Package name                     |        PR opened             | PR merged            | Notes    |
|:------------:|:------------:|:--------------------------------:|:----------------------------:|:--------------------:|:--------:|
|moveit_ros    |              |                                  | :white_medium_square:        | :white_medium_square:|          |
|              | planning     |                                  | :white_medium_square:        | :white_medium_square:|          |
|              |              | robot_model_loader               | :heavy_check_mark:           | :white_medium_square:|          |
|              |              | trajectory_execution_manager     | :heavy_check_mark:           | :white_medium_square:|          |
|              |              | planning_request_adapter_plugins | :heavy_check_mark:           | :white_medium_square:|          |
|              |              | constraint_sampler_manager_loader| :heavy_check_mark:           | :white_medium_square:|          |
|              |              | rdf_loader                       | :heavy_check_mark:           | :white_medium_square:|          |
|              |              | planning_pipeline                | :white_medium_square:        | :white_medium_square:|New commits added after merging - Need Update|
|              |              | plan_execution                   | :heavy_check_mark:           | :white_medium_square:|          |
|              |              | kinematics_plugin_loader         | :heavy_check_mark:           | :white_medium_square:|          |
|              |              | planning_components_tools        | :white_medium_square:        | :white_medium_square:|No progress made for porting|
|              |              | collision_plugin_loader          | :heavy_check_mark:           | :white_medium_square:|          |
|              |              | planning_scene_monitor           | :heavy_check_mark:           | :white_medium_square:|          |
|              |benchmarks    |                                  | :white_medium_square:        | :white_medium_square:|No progress made for porting|
|              |manipulation  |                                  | :white_medium_square:        | :white_medium_square:||
|              |move_group    |                                  | :white_medium_square:        | :white_medium_square:||
|              |perception    |                                  | :white_medium_square:        | :white_medium_square:|No progress made for porting|
|              |planning_interface|                              | :white_medium_square:        | :white_medium_square:||
|              |              | py_bindings_tools        | :white_medium_square:        | :white_medium_square:||
|              |              | common_planning_interface_objects        | :white_medium_square:        | :white_medium_square:||
|              |              | planning_scene_interface        | :white_medium_square:        | :white_medium_square:||
|              |              | move_group_interface        | :white_medium_square:        | :white_medium_square:||
|              |              | robot_interface        | :white_medium_square:        | :white_medium_square:||
|              |              | test        | :white_medium_square:        | :white_medium_square:||
|              |robot_interaction|                               | :white_medium_square:        | :white_medium_square:||
|              |visualization |                                  | :white_medium_square:        | :white_medium_square:||
|              |warehouse     |                                  | :white_medium_square:        | :white_medium_square:||
|moveit_core   |              |                                  | :heavy_check_mark:           | :heavy_check_mark:   ||

JafarAbdi, 2019-11-04:
This's a [link](https://docs.google.com/spreadsheets/d/1aPb3hNP213iPHQIYgcnCYh9cGFUlZmi_06E_9iTSsOI/edit?usp=sharing) to GSheet that show the progress

henningkayser, 2020-05-13:
Closing this since it's documented in the Readme

closed 2020-05-13 by henningkayser