moveit/moveit2 #121: Port all package dependencies

henningkayser, 2019-11-07:
I've given you access to our forked package dependencies which are partially migrated but still need some fixup (see warnings and upstream PRs).
https://github.com/PickNikRobotics/geometric_shapes
https://github.com/PickNikRobotics/object_recognition_msgs
https://github.com/PickNikRobotics/octomap
https://github.com/PickNikRobotics/octomap_msgs
https://github.com/PickNikRobotics/srdfdom

@RoboticsYY Could you start fixing these up and filing new PRs? That would be really helpful.

RoboticsYY, 2019-11-08:
@henningkayser Thanks for giving me access! I have fixed up some warnings of moveit_core, I will continue to fix the warnings in these dependency packages as well.

RoboticsYY, 2019-11-27:
Submitted PRs to the upstream projects:

- [ ] object_recognition_msgs: [PR](https://github.com/wg-perception/object_recognition_msgs/pull/8)
- [ ] octomap_msgs: [PR](https://github.com/OctoMap/octomap_msgs/pull/13)

closed 2020-01-31 by henningkayser