moveit/moveit2 #129: Re-implement dynamics_reconfigure

henningkayser, 2019-11-25:
We need to re-implement runtime updates of node parameters dynamics_reconfigure.
A rather straight-forward approach would use parameter event callbacks as described in this example:
https://github.com/ros2/tutorials/blob/master/rclcpp_tutorials/src/parameters/parameter_events.cpp

The only downside I see is that we don't automatically register "configurable nodes" and there is no `reconfigure_gui` anymore.

mkhansen-intel, 2019-11-26:
Henning, I believe rqt_reconfigure is available now for dashing:
https://github.com/ros-visualization/rqt_reconfigure/tree/dashing

We have not implemented the reconfigure callbacks as of yet, but here is one PR to do so for AMCL.
https://github.com/ros-planning/navigation2/pull/1378

henningkayser, 2019-11-26:
> Henning, I believe rqt_reconfigure is available now for dashing:
> https://github.com/ros-visualization/rqt_reconfigure/tree/dashing
> 

Thanks, I wasn't aware of this one.

> We have not implemented the reconfigure callbacks as of yet, but here is one PR to do so for AMCL.
> [ros-planning/navigation2#1378](https://github.com/ros-planning/navigation2/pull/1378)

Great to have this as an example.

mkhansen-intel, 2019-11-26:
Henning, there is also a PR in progress to create a ParameterEventsSubscriber class in rclcpp:
https://github.com/ros2/rclcpp/pull/829

If you have time please take a look at it and provide any feedback if it is useful for MoveIt2.

henningkayser, 2022-10-10:
With https://github.com/PickNikRobotics/generate_parameter_library being applied, this issue can be closed.

closed 2022-10-10 by henningkayser