moveit/moveit2 #146: Convert to ROS 2 launch files

JafarAbdi, 2019-12-19:
In ROS 2 we have two ways to write launch files 

- Python API
- XML based (It's not supported in [dashing](https://github.com/ros2/launch/tree/dashing), but it does in [eloquent](https://github.com/ros2/launch/tree/eloquent))

Even if we decided to move on with XML based one we have to do some migration steps [see](https://index.ros.org/doc/ros2/Tutorials/Launch-files-migration-guide/) 

[Reference](https://design.ros2.org/articles/roslaunch.html)

JafarAbdi, 2019-12-23:
Python launch file [example](https://gist.github.com/JafarAbdi/167016893126e0adc477530238357b59) planning_context + kdl_tests

ddengster, 2020-04-07:
i recommend using the Python API; it's better refined

AliaChe, 2020-07-08:
Hello, in moveit1, there is launch files generated by the setup_assistant that permits to launch the move_group node and all necessary plugins. Is there any other examples of these launch files in moveit2?