moveit/moveit2 #151: stderr: moveit_core

1000sprites, 2019-12-30:
/usr/bin/ld: cannot find -lEigen3::Eigen
collect2: error: ld returned 1 exit status
make[2]: *** [robot_model/libmoveit_robot_model.so.2.0.0] Error 1
make[1]: *** [robot_model/CMakeFiles/moveit_robot_model.dir/all] Error 2
make[1]: *** Waiting for unfinished jobs....

1000sprites, 2019-12-30:
Starting >>> moveit_core
--- stderr: moveit_core
CMake Error at /home/ssw/ros2_dashing/install/ament_cmake_gtest/share/ament_cmake_gtest/cmake/ament_add_gtest_executable.cmake:50 (add_executable):
  Target "test_robot_model" links to target "Eigen3::Eigen" but the target
  was not found.  Perhaps a find_package() call is missing for an IMPORTED
  target, or an ALIAS target is missing?
Call Stack (most recent call first):
  /home/xxx/ros2_dashing/install/ament_cmake_gtest/share/ament_cmake_gtest/cmake/ament_add_gtest_executable.cmake:37 (_ament_add_gtest_executable)
  /home/xxx/ros2_dashing/install/ament_cmake_gtest/share/ament_cmake_gtest/cmake/ament_add_gtest.cmake:66 (ament_add_gtest_executable)
  robot_model/CMakeLists.txt:33 (ament_add_gtest)

1000sprites, 2019-12-30:
Starting >>> moveit_core
--- stderr: moveit_core
CMake Error at /home/xxx/ros2_dashing/install/ament_cmake_gtest/share/ament_cmake_gtest/cmake/ament_add_gtest_executable.cmake:50 (add_executable):
  Target "test_robot_model" links to target "Eigen3::Eigen" but the target
  was not found.  Perhaps a find_package() call is missing for an IMPORTED
  target, or an ALIAS target is missing?
Call Stack (most recent call first):
  /home/xxx/ros2_dashing/install/ament_cmake_gtest/share/ament_cmake_gtest/cmake/ament_add_gtest_executable.cmake:37 (_ament_add_gtest_executable)
  /home/xxx/ros2_dashing/install/ament_cmake_gtest/share/ament_cmake_gtest/cmake/ament_add_gtest.cmake:66 (ament_add_gtest_executable)
  robot_model/CMakeLists.txt:33 (ament_add_gtest)

JafarAbdi, 2019-12-30:
Hi @1000sprites,

Did you follow the instruction in the main [README](https://github.com/ros-planning/moveit2#build-from-source) .?

Seem like you don't have Eigen3 installed,

You should run ` rosdep install -r --from-paths . --ignore-src --rosdistro dashing -y` before compiling

1000sprites, 2019-12-31:
@ JafarAbdi  I have installed Eigen3.  I spent a lot of time trying various methods, and finally solved this problem.  The method is to comment the code Eigen3::Eigen of target_link_libraries in ~/src/moveit2/moveit_core/robot_model/CMakeLists.txt.

henningkayser, 2020-01-31:
@1000sprites You're right, linking Eigen3 should have been implemented using  https://github.com/ros2/eigen3_cmake_module.
Please verify if #164 solves this for you.

closed 2020-01-31 by mamoll

ostifede02, 2024-05-16:
Hello!
Unfortunately I get a similar issue. I have re-installed ROS2 and followed the tutorial from [here](https://moveit.ros.org/install-moveit2/source/).

Thank you in advance for your help.

`--- stderr: moveit_core
CMake Error at /opt/ros/humble/share/kdl_parser/cmake/export_kdl_parserExport.cmake:56 (set_target_properties):
  The link interface of target "kdl_parser::kdl_parser" contains:

    urdfdom_headers::urdfdom_headers

  but the target was not found.  Possible reasons include:

    * There is a typo in the target name.
    * A find_package call is missing for an IMPORTED target.
    * An ALIAS target is missing.

Call Stack (most recent call first):
  /opt/ros/humble/share/kdl_parser/cmake/ament_cmake_export_targets-extras.cmake:9 (include)
  /opt/ros/humble/share/kdl_parser/cmake/kdl_parserConfig.cmake:41 (include)
  CMakeLists.txt:19 (find_package)

CMake Error at /opt/ros/humble/share/srdfdom/cmake/srdfdomTargetsExport.cmake:56 (set_target_properties):
  The link interface of target "srdfdom::srdfdom" contains:

    urdfdom_headers::urdfdom_headers

  but the target was not found.  Possible reasons include:

    * There is a typo in the target name.
    * A find_package call is missing for an IMPORTED target.
    * An ALIAS target is missing.

Call Stack (most recent call first):
  /opt/ros/humble/share/srdfdom/cmake/ament_cmake_export_targets-extras.cmake:9 (include)
  /opt/ros/humble/share/srdfdom/cmake/srdfdomConfig.cmake:41 (include)
  CMakeLists.txt:29 (find_package)

CMake Error at /opt/ros/humble/share/urdf/cmake/urdfExport.cmake:56 (set_target_properties):
  The link interface of target "urdf::urdf" contains:

    urdfdom::urdfdom_sensor

  but the target was not found.  Possible reasons include:

    * There is a typo in the target name.
    * A find_package call is missing for an IMPORTED target.
    * An ALIAS target is missing.

Call Stack (most recent call first):
  /opt/ros/humble/share/urdf/cmake/ament_cmake_export_targets-extras.cmake:9 (include)
  /opt/ros/humble/share/urdf/cmake/urdfConfig.cmake:41 (include)
  /opt/ros/humble/share/srdfdom/cmake/ament_cmake_export_dependencies-extras.cmake:21 (find_package)
  /opt/ros/humble/share/srdfdom/cmake/srdfdomConfig.cmake:41 (include)
  CMakeLists.txt:29 (find_package)

CMake Error at /opt/ros/humble/share/urdf_parser_plugin/cmake/urdf_parser_plugin-exportExport.cmake:56 (set_target_properties):
  The link interface of target "urdf_parser_plugin::urdf_parser_plugin"
  contains:

    urdfdom_headers::urdfdom_headers

  but the target was not found.  Possible reasons include:

    * There is a typo in the target name.
    * A find_package call is missing for an IMPORTED target.
    * An ALIAS target is missing.

Call Stack (most recent call first):
  /opt/ros/humble/share/urdf_parser_plugin/cmake/ament_cmake_export_targets-extras.cmake:9 (include)
  /opt/ros/humble/share/urdf_parser_plugin/cmake/urdf_parser_pluginConfig.cmake:41 (include)
  /opt/ros/humble/share/urdf/cmake/ament_cmake_export_dependencies-extras.cmake:21 (find_package)
  /opt/ros/humble/share/urdf/cmake/urdfConfig.cmake:41 (include)
  /opt/ros/humble/share/srdfdom/cmake/ament_cmake_export_dependencies-extras.cmake:21 (find_package)
  /opt/ros/humble/share/srdfdom/cmake/srdfdomConfig.cmake:41 (include)
  CMakeLists.txt:29 (find_package)

CMake Generate step failed.  Build files cannot be regenerated correctly.
---
Failed   <<< moveit_core [2.31s, exited with code 1]

`

PR #164 (2020-01-31): Fix Eigen3 linking issue in moveit_core
Fixes #151 using https://github.com/ros2/eigen3_cmake_module
commit ddc9292a (2020-01-31): Fix Eigen3 linking issue in moveit_core (#164)

* Closes #151