moveit/moveit2 #156: eigen_stl_containers missing

mhubii, 2020-01-20:
The `geometric_shapes` dependency ([moveit2.repos](https://github.com/ros-planning/moveit2/blob/defd82fbbf29ad65c4bd75af042eedba29bc70b8/moveit2.repos#L26)) depends on `eigen_stl_containers`, but this dependency is not mentioned in the `.repos` file

JafarAbdi, 2020-01-20:
Hi @mhubii,

Please see my comment on the PR you opened

closed 2020-01-20 by mhubii

camrbuss, 2020-03-21:
I am getting the following error even after following the README and the PR instructions.

After installing the apt package and the rosdep install
`sudo apt install ros-eloquent-eigen-stl-containers`
`rosdep install -r --from-paths . --ignore-src --rosdistro eloquent -y`

I get: 

```
Starting >>> geometric_shapes
--- stderr: geometric_shapes
CMake Error at CMakeLists.txt:37 (find_package):
  By not providing "Findeigen_stl_containers.cmake" in CMAKE_MODULE_PATH this
  project has asked CMake to find a package configuration file provided by
  "eigen_stl_containers", but CMake did not find one.

  Could not find a package configuration file provided by
  "eigen_stl_containers" with any of the following names:

    eigen_stl_containersConfig.cmake
    eigen_stl_containers-config.cmake

  Add the installation prefix of "eigen_stl_containers" to CMAKE_PREFIX_PATH
  or set "eigen_stl_containers_DIR" to a directory containing one of the
  above files.  If "eigen_stl_containers" provides a separate development
  package or SDK, be sure it has been installed.
```

yaozhixin3190, 2020-06-29:
> I am getting the following error even after following the README and the PR instructions.
>
> After installing the apt package and the rosdep install
> `sudo apt install [...]

I have the same problem, do you solve it?

doitmaan, 2020-09-07:
Issue Reproduced on Ubuntu 20.04 LTS during building the workspace

> Starting >>> urdfdom_headers
Finished <<< urdfdom_headers [0.86s]
Starting >>> geometric_shapes
--- stderr: geometric_shapes
CMake Error at CMakeLists.txt:41 (find_package):
  By not providing "Findeigen_stl_containers.cmake" in CMAKE_MODULE_PATH this
  project has asked CMake to find a package configuration file provided by
  "eigen_stl_containers", but CMake did not find one.

 

>  Could not find a package configuration file provided by

  "eigen_stl_containers" with any of the following names:

    eigen_stl_containersConfig.cmake
    eigen_stl_containers-config.cmake

  

> Add the installation prefix of "eigen_stl_containers" to CMAKE_PREFIX_PATH
or set "eigen_stl_containers_DIR" to a directory containing one of the
  above files.  If "eigen_stl_containers" provides a separate development
  package or SDK, be sure it has been installed.
---

> Failed   <<< geometric_shapes [1.41s, exited with code 1]
> Summary: 1 package finished [2.55s]
>   1 package failed: geometric_shapes
>   1 package had stderr output: geometric_shapes
>   35 packages not processed

`

mhubii, 2020-09-16:
well as far as I remember, I added it to the `.repos` file, so it got downloaded for me anyways back then. But I am not the provider of this package, so its not up to me to decide

Drojas251, 2021-06-17:
@JafarAbdi  I am having the same issue when building ROS2 Foxy and Moveit2 from source even when running 
`rosdep install -r --from-paths . --ignore-src --rosdistro foxy -y`

Any ideas why this dependency is not found?

JafarAbdi, 2021-06-17:
> @JafarAbdi I am having the same issue when building ROS2 Foxy and Moveit2 from source even when running
> `rosdep install -r --from-paths . --ignore-src --rosdistro foxy -y`
> 
> Any ideas why this dependency is not found?

Did you run `source /opt/ros/foxy/setup.bash` before building .?

Drojas251, 2021-06-17:
@JafarAbdi Forgive my ignorance, but when you install ROS2 from source you should source the workspace correct? 
`source ros2_ws/install/setup.bash`

I sourced the workspace prior to building. Should I run this instead?
`source /opt/ros/foxy/setup.bash`

JafarAbdi, 2021-06-17:
> @JafarAbdi Forgive my ignorance, but when you install ROS2 from source you should source the workspace correct?
> `source ros2_ws/install/setup.bash`
> 
> I sourced the workspace prior to building. Should I run this instead?
> `source /opt/ros/foxy/setup.bash`

That's right if you built ROS2 from source you need to run `source ros2_ws/install/setup.bash` but in this case, you need to install the dependencies from source otherwise, you may run into bugs/segfaults due to ABI incompatibility or get errors about not finding some shared libraries in case building moveit2 succeeded

Drojas251, 2021-06-17:
@JafarAbdi Okay great, that is what I thought. Is there a direct way I can find this list of dependencies needed to build Moveit2 with a ROS2 install from source? Is there a README.md somewhere that walks through the steps up building Moveit2 with ROS2 installed form source?

JafarAbdi, 2021-06-17:
> @JafarAbdi Okay great, that is what I thought. Is there a direct way I can find this list of dependencies needed to build Moveit2 with a ROS2 install from source? Is there a README.md somewhere that walks through the steps up building Moveit2 with ROS2 installed form source?

You could run the following command in the workspace to get all the dependencies `rosdep install --reinstall --simulate --ignore-src --from-paths .`

Drojas251, 2021-06-17:
@JafarAbdi This command should be ran in the ros2 workspace or moveit2 workspace?

JafarAbdi, 2021-06-17:
> @JafarAbdi This command should be ran in the ros2 workspace or moveit2 workspace?

In moveit2 workspace

Drojas251, 2021-06-17:
@JafarAbdi Great, thank you very much for you help!

peterdavidfagan, 2022-01-04:
Hi @JafarAbdi, 

I am facing the same error when building MoveIt 2 from source. I am receiving the message that "All required rosdeps installed successfully" when I look to install package dependencies in my ws_moveit2 workspace. However when I try to build MoveIt 2 I receive an error in cmake finding eigen_stl_containers. Below is a screenshot example:

[image]

It also seems that eigen_stl_containers is installed for the rolling distribution of ROS 2:

[image]

I have installed ROS 2 from source and I have also sourced the setup.bash file within my current bash session. 

I believe I am missing something basic here, I would very much appreciate your guidance on this.

peterdavidfagan, 2022-01-05:
Hi @JafarAbdi,

I believe the above error people are encountering is due to the management of dependencies and how ROS 2 is being built. For instance if I build ROS 2 rolling from source then I would expect to source `~/ros2_rolling/install/local_setup.bash`  , however, if I subsequently use rosdep to install the dependencies required for building MoveIt 2 the dependent packages will be included at path `/opt/ros/rolling/include/` and hence if I only source `~/ros2_rolling/install/local_setup.bash` before attempting to build MoveIt 2 I will encounter build errors (eigen_stl_containers error). 

The MoveIt 2 build guide actually appears to highlight that it only applies to ROS 2 installations from Debian packages. I have opened an issue on this for the guide above in case it is helpful to update the guide with additional details for when ROS 2 is built from source.

If there are additional notes you believe would be useful to include in the guide based on the above I'd be happy to work on a PR for these.

tsadarsh, 2022-06-10:
> That's right if you built ROS2 from source you need to run `source ros2_ws/install/setup.bash` but in this case, you need to install the dependencies from source otherwise, you may run into [...]

I built ros2 from the available binary. But I still had to *source* using the same command: `source ros2_ws/install/setup.bash` to solve this issue. 
But unfortunately the build fails again in the second step. 
```
Failed   <<< moveit_visual_tools [7.89s, exited with code 1]

Summary: 1 package finished [9.68s]
  1 package failed: moveit_visual_tools
  1 package had stderr output: moveit_visual_tools
  1 package not processed
```
I have referenced this issue [here](https://github.com/ros-planning/moveit_visual_tools/issues/106#issuecomment-1152300375). It would be great if someone could help me here.

LitchiCheng, 2025-01-04:
It works by below steps:
```
sudo apt install ros-foxy-eigen-stl-containers
rosdep install -r --from-paths . --ignore-src --rosdistro foxy -y
source /opt/ros/foxy/setup.bash
```