moveit/moveit2 #176: type conversion error in moveit_core

iamrajee, 2020-03-09:
### Description
Title: Error in moveit_core while building it for the first time in workspace as
`colcon build --event-handlers desktop_notification- status- --cmake-args -DCMAKE_BUILD_TYPE=Release`

### Environment
* ROS Distro: [eloquent]
* OS Version: e.g. Ubuntu 18.04
* Binary build
* distribution status    : active
* moveit2 : cloned from ros-planning
/moveit2(master branch) on 9/03/2020.

shapes::constructShapeFromMsg() function in [conversions.cpp](https://github.com/ros-planning/moveit2/blob/master/moveit_core/robot_state/src/conversions.cpp#L272) file expect argument of type `const SolidPrimitive& {aka const shape_msgs::SolidPrimitive_<std::allocator<void> >&}` but we are passing an argument of type `const value_type {aka const shape_msgs::msg::SolidPrimitive_<std::allocator<void> >}`, thus generating build time error as
```
moveit2/moveit_core/robot_state/src/conversions.cpp: In function ‘void moveit::core::{anonymous}::_msgToAttachedBody(const moveit::core::Transforms*, const AttachedCollisionObject&, moveit::core::RobotState&)’:
.../moveit_core/robot_state/src/conversions.cpp:272:84: error: **no matching function for call** to ‘constructShapeFromMsg(const value_type&)’
           shapes::Shape* s = shapes::constructShapeFromMsg(aco.object.primitives[i]);
                                                                                    ^
In file included from ../moveit_core/robot_state/src/conversions.cpp:39:0:
/opt/ros/melodic/include/geometric_shapes/shape_operations.h:49:8: note: candidate: shapes::Shape* shapes::constructShapeFromMsg(const SolidPrimitive&)
 Shape* constructShapeFromMsg(const shape_msgs::SolidPrimitive& shape_msg);
        ^~~~~~~~~~~~~~~~~~~~~
/opt/ros/melodic/include/geometric_shapes/shape_operations.h:49:8: note:   no known conversion for argument 1 from ‘**const value_type** {aka const shape_msgs::msg::SolidPrimitive_<std::allocator<void> >}’ to ‘**const SolidPrimitive&** {aka const shape_msgs::SolidPrimitive_<std::allocator<void> >&}’
```
You can see the full error msg can be found [here](https://gist.github.com/iamrajee/c761766bc338beef1b7fbb1e9b774986).

I tried several to approach this error e.g
Calling the above mentioned function like this
`shapes::Shape* s = shapes::constructShapeFromMsg(aco.object.primitives[i]);`
insead of like this
`shapes::Shape* s = shapes::constructShapeFromMsg((const shape_msgs::msg::SolidPrimitive&)aco.object.primitives[i]);`

Now this generated the error that
`no known conversion for argument 1 from ‘**const SolidPrimitive** {aka const shape_msgs::msg::SolidPrimitive_<std::allocator<void> >}’ to ‘**const SolidPrimitive&** {aka const shape_msgs::SolidPrimitive_<std::allocator<void> >&}’`
I tried few this to resolve this too, but with no luck.

JafarAbdi, 2020-03-09:
You're mixing ROS 1 and ROS 2 as you can see in the log file you shared the included header are from melodic whereas you're building a ROS 2 package

iamrajee, 2020-03-09:
> You're mixing ROS 1 and ROS 2 as you can see in the log file you shared the included header are from melodic whereas you're building a ROS 2 package

Yes, I have ROS1 Melodic and ROS eloquent in the same environment. I had properly sourced ROS2 environment in the ROS2 workspace as a new terminal as
```
source /opt/ros/eloquent/setup.bash
source install/local_setup.bash
source install/setup.bash
```
How else do I ensure that I'm not mixing ROS1 and ROS2 environment?

In my **~/.bashrc** I have cmd to source ROS1 environment, could it be because of it?

JafarAbdi, 2020-03-09:
Yes, it could be the reason, I personally have a prompt that asks which version I want to use and source it accordingly,  

```bash
  while [[ "$resp" != "ros" && "$resp" != "ros2" ]]; do
    read -p "Which version of ROS you want? (ros/ros2)" resp;
  done

  if [ "$resp" == "ros" ]; then
    export ROS_DISTRO="melodic"
  elif [ "$resp" == "ros2" ]; then
    export ROS_DISTRO="eloquent"
  fi
  source /opt/ros/$ROS_DISTRO/setup.bash
```

iamrajee, 2020-03-09:
> Yes, it could be the reason, I personally have a prompt that asks which version I want to use and source it accordingly,
Many thanks, this solved my issue.

closed 2020-03-09 by iamrajee