moveit/moveit2 #180: computeCartesianPath(...) deprecated in robot_state.cpp

iamrajee, 2020-03-24:
### Environment
ROS Distro: [eloquent]
OS Version: e.g. Ubuntu 18.04
Binary build
distribution status: active
moveit2: cloned from ros-planning:moveit2:master branch at this [state](https://github.com/ros-planning/moveit2/tree/6a23d3284d6d1e1d02347612da348b01519e8745).

### Description
I'm porting [MTC](https://github.com/iamrajee/ros2eloquent_moveit_ws/tree/master/src/moveit_task_constructor) to ros2 eloquent and currently ported cartesian_position_motion.cpp as [this](https://github.com/ros-planning/moveit_task_constructor/commit/be828860f469c7dd81bb95968220ae0bd317930f). While building the MTC I got the error stating that the `computeCartesianPath(...)` function is deprecated. I can also see that same function has deprecated here at [robot_state.h#L1058](https://github.com/ros-planning/moveit2/blob/master/moveit_core/robot_state/include/moveit/robot_state/robot_state.h#L1058). Please suggested to me how should I proceed ahead. 
Would I be able to solve this if I make appropriate modification in `computeCartesianPath(...)` function similar to what is been done [here](https://github.com/AcutronicRobotics/moveit2/blob/master/moveit_core/robot_state/src/robot_state.cpp#L1884) in Acutronic fork of moveit2?

See my full error report [here](https://gist.github.com/iamrajee/ad7bf93ebd804231881ff455231466ab).

 Any help would be greatly appreciated.

v4hn, 2020-03-24:
I do not understand how you end up in this state at all, or why you reference a 3 years old commit from mtc, but anyway...

Is anything wrong with simply migrating to the newer [cartesian_interpolator.h](https://github.com/ros-planning/moveit2/blob/master/moveit_core/robot_state/include/moveit/robot_state/cartesian_interpolator.h)?

iamrajee, 2020-03-24:
> why you reference a 3 years old commit from mtc?
I was following the full commit history of MTC to build it for ros2 eloquent by making appropriate changes as and when required. This way I could better understand the reason for a particular design choice of MTC. Could you suggest me a better way to port MTC?

> Is anything wrong with simply migrating to the newer cartesian_interpolator.h?
I haven't tried that yet. I would rather do that.

Thanks for your response!

v4hn, 2020-03-24:
> I was following the full commit history of MTC to build it for ros2 eloquent by making appropriate changes as and when required. This way I could better understand the reason for a particular design choice of MTC. Could you suggest me a better way to port MTC?

Knowing the history of the project is an awful idea! The only things subject to be changed for ROS2 should be the includes/message types and the introspection code (+ potential changes in rviz for `moveit_task_constructor_visualization`).
There is no merit at all in porting the history like that and, as you noticed already, you would have to look at the old interfaces of MoveIt for that too.

closed 2020-03-24 by v4hn

iamrajee, 2020-03-24:
Alright, I would do as suggested. By the way, Which distro should I target while porting MTC, dashing or eloquent? As I could see there is [some](https://github.com/ros-planning/moveit2/issues/179#issuecomment-601654432) difference in function definition in moveit2 for ROS dashing(Acutronic fork) and ROS eloquent(ros-planning master branch)?

v4hn, 2020-03-24:
> Which distro should I target while porting MTC

I do not know *why* you work on this port right now and your motivation (or employer) should define which distro is targeted. Personally, I would always go with the latest release if it's actually testable on real hardware.

iamrajee, 2020-03-24:
>  _why_ you work on this port right now?

I have been working on a multi-manipulator system using MTC for some time. I got some difficulties in concurrently executing the tasks. I had worked on some [solution](https://github.com/ros-planning/moveit_task_constructor/issues/138#issuecomment-591305341) suggested by you with limited success. Then I came across [this](https://youtu.be/zGM2lYAHrm4) multi-arm demonstration using moveit2. Thus, I was motivated to port MTC to ROS2 by myself. Also, I'm hoping to take this as GSoC.