moveit/moveit2 #181: Method for changing planning algorithm for a particular instance of a PlanningComponent.

Mariunil, 2020-03-25:
We are a couple of master students using Moveit 2 as our motion planning framework in our master's thesis, and are having difficulties changing the planning algorithm that is used for a chosen `PlanningComponent`. We are using OMPL as our planning pipeline, similar to the "run_moveit_cpp" demos. 

In the `PlanningComponent` class, we would like to have a method allowing us to specify what planner  from our loaded planning_pipeline we would like to use for this particular `PlanningComponent`.  

An alternative could be an option to change the default planner in `MoveitCpp`. For instance, it could be an optional parameter in the construction of the `MoveitCpp` instance, or a `set_default_planner(...)` method in the `MoveitCpp` class.

If there already exists ways of changing either the planner used for a particular `PlanningComponent` or the default planner, help finding these would be greatly appreciated.

JafarAbdi, 2020-04-15:
Hi @Mariunil, Shouldn't calling `PlanningComponent::plan(const PlanRequestParameters& parameters)` with `planner_id` and `planning_pipeline` set do what you're looking for .?

Cheers,

Mariunil, 2020-04-17:
Yes it would, must have missed that option. Thank you very much.

closed 2020-05-11 by henningkayser