moveit/moveit2 #182: moveit simple controller build error

jrgnicho, 2020-04-08:
### Description

Overview of your issue here.
While building moveit2 following the instructions given in the main README.md I get the following build error

```
Failed   <<< moveit_simple_controller_manager	[ Exited with code 1 ]
--- stderr: moveit_ros_occupancy_map_monitor
CMake Error at .../moveit2_ws/install/moveit_core/share/moveit_core/cmake/ament_cmake_export_libraries-extras.cmake:48 (message):
  Package 'moveit_core' exports the library 'moveit_exceptions' which
  couldn't be found
Call Stack (most recent call first):
  ...t/moveit2_ws/install/moveit_core/share/moveit_core/cmake/moveit_coreConfig.cmake:38 (include)
  CMakeLists.txt:25 (find_package)
```

### Your environment
* ROS Distro: [eloquent]
* OS Version: Ubuntu 18.04
* Source  build
* master branch
### Steps to reproduce
Follow build instructions in README.md
```
 git clone https://github.com/ros-planning/moveit2.git -b master
 vcs import < moveit2/moveit2.repos
 rosdep install -r --from-paths . --ignore-src --rosdistro eloquent -y
```

```
 cd $COLCON_WS
 colcon build --event-handlers desktop_notification- status- --cmake-args -DCMAKE_BUILD_TYPE=Release
```

### Expected behaviour
Build should finish without errors

### Actual behaviour
Build errors out with the error message shown above

JafarAbdi, 2020-04-08:
Unfortunately, I can't reproduce this locally, did you try to remove the `log`, `install`, and `build` folder and recompile again .? I fixed similar issue by doing so

jrgnicho, 2020-04-08:
I removed `install` and `build` but not `log`  I didn't think `log` mattered but I'll try that as well

JafarAbdi, 2020-04-08:
Yes, you're right keeping `log` should effect anything

jrgnicho, 2020-04-08:
I did a clean build just now by removing all three directories (build, install and log) and the result was the same

jrgnicho, 2020-04-08:
OK somehow I ran into this old post describing a similar [issue](https://www.gitmemory.com/issue/AcutronicRobotics/moveit2/104/500981166) and follow the build command used [here](https://github.com/AcutronicRobotics/moveit2#install-and-test-moveit-2)
```
colcon build --merge-install
```

And the build completed successfully

JafarAbdi, 2020-04-08:
That's the reason why it was failing, ament_cmake have an issue with the versioned shared library, it got fixed by https://github.com/ament/ament_cmake/pull/216/ but it's not released yet for eloquent

jrgnicho, 2020-04-08:
Ok for now I won't be using the `--symlink-install` flag

closed 2020-05-11 by henningkayser