moveit/moveit2 #184: Throws wrong type error while parsing ompl_planning.yaml

jrgnicho, 2020-04-29:
### Description
In loading the config yaml files for MoveItCPP in my [application](https://github.com/swri-robotics/collaborative-robotic-sanding/blob/moveit2-integration-test/crs_application/launch/main.launch.py), 
an incorrect yaml type error is thrown when parsing the ompl_planning.yaml file

Overview of your issue here.
I don't have a screen capture of the actual error message but it seemed to suggest that it expected a string but got a double instead.  I was able to get it to accept the yaml file by placing quotes around the value assigned to `longest_valid_segment_fraction ` field in order to make it a string, [see here ](https://github.com/swri-robotics/collaborative-robotic-sanding/blob/adc30bdfcbf48c0981dc56bfe3998baf6c9ee136/crs_moveit2_config/config/ompl_planning.yaml#L183)

### Your environment
* ROS Distro: [Eloquent]
* OS Version: e.g. Ubuntu 18.04
* Source 
* master branch

### Steps to reproduce
Load ompl_planning.yaml file which includes the `longest_valid_segment_fraction ` set to a double

### Expected behaviour
MoveItCPP should load the yaml file and configure itself from it

### Actual behaviour
Throws exception saying that it expected a string but got a double instead

### Backtrace or Console output

Did't capture :(

JafarAbdi, 2020-04-29:
You're right, it's a bug in [ompl_interface.cpp#L181](https://github.com/ros-planning/moveit2/blob/master/moveit_planners/ompl/ompl_interface/src/ompl_interface.cpp#L181) it loop over the parameters' name and expect string for all of them which isn't the case here

In ROS1 calling a getParam with a different type return false, in ROS2 it throws an exception

jrgnicho, 2020-04-29:
I see, that'll be tricky to get to work right if it throws an exception

JafarAbdi, 2020-04-29:
@jrgnicho How about adding a map of KNOWN_GROUP_PARAMS and ParameterType and getting the parameter that matches the parameter type .?

ddengster, 2020-04-30:
```
rclcpp::Parameter param = node_->get_parameter(group_name_param + "." + k);
        if (param.get_type() == rclcpp::ParameterType::PARAMETER_STRING)
        {
          std::string value = param.get_value<std::string>();
          if (!value.empty())
            specific_group_params[k] = value;
          continue;
        }
        else if (param.get_type() == rclcpp::ParameterType::PARAMETER_DOUBLE)
        {
          double value_d = param.get_value<double>();
          // convert to string using no locale
          specific_group_params[k] = moveit::core::toString(value_d);
          continue;
        }
        else if (param.get_type() == rclcpp::ParameterType::PARAMETER_INTEGER)
        {
          int value_i = param.get_value<int>();
          specific_group_params[k] = std::to_string(value_i);
          continue;
        }
        else if (param.get_type() == rclcpp::ParameterType::PARAMETER_BOOL)
        {
          bool value_b = param.get_value<bool>();
          specific_group_params[k] = std::to_string(value_b);
          continue;
        }
```

Something like this should work.

sidetrack: I have a working move_group node, i'm a bug away from a working version (and a draft PR). If you happen to be fixing a bug that occurs with the srdf injesting the joint values of a group_state with the wrong joint order let me know if you've found a solution.

henningkayser, 2020-05-13:
> Something like this should work.

Probably, but in general I would prefer either a template function (instead of explicit type checks) or introducing a convention that forbids variable types. I also thought that there are already some implicit conversions available, at least for double and int.

> sidetrack: I have a working move_group node, i'm a bug away from a working version (and a draft PR). If you happen to be fixing a bug that occurs with the srdf injesting the joint values of a group_state with the wrong joint order let me know if you've found a solution.

That's interesting, could you start an issue on that?

jrgnicho, 2020-05-13:
> ```
> rclcpp::Parameter param = node_->get_parameter(group_name_param + "." + k);
>         if (param.get_type() == rclcpp::ParameterType::PARAMETER_STRING)
>         {
>           std::string value = [...]

That should work but I'm with @henningkayser on preferring template functions

ddengster, 2020-05-14:
> That's interesting, could you start an issue on that?
 turns out the solution was to port joint reordering functionality over in joint_trajectory_controller. I've integrated it into the [pr](https://github.com/ros-controls/ros2_controllers/pull/36).

henningkayser, 2020-05-14:
> > That's interesting, could you start an issue on that?
> 
> turns out the solution was to port joint reordering functionality over in joint_trajectory_controller. I've integrated it into the jtc [pr](https://github.com/ros-controls/ros2_controllers/pull/36).

Great, thanks!

closed 2021-05-04 by henningkayser