moveit/moveit2 #185: can you add instructions for install moveit2 on windows?

DreamLFairy, 2020-05-10:
Evironment:   windows10  eloquent.
when I failed to build, I can't serach the instructions on windows.
c:\ws>colcon build --merge-install
Starting >>> object_recognition_msgs
Starting >>> octomap_msgs
Starting >>> random_numbers
Starting >>> urdfdom_py
Starting >>> joint_state_publisher
Starting >>> ompl
Starting >>> turtlesim
Finished <<< urdfdom_py [2.09s]
Starting >>> srdfdom
Finished <<< joint_state_publisher [2.25s]
Starting >>> moveit_resources
Starting >>> joint_state_publisher_gui
Finished <<< joint_state_publisher_gui [2.59s]
Finished <<< moveit_resources [12.6s]
Failed   <<< random_numbers     [ Exited with code 1 ]
Aborted  <<< srdfdom
Aborted  <<< octomap_msgs
Aborted  <<< ompl
Aborted  <<< turtlesim
Aborted  <<< object_recognition_msgs

Summary: 4 packages finished [34.0s]
  1 package failed: random_numbers
  5 packages aborted: object_recognition_msgs octomap_msgs ompl srdfdom turtlesim
  5 packages had stderr output: object_recognition_msgs octomap_msgs ompl srdfdom turtlesim
  11 packages not processed

henningkayser, 2020-05-13:
Disclaimer: MoveIt 2 doesn't really support Windows yet, even though it should compile. Please try compiling without "--merge-install" as instructed in the Readme.

tangguotongling, 2021-11-08:
> @henningkayser Adding instructions for source install and binary install for Foxy is on my list. I'm working on fixing a building issue with moveit2 that's cropped up since the 09-02 Foxy release (relating to pybind11_vendor), will add the instructions after it's building and running nicely again.

Thank you for your work and look forward to it~

dsauval, 2021-12-21:
Hi ! Any updates on running things on Windows :) ?

lilustga, 2022-01-03:
Hi @dsauval, this is unfortunately stuck behind the new ms-iot foxy release. It has been delayed for issues unrelated to moveit. With the holidays over things should be moving quicker again.

henningkayser, 2022-06-02:
I'm closing this since https://github.com/ros-planning/moveit.ros.org/pull/688 has been merged already a while ago

closed 2022-06-02 by henningkayser