moveit/moveit2 #188: moveit_exceptions library not found

AndyZe, 2020-05-15:
### Description

When running:

 `colcon build --event-handlers desktop_notification- status- --cmake-args -DCMAKE_BUILD_TYPE=Release`

I get:

```
--- stderr: moveit_ros_occupancy_map_monitor
CMake Error at /home/andyz/ws_ros2/install/moveit_core/share/moveit_core/cmake/ament_cmake_export_libraries-extras.cmake:48 (message):
  Package 'moveit_core' exports the library 'moveit_exceptions' which
  couldn't be found
```

### Your environment
* ROS Distro: Eloquent, binary installation
* OS Version: e.g. Ubuntu 18.04
* Building moveit2 from source (master branch)
* I've triple-checked that I'm sourcing install/setup.bash

JafarAbdi, 2020-05-15:
This seems similar to this https://github.com/ros-planning/moveit2/issues/182, could you please check if the comments there helps .?

AndyZe, 2020-05-15:
^ that did work. Previous usage of `--symlink-install` was my issue too. Thanks, and sorry for not finding it myself

tylerjw, 2021-01-21:
I just ran into this... maybe we should add ament_cmake as a source dependency?

JafarAbdi, 2021-06-30:
This should be fixed in the latest main branch

closed 2021-06-30 by JafarAbdi