moveit/moveit2 #189: resolveConstraintFrames() referenced but not implemented

CapedOpossum, 2020-05-17:
### Description

When building moveit2 for ROS 2 eloquent, the build fails at the `moveit_ros` package, specifically when building the `planning_request_adapter_plugins` in the `planning` sub-directory. The build fails because the `default_planner_request_adapters::ResolveConstraintFrame` class implementation (located at `moveit_ros/planning/planning_request_adapter_plugins/src/resolve_constrant_frames.cpp`) references the `resolveConstraintFrames()` function in the `moveit::kinematics_constrant` namespace, but said function has been commented out for a while. The comment block just above the function implementation in the `moveit_core/kinematics_constrant/utils.cpp` file hinted at the function not being used anywhere, despite its use in `default_planner_request_adapters::ResolveConstraintFrames`. The function declaration, however, was not commented out.

An article published around February this year hinted at plugin functionality not being available, and given the name of the subdirectory containing the broken file (`planning_request_adapter_plugins`), I wonder if that's a directory that must be excluded from the build.

### Your environment
* ROS Distro: Eloquent (ROS2)
* OS Version: macOS 10.13.6
* Source or Binary build? __Source Build__
* If binary, which release version? _N/A_
* If source, which branch? `master`

### Steps to reproduce
1. Install and build ROS 2 eloquent from source, along with any required dependencies.
1. Create a "workspace" directory as the root of the ROS 2 tree for __MoveIt! 2__ (e.g., `moveit2_ws`)
1. Clone `ros-planning/moveit2` onto a `src` folder under the aforementioned "workspace" folder.
1. Do a `colcon build --symlink-install` from the root of the ROS 2 tree.

### Expected behaviour
Packages should build and install under the ROS 2 tree in the `install/` subfolder.

### Actual behaviour
After successfully building `moveit_core`, `moveit_msgs`, `moveit_resources`, `moveit_simple_controller_manager` (not necessarily in that order, and not explicitly listing pre-requisites), build fails at `moveit_ros_planning`.
### Backtrace or Console output

Abbreviated build output follows:

```
Undefined symbols for architecture x86_64:
  "kinematic_constraints::resolveConstraintFrames(moveit::core::RobotState const&, moveit_msgs::msg::Constraints_<std::__1::allocator<void> >&)", referenced from:
      default_planner_request_adapters::ResolveConstraintFrames::adaptAndPlan(boost::function<bool (std::__1::shared_ptr<planning_scene::PlanningScene const> const&, moveit_msgs::msg::MotionPlanRequest_<std::__1::allocator<void> > const&, planning_interface::MotionPlanResponse&)> const&, std::__1::shared_ptr<planning_scene::PlanningScene const> const&, moveit_msgs::msg::MotionPlanRequest_<std::__1::allocator<void> > const&, planning_interface::MotionPlanResponse&, std::__1::vector<unsigned long, std::__1::allocator<unsigned long> >&) const in resolve_constraint_frames.cpp.o
ld: symbol(s) not found for architecture x86_64
clang: error: linker command failed with exit code 1 (use -v to see invocation)
gmake[2]: *** [planning_request_adapter_plugins/CMakeFiles/moveit_default_planning_request_adapter_plugins.dir/build.make:393: planning_request_adapter_plugins/libmoveit_default_planning_request_adapter_plugins..dylib] Error 1
gmake[1]: *** [CMakeFiles/Makefile2:565: planning_request_adapter_plugins/CMakeFiles/moveit_default_planning_request_adapter_plugins.dir/all] Error 2
```

henningkayser, 2020-05-20:
@CapedOpossum thanks for reporting this. I'm right now working on a MoveIt 1 master branch sync and also ran into this, even though it apparently wasn't occurring with the recommended target setup (Ubuntu 18.04, colcon build without --symlink-install). I'll look into enabling this function again when the sync gets merged.

henningkayser, 2020-10-14:
This has been fixed with https://github.com/ros-planning/moveit2/commit/ff27996375ecdd020e069642368e41f6a94fd005. Full macOS support should have been provided with https://github.com/ros-planning/moveit2/pull/271 (for Foxy).

closed 2020-10-14 by henningkayser