moveit/moveit2 #190: Foxy Fitzroy 2.1 Beta Release - July 2020

henningkayser, 2020-05-19:
As already discussed and announced on [moveit.ros.org](moveit.ros.org), we're aiming for a MoveIt 2 Beta release targeting [Foxy Fitzroy](https://index.ros.org/doc/ros2/Releases/Release-Foxy-Fitzroy/) in July. Feature-wise, this release can be equal to the Eloquent Beta version and, so the roadmap is somewhat orthogonal to the overall migration process. However, there are some [changes since Eloquent](https://index.ros.org/doc/ros2/Releases/Release-Foxy-Fitzroy/#changes-since-the-eloquent-release) that require fixing or at least should be addressed.

In particular, we should review:

* Proper use of "modern" CMake interface targets with `ament_target_dependencies`
* Update Node Interface getters if applicable
* Verify tests work with new working directory

Additional steps:
* Review/update rosinstall dependencies + release
* Setup docker images
* Review release status of debian dependencies -> Make update strategy

Usually, we would do all these steps on `master` and then create the release branch from it. However, this would probably interfere with the general ROS 2 migration progress, so in this case I would actually create the `foxy` release branch first and then merge the changes back to master after the release. Alternatively, we could continue migration on the `eloquent` branch and sync that progress back once Foxy support is completed.

### Open questions:

**Can `master` support Foxy and Eloquent?**
Considering the required changes it seems feasible to support both versions, possibly using separate rosinstall files. However, this might not be a long-term solution as we will start adding features that make use of affected API's (subscription callbacks, node interface getters).

**What's our sync strategy?**
I would suggest syncing all features from master into both release branches `eloquent` and `foxy`, until both are ready for a full release (i.e. all packages are ported). Depending on `master` support for both versions, MoveIt 2 roadmap progress will continue targeting `foxy`, where changes are synced back to `eloquent` on a frequent basis. I don't expect too much overhead there, considering the ROS2 changes between Eloquent and Foxy.

davetcoleman, 2020-05-19:
Your first link is dead. Did you mean [this PR](https://github.com/ros-planning/moveit.ros.org/pull/438) by chance?

Is the overhead of maintaining two branches worthwhile, or should we just wait a month (after the official Foxy release) to do the final porting? Tackling two efforts at once sounds hard.

henningkayser, 2020-05-19:
> Your first link is dead. Did you mean [this PR](https://github.com/ros-planning/moveit.ros.org/pull/438) by chance?
> 

Thanks! Fixed it. I just wanted to link the release versions on the website.

> Is the overhead of maintaining two branches worthwhile, or should we just wait a month (after the official Foxy release) to do the final porting? Tackling two efforts at once sounds hard.

To me it seems like the diff between both branches will be pretty small. We would possibly even get the master branch to compile under Foxy and Eloquent so syncing changes to the release branches would be trivial. The only issues I see is that some dependencies won't be available yet and it can take some time to get them migrated and released. I'm totally fine with pushing back the deadline, but I think we should already make plans as it doesn't conflict with the remaining work.

henningkayser, 2020-07-01:
We need to sync+release following dependencies:

* https://github.com/ros-planning/moveit_msgs
* https://github.com/ros-planning/random_numbers
* https://github.com/ros-planning/moveit_resources
* https://github.com/ros-planning/geometric_shapes
* https://github.com/ros-planning/srdfdom
* https://github.com/OctoMap/octomap_msgs

Unclear:
* https://github.com/wg-perception/object_recognition_msgs

Pending releases:
* Octomap: https://github.com/ros/rosdistro/pull/25637

davetcoleman, 2020-09-17:
@henningkayser what is blocking the Foxy debian release?

henningkayser, 2020-09-21:
> @henningkayser what is blocking the Foxy debian release?

This issue https://github.com/ros-planning/moveit2/issues/242

davetcoleman, 2020-09-21:
I just emailed Chris at Open Robotics, perhaps that will help...

closed 2021-05-04 by henningkayser