moveit/moveit2 #216: Port & Fix moveit2 tests

sjahr, 2020-06-15:
This issue gives an overview on the test classes in this repository and their current status to get a better overview which ones still need to be ported and/or fixed. As this is work in progress the table is not yet completed. If you notice any code missing or falsely identified as test code please let me know.

||| ported | working |
| ------------- | ------------ |:----:| :-----:|
|**moveit_ros_planning**||||
||demo_scene |[X]|[X]|
||moveit_compare_collision_checking_speed_fcl_bullet|[ ]|[ ]|
||moveit_evaluate_state_operations_speed |[TODO]| [TODO]|
||moveit_print_planning_model_info |[TODO]|[TODO]|
||moveit_visualize_robot_collision_volume |[TODO]|[TODO]|
||moveit_display_random_state |[TODO]|[TODO]|
||moveit_kinematics_speed_and_validity_evaluator |[TODO]|[TODO]|
||moveit_print_planning_scene_info |[TODO]|[TODO]|
||moveit_evaluate_collision_checking_speed |[TODO]|[TODO]|
||moveit_list_request_adapter_plugins |[TODO]|[TODO]|
||moveit_publish_scene_from_text |[TODO]|[TODO]|
||test_controller_manager |[TODO]|[TODO]|

ddengster, 2020-06-18:
Here's an update on moveit_ros_move_group's test: test_cancel_before_plan_execution. 

The test is a python file with a launch xml, it depends on moveit_commander which has yet to be ported, which in turn depends on moveit_ros_planning_interface and py_bindings_tools for it's bindings. py_bindings_tools is notably struck out in the Moveit2 migration progress document, so it seems like I'll have to leave it alone for now.

nbbrooks, 2020-06-25:
There is a TODO in the codebase, but moveit_kinematics [benchmark_ik.cpp](https://github.com/ros-planning/moveit2/blob/master/moveit_kinematics/test/benchmark_ik.cpp#L83) needs its parameters adjusted so it loads successfully.

sjahr, 2020-06-28:
Thank you for your input! I added comments in the table to point that out.