moveit/moveit2 #229: [moveit_servo] Handling moveit_servo's use of control_msgs/msg/JointJog

AdamPettinger, 2020-06-30:
~~`movit_jog_arm`~~ `moveit_servo` depends on [control_msgs/msg/JointJog](https://github.com/ros-controls/control_msgs/blob/kinetic-devel/control_msgs/msg/JointJog.msg) as an input to the servo node (for direct joint control). 

I believe this message was added to `control_msgs` specifically for `moveit_jog_arm`, and there was some discussion at the time between placing it in `control_msgs` vs `moveit_msgs` ([https://github.com/ros-planning/moveit_msgs/pull/48](https://github.com/ros-planning/moveit_msgs/pull/48) and [https://github.com/ros-controls/control_msgs/pull/26](https://github.com/ros-controls/control_msgs/pull/26))

I am bringing this up now because

1. We recently moved away from the "jog" terminology in `moveit_jog_arm` in favor of "servo".
2. The `JointJog` msg is not yet ported to ROS 2 in [control_msgs](https://github.com/ros-controls/control_msgs/tree/foxy-devel/control_msgs/msg)

I don't have strong opinions on how, but the `JointJog` message (or similar) will need to be ported to ROS 2 for use with `moveit_servo`. I don't want to mess with the internals of that message type, but thought that we might want to rename it to bring it in line with the `moveit_servo` renaming? I figure there are some options moving forward:

1. Copy the `JointJog` message to the `control_msgs` `foxy-devel` branch and keep it the same as it was in ROS 1
2. Copy as above, but rename the message to `JointServo` (or something else?)
3. If we think it's only being used in `moveit_servo`, we can move it to `moveit_msgs` and potentially rename as wanted
4. Something else

I am looking for feedback before pursuing any of the options, especially from @AndyZe (original commit), and anybody else around when that message was added

AndyZe, 2020-06-30:
I think it's fine to keep the message name as **JointJog** because that has meaning in the robot world. For example:

> Joint – select this coordinate system to jog each joint individually or at the same time.

from jogging instructions for an industrial robot. (https://motioncontrolsrobotics.com/jogging-the-robot/)

AndyZe, 2020-06-30:
I vote for Option 1 to stay in line with our original discussion

henningkayser, 2020-07-02:
I agree that it should be fine to keep it as JointJog. Otherwise, we could rename it to `JointServoCommand` but only if this message is really used for moveit_servo only. Option 1 should work for now. Could you file a PR for the foxy-devel branch?

AdamPettinger, 2020-07-03:
Done and merged [https://github.com/ros-controls/control_msgs/pull/47](https://github.com/ros-controls/control_msgs/pull/47)

closed 2020-07-03 by AdamPettinger