moveit/moveit2 #234: /move_group node and launch files

AliaChe, 2020-07-15:
Hello, 

I plan to use moveit2 in order to plan trajectories for the turtlebot3 and eventually for a drone. 
In moveit1, the launch files necessary to launch the move_group and the different functionalities are generated by the setup-assistant. Since the setup-assistant is not yet ported in moveit2, is there any planned porting or any porting in progress for the move_group node and the corresponding launch files ? 

Thank you

henningkayser, 2020-07-16:
Hi @AliaChe,

Currently, MoveGroup has not been fully ported yet, this feature is WIP. We are also planning to finalize the straight migration including MSA in the near future. For now, you could try to run your setup with MoveItCpp which is actually very similar to the MoveGroupInterface. Simply check out the demo at https://github.com/ros-planning/moveit2/tree/master/moveit_demo_nodes/run_moveit_cpp. Would this work for you?

AliaChe, 2020-07-20:
Hi @henningkayser , 

Thank you for your reply. Do you have an idea about the schedule of this porting?  I'll check the demo you mentioned. Is there any other tips or hints that can help? 

Thank you

JafarAbdi, 2020-10-05:
Hi @AliaChe,

MoveGroup is available in the main branch, please see [run_move_group](https://github.com/ros-planning/moveit2/tree/main/moveit_demo_nodes/run_move_group) on how to use it, let me know if you run into issues while testing it

closed 2020-10-05 by JafarAbdi