moveit/moveit2 #235: Switch to clang-format-6 or *-10

henningkayser, 2020-07-23:
With Foxy/Noetic (Ubuntu Focal) `clang-format-3.9` is not supported anymore. In order to allow syncing between MoveIt 1 and MoveIt 2 we need to find a common clang-format version can be applied for all distros. For now, I suggest running clang-format-3.9 on Eloquent (see https://github.com/ros-planning/moveit2/pull/227/commits/7aea7a713c66b13ca9c47fe4dc5c098512a6886d) until we found a good replacement.

Related:
* Noetic Release issue - https://github.com/ros-planning/moveit/issues/2015
* Example configs - https://github.com/ros-planning/moveit/pull/2169, https://github.com/ros-planning/moveit/pull/2177

henningkayser, 2020-08-11:
https://github.com/ros-planning/moveit/pull/2177 has been merged, I will perform a new repo sync and then file a separate PR fixing clang-format-10.

JafarAbdi, 2020-10-05:
Fixed in https://github.com/ros-planning/moveit2/pull/282

closed 2020-10-05 by JafarAbdi