moveit/moveit2 #250: Compile error with updateInternalDataStructures override

ddengster, 2020-08-18:
### Description

```
In file included from /home/ddeng/moveit_ws/src/moveit2/moveit_core/constraint_samplers/test/pr2_arm_kinematics_plugin.cpp:43:
/home/ddeng/moveit_ws/src/moveit2/moveit_core/constraint_samplers/test/pr2_arm_kinematics_plugin.h:87:8: error: ‘void pr2_arm_kinematics::PR2ArmIKSolver::updateInternalDataStructures()’ marked ‘override’, but does not override
   void updateInternalDataStructures() override;
        ^~~~~~~~~~~~~~~~~~~~~~~~~~~~
make[2]: *** [constraint_samplers/CMakeFiles/test_constraint_samplers.dir/test/pr2_arm_kinematics_plugin.cpp.o] Error 1
make[2]: *** Waiting for unfinished jobs....
In file included from /home/ddeng/moveit_ws/src/moveit2/moveit_core/constraint_samplers/test/test_constraint_samplers.cpp:54:
/home/ddeng/moveit_ws/src/moveit2/moveit_core/constraint_samplers/test/pr2_arm_kinematics_plugin.h:87:8: error: ‘void pr2_arm_kinematics::PR2ArmIKSolver::updateInternalDataStructures()’ marked ‘override’, but does not override
   void updateInternalDataStructures() override;
```

Related commit:
https://github.com/ros-planning/moveit2/commit/d9e99cba339a0a0a8ef8d519d5ee08bb64bfd7bd#diff-8404d089489122c76cd63a908826c88e

### Your environment
* ROS Distro: Eloquent
* OS Version: Ubuntu 18.04
* Source build: master 9e6868a

### Steps to reproduce
`colcon build`

### Expected behaviour
Should build

### Actual behaviour
Stated in descript

henningkayser, 2020-08-18:
The debian `ros-eloquent-orocos-kdl` was released with a really out of date kdl version which was only synced this year https://github.com/ros2/orocos_kinematics_dynamics/pull/13 (changes go back to 2015). The synced release only made it to Foxy, though. As discussed over the last months, we will not support Eloquent on master anymore since there are too many breaking changes we don't want to support while the port is not completed (dependency issues like this, launch file changes, ...). For testing and development, you can use the `eloquent` branch if you want, but ideally we should all move to Foxy on `master`. If you still want to compile `master` in Eloquent, you have to pull in [orocos_kdl](https://github.com/ros2/orocos_kinematics_dynamics) from source and probably fix some other missing dependencies and launch file changes.

JafarAbdi, 2020-10-05:
This should be fixed in the main branch

closed 2020-10-05 by JafarAbdi