moveit/moveit2 #252: Duplicate nodes in run_moveit_cpp demo

AdamPettinger, 2020-08-20:
`###` Description

Running both demo launch files in `run_moveit_cpp` give duplicate nodes:
[image]

I believe this is caused by #230 as I can rebase to remove those commits and do not get duplicate nodes.

### Your environment
* ROS Distro: Foxy
* OS Version: Ubuntu 20.04
* Source build: master branch

### Steps to reproduce
[Build moveit2 from source](https://moveit.ros.org/install-moveit2/source/)
`ros2 launch run_moveit_cpp run_moveit_cpp.launch.py`
In a new terminal while that is running:
`ros2 node list`

### Expected behaviour
Rebasing to before that PR was merged gives what I would expect:
[image]

henningkayser, 2020-08-26:
@AdamPettinger are you sure this is still an issue? I thought this was fixed before the referenced PR was merged. I'm not able to reproduce this.

AdamPettinger, 2020-08-26:
> Possibly useable information: a recent fix for the same kind of problem in navigation2

@mikeferguson That worked like a charm, thanks! Shamelessly copied

closed 2020-08-27 by AdamPettinger