moveit/moveit2 #268: High CPU load

AndyZe, 2020-08-30:
### Description

The moveit_cpp and moveit_servo demos show very high CPU load.

### Your environment
* ROS Distro: Foxy
* OS Version: Ubuntu 20.04
* Source build, master branch

### Steps to reproduce
Run the moveit_servo demo:

`ros2 launch moveit_servo servo_cpp_interface_demo.launch.py`

Run `top`. Observe high CPU load for several nodes (>90%, sometimes >200%). These are the affected nodes: fake_joint_driver, servo_demo, robot_state_pub.

You should see something similar if you run `ros2 launch run_moveit_cpp run_moveit_cpp.launch.py`

### Expected behaviour
Reasonable CPU loads, probably less than 20%.

AndyZe, 2020-08-30:
Things these 3 nodes have in common:

- They all subscribe to something
- They all publish something

Things they do not have in common:

- They do not all use a MultiThreadedExecutor

AndyZe, 2020-08-30:
Commenting this line:

`executor->spin();`

And replacing with a loop with sleep drops CPU load by 80%:

```
  while (std::cin.get() != '\n')
  {
    loop_rate.sleep();
  }
```

AndyZe, 2020-08-31:
Related:  https://github.com/ros-planning/navigation2/pull/866

AdamPettinger, 2020-08-31:
#270 Does drop the CPU load for me

One thing I have noticed is that the 3 nodes you mentioned (`fake_joint_driver`, `servo_demo`, `robot_state_pub`) all publish or subscribe to `/joint_states`

On both master and with your PR, the `moveit_servo` and `run_moveit_cpp` demos both publish `/joint_states` incredibly fast: 2700 - 3500 hz. I wonder if that is why those nodes are the ones with high load? No other nodes deal with `/joint_states`

AndyZe, 2020-08-31:
It sounds like we are close! I bet the issue is in fake_joint_driver

AndyZe, 2020-08-31:
Making this change in fake_joint_driver.cpp is one way to do it. The CPU load is great after this:

```
void FakeJointDriver::update(void)
{
  // only do loopback
  act_dis = cmd_dis;
}
```

```
void FakeJointDriver::update(void)
{
  rclcpp::sleep_for(std::chrono::milliseconds(100));
  // only do loopback
  act_dis = cmd_dis;
}
```

AdamPettinger, 2020-08-31:
Tested adding that line, (with 10 ms sleep for 100 hz) and `top` showed 45% for Rviz, 63% for servo_node, and everything else under 5%. Pretty sure that will do it

AndyZe, 2020-09-01:
Made a little PR to fix this on Jafar's branch here:

`https://github.com/JafarAbdi/fake_joint/pull/3`

I think the root case was integer division in C++ causing a spin rate of infinity.  :0

closed 2020-09-01 by AndyZe