moveit/moveit2 #274: Missing "random_numbers" Package on building from source on 20.04 LTS

doitmaan, 2020-09-08:
### Description

Failure to Build Moveit2 on ubuntu 20.04 focal Due to the inability of finding the **random_numbers** package as a dependency for **geometric_shapes**

### Your environment
* ROS Distro: [foxy]
* OS Version: e.g. Ubuntu 20.04.1
* Building the Source | Pulling from master 

### Steps to reproduce

building on 

> VERSION="20.04.1 LTS (Focal Fossa)"
ID=ubuntu
ID_LIKE=debian
PRETTY_NAME="Ubuntu 20.04.1 LTS"
VERSION_ID="20.04"

Using:

`
colcon build --event-handlers desktop_notification- status- --cmake-args -DCMAKE_BUILD_TYPE=Release`

### Expected behaviour

Ability to pull all dependencies from the .repo file

### Backtrace or Console output

> Starting >>> urdfdom_headers
> Finished <<< urdfdom_headers [0.15s]
> Starting >>> geometric_shapes
> --- stderr: geometric_shapes
> CMake Error at CMakeLists.txt:42 (find_package):
>   By not providing "Findrandom_numbers.cmake" in CMAKE_MODULE_PATH this
>   project has asked CMake to find a package configuration file provided by
>   "random_numbers", but CMake did not find one.
> 
>   Could not find a package configuration file provided by "random_numbers"
>   with any of the following names:
> 
>     random_numbersConfig.cmake
>     random_numbers-config.cmake
> 
>   Add the installation prefix of "random_numbers" to CMAKE_PREFIX_PATH or set
>   "random_numbers_DIR" to a directory containing one of the above files.  If
>   "random_numbers" provides a separate development package or SDK, be sure it
>   has been installed.
> 
> 
> ---
> Failed   <<< geometric_shapes [0.59s, exited with code 1]
> 
> Summary: 1 package finished [0.97s]
>   1 package failed: geometric_shapes
>   1 package had stderr output: geometric_shapes
>   35 packages not processed

Use [gist.github.com](gist.github.com) to copy-paste the console output or segfault backtrace using gdb.

doitmaan, 2020-09-08:
### similar behavior but a different missing package:

[https://github.com/ros-planning/moveit2/issues/156#issuecomment-688543800](url)

possible package support for the above can be:

[https://github.com/ros-planning/moveit2/pull/157#issuecomment-688568989](url)

mikeferguson, 2020-09-08:
moveit_core has a dependency on random_numbers - I think you missed the rosdep install step of the installation instructions:

```
rosdep install -r --from-paths . --ignore-src --rosdistro foxy -y
```

(it's right after VCS import in the source install instructions: https://moveit.ros.org/install-moveit2/source/)

doitmaan, 2020-09-08:
Hi, 
Thank you very much for the reply but

On

`rosdep install -r --from-paths . --ignore-src --rosdistro foxy -y`

Resulted in 

`#All required rosdeps installed successfully`

I made sure to follow the steps but added it to the .repos file 

PR: [https://github.com/ros-planning/moveit2/pull/275#issue-481749827](url)

Kindly advice if I missing something

henningkayser, 2020-09-08:
Both `moveit_core` and `geometric_shapes` list `random_numbers` in their `package.xml`. Could you check if `ros-foxy-random-numbers` is installed? This seems like an issue with rosdep on your system, adding it to the .repos file should not be necessary.

JafarAbdi, 2020-09-08:
Make sure to source the `setup.bash` file by running `source /opt/ros/foxy/setup.bash
` or simply add it to ` ~/.bashrc`, otherwise CMake will not be able to find the dependencies

doitmaan, 2020-09-08:
> Both `moveit_core` and `geometric_shapes` list `random_numbers` in their `package.xml`. Could you check if `ros-foxy-random-numbers` is installed? This seems like an issue with rosdep on your system, adding it to the .repos file should not be necessary.

Thank you for your reply, I was able to find 

> ros-foxy-random-numbers-dbgsym/focal 1.0.0-1focal.20200714.220700 amd64
ros-foxy-random-numbers/focal,now 1.0.0-1focal.20200714.220700 amd64 
[installed]

both of the above packages are installed, I believe they have been installed correctly

doitmaan, 2020-09-08:
> Make sure to source the `setup.bash` file by running `source /opt/ros/foxy/setup.bash ` or simply add it to ` ~/.bashrc`, otherwise CMake will not be able to find the dependencies

Thank you very much, I accidentally  forgot to source the environment

closed 2020-09-08 by doitmaan