moveit/moveit2 #280: Example fails

khssnv, 2020-09-15:
### Description

I have MoveIt2 installed by [guide](https://moveit.ros.org/install-moveit2/source/) on fresh Ubuntu 20.04 with ROS2 Foxy. An attempt to launch the [demo](https://github.com/ros-planning/moveit2/tree/main/moveit_demo_nodes/run_moveit_cpp) fails with output you may find in a gist by the link below.

### Your environment
* ROS Distro: ROS2 Foxy
* OS Version: Ubuntu 20.04.1 LTS 
* Source build
* Branch main at d95e4aecf

### Steps to reproduce
1. Get fresh Ubuntu 20.04.1 LTS
2. Install ROS2 Foxy with build tools
3. Build moveit2 by the [guide](https://moveit.ros.org/install-moveit2/source/)
4. Launch the [demo](https://github.com/ros-planning/moveit2/tree/main/moveit_demo_nodes/run_moveit_cpp)

### Expected behaviour
Demo works as described [here](https://github.com/ros-planning/moveit2/tree/main/moveit_demo_nodes/run_moveit_cpp#running-the-demo)

### Actual behaviour
Demo run fails with error
```console
[run_moveit_cpp-4] [ERROR] [1600161898.346785397] [moveit.ros_planning_interface.planning_component]: Could not compute plan successfully
```
### Backtrace or Console output

[out.md](https://gist.github.com/khssnv/834ec27d710c41f787d44f104d054f13)

henningkayser, 2020-09-15:
@khssnv thanks for reporting this. The log shows that the current state used as start state is in self-collision which means that the joint state wasn't updated in time so it uses the self-colliding zero state. A couple lines above you can see that the fake controller fails to initialize. Unfortunately, I wasn't able to to reproduce this yet, but I wouldn't rule out that there were some breaking changes in `ros2_control`. Does this happen every time you run this demo? What is the behavior for `run_moveit_cpp_fake_controller.launch.py` or the `moveit_servo` and `move_group` demos? If possible, please post your output of `vcs validate < moveit2/moveit2.repos` from your source directory and check if there are any packages out of sync.

@JafarAbdi any insights?

JafarAbdi, 2020-10-05:
@khssnv I can't reproduce this, could you please share the information that Henning asked you to provide .?