moveit/moveit2 #283: Source build fails

ShotaAk, 2020-09-17:
### Description

The `colcon build` command of [MoveIt 2 Source Build - Linux](https://moveit.ros.org/install-moveit2/source/) fails with log message  `hardware_interface/joint_command_handle.hpp: No such file or directory`.

```sh
In file included from /home/shotaak/moveit_ws/src/fake_joint/fake_joint_driver/src/fake_joint_driver_node.cpp:10:
/home/shotaak/moveit_ws/src/fake_joint/fake_joint_driver/include/fake_joint_driver/fake_joint_driver.h:12:10: fatal error: hardware_interface/joint_command_handle.hpp: No such file or directory
   12 | #include <hardware_interface/joint_command_handle.hpp>
      |          ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
compilation terminated.
```

### Your environment
* ROS Distro: Foxy
* OS Version: Ubuntu 20.04
* Source build
* If source, which branch? ddf585cf5a9fe303da5900a8c49c1d39b40d3398

### Steps to reproduce
Reproduce procedure is same as the [MoveIt 2 Source Build guide](https://moveit.ros.org/install-moveit2/source/).

### Expected behaviour
The build command results success.

### Actual behaviour
The build has failed.

```sh
In file included from /home/shotaak/moveit_ws/src/fake_joint/fake_joint_driver/src/fake_joint_driver_node.cpp:10:
/home/shotaak/moveit_ws/src/fake_joint/fake_joint_driver/include/fake_joint_driver/fake_joint_driver.h:12:10: fatal error: hardware_interface/joint_command_handle.hpp: No such file or directory
   12 | #include <hardware_interface/joint_command_handle.hpp>
      |          ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
compilation terminated.
```

### Additional context

The file `joint_command_handle.hpp` has removed by https://github.com/ros-controls/ros2_control/pull/134.

This causes above fake_joint_driver's build error.

yongxf, 2020-09-17:
I went through the same issue. joint_command_handle.hpp cannot be found in moveit2.

henningkayser, 2020-09-17:
@ShotaAk Thanks for reporting, we've already realized that `ros2_control` had some breaking changes lately. Until then, it probably makes most sense to fix the commit hash in the repos file and bump the version from time to time until there is a stable API. Were you able to run the demo successfully with an older version?

ShotaAk, 2020-09-17:
Yes. I had checkout `ros2_control` to [2c3aad1cb8ad4d23925bceeb5c26e22d6cf859ef](https://github.com/ros-controls/ros2_control/tree/2c3aad1cb8ad4d23925bceeb5c26e22d6cf859ef) and `ros2_controllers` to [d52902538a970e792ecc604416e22ef90a57f29e](https://github.com/ros-controls/ros2_controllers/tree/d52902538a970e792ecc604416e22ef90a57f29e), then the demo successfully worked.

destogl, 2020-09-20:
@henningkayser @ShotaAk we are braking a lot currently in `ros2_control` in a month or so we should fix new interfaces. If you could use the versions from the [comment](https://github.com/ros-planning/moveit2/issues/283#issuecomment-694193363) it would be great :)

henningkayser, 2020-09-21:
I've created #284 for freezing the `ros2_control` commits in the workspace. @ShotaAk could you rename this issue to something like "Use new ros2_control interfaces" so that we can continue tracking it in here?

ShotaAk, 2020-09-21:
@henningkayser Sure. :+1:

ShotaAk, 2020-09-25:
@henningkayser @JafarAbdi I have replaced old joint handles of `fake_joint_driver` to new joint handles to solve this issue.
(See https://github.com/ShotaAk/fake_joint/commit/e888e4988563c2e671f4239f2fe9de0938d2e0bf)

May I create a pull request to `JafarAbdi/fake_joint`?

(Or should we wait and see the ros2_control proggress?)

henningkayser, 2020-09-25:
@ShotaAk That's great, thanks! In general, I think it's preferable to stay up to date, but I'd like to first make sure that the API is somewhat stable. The PR would still be very helpful, even if we don't merge it right away.

ShotaAk, 2020-09-25:
> I'd like to first make sure that the API is somewhat stable
I understand. Anyway, I will create a PR.

bmagyar, 2020-09-26:
Hey guys, the current version is functional to a limited extent but the upcoming changes may break API/ABI multiple times, hence the recommendation to stick to a fixed version for the time being. I'll make sure to update you when we reach a stable version.

ShotaAk, 2020-09-27:
@bmagyar Thanks. It looks like there will be more updates than just joint handle.

(I should calm down and see the progress.）

destogl, 2020-12-12:
Hey people,

we have now fixed the API for Foxy and you can start to handle this. If you need any help just ping me or @bmagyar

ShotaAk, 2020-12-17:
@destogl Thank you for your information.

I'm trying to use new ros2_control API in our robot (CRANE+V2, https://github.com/rt-net/crane_plus ).
Is there any migration guide for new API?
(It seem the `joint_command_handle`, `joint_state_handle` and `robot_hardware` in `hardware_interface` are deleted.)

destogl, 2021-01-05:
@ShotaAk I just wrote short documentation in ros-controls/roadmap#36

It would be great if you check it and try to use it. We are very happy to get any feedback on it. Especially hot to make it better.

ShotaAk, 2021-01-06:
@destogl That's great documentation!
The contents of `Differences to ros_control (ROS1)` and `Migration Guide to ros2_control` are exactly what I wanted to read.
Thank you!

closed 2021-03-25 by henningkayser