moveit/moveit2 #287: Received trajectory with non zero time start time (1601439816.637827) that ends on the past (1601439816.637827)

sachinkum0009, 2020-09-30:
### Description

Overview of your issue here.

### Your environment
* ROS Distro: [Foxy]
* OS Version: e.g. Ubuntu 20.04
* Binary build
* If binary, 
```bash
git clone https://github.com/ros-planning/moveit2.git -b main
vcs import < moveit2/moveit2.repos
rosdep install -r --from-paths . --ignore-src --rosdistro foxy -y
```

### Steps to reproduce
```bash
ros2 launch moveit_servo servo_cpp_interface_demo.launch.py
```
### Expected behaviour

Robot is not moving

### Actual behaviour
Robot should move as showed in running_demo readme
https://github.com/ros-planning/moveit2/blob/main/moveit_ros/moveit_servo/doc/running_the_demos.md?s=09

### Backtrace or Console output
```bash
[INFO] [launch]: Default logging verbosity is set to INFO
[INFO] [rviz2-1]: process started with pid [41803]
[INFO] [static_transform_publisher-2]: process started with pid [41805]
[INFO] [servo_demo-3]: process started with pid [41807]
[INFO] [fake_joint_driver_node-4]: process started with pid [41809]
[INFO] [robot_state_publisher-5]: process started with pid [41811]
[fake_joint_driver_node-4] Parsing robot urdf xml string.
[static_transform_publisher-2] [INFO] [1601439784.058746096] [static_transform_publisher]: Spinning until killed publishing transform from 'world' to 'panda_link0'
[robot_state_publisher-5] Parsing robot urdf xml string.
[robot_state_publisher-5] Link panda_link1 had 1 children
[robot_state_publisher-5] Link panda_link2 had 1 children
[robot_state_publisher-5] Link panda_link3 had 1 children
[robot_state_publisher-5] Link panda_link4 had 1 children
[robot_state_publisher-5] Link panda_link5 had 1 children
[robot_state_publisher-5] Link panda_link6 had 1 children
[robot_state_publisher-5] Link panda_link7 had 1 children
[robot_state_publisher-5] Link panda_link8 had 1 children
[robot_state_publisher-5] Link panda_hand had 2 children
[robot_state_publisher-5] Link panda_leftfinger had 0 children
[robot_state_publisher-5] Link panda_rightfinger had 0 children
[robot_state_publisher-5] [INFO] [1601439784.118275663] [robot_state_publisher]: got segment panda_hand
[robot_state_publisher-5] [INFO] [1601439784.118833171] [robot_state_publisher]: got segment panda_leftfinger
[robot_state_publisher-5] [INFO] [1601439784.118896429] [robot_state_publisher]: got segment panda_link0
[robot_state_publisher-5] [INFO] [1601439784.118924240] [robot_state_publisher]: got segment panda_link1
[robot_state_publisher-5] [INFO] [1601439784.118946591] [robot_state_publisher]: got segment panda_link2
[robot_state_publisher-5] [INFO] [1601439784.118968256] [robot_state_publisher]: got segment panda_link3
[robot_state_publisher-5] [INFO] [1601439784.118989938] [robot_state_publisher]: got segment panda_link4
[robot_state_publisher-5] [INFO] [1601439784.119011251] [robot_state_publisher]: got segment panda_link5
[robot_state_publisher-5] [INFO] [1601439784.119032416] [robot_state_publisher]: got segment panda_link6
[robot_state_publisher-5] [INFO] [1601439784.119053959] [robot_state_publisher]: got segment panda_link7
[robot_state_publisher-5] [INFO] [1601439784.119075635] [robot_state_publisher]: got segment panda_link8
[robot_state_publisher-5] [INFO] [1601439784.119097532] [robot_state_publisher]: got segment panda_rightfinger
[fake_joint_driver_node-4] [INFO] [1601439784.276648564] [controller_manager]: Loading controller 'fake_joint_state_controller'
[fake_joint_driver_node-4] 
[fake_joint_driver_node-4] [INFO] [1601439784.495835246] [controller_manager]: Loading controller 'fake_joint_trajectory_controller'
[fake_joint_driver_node-4] 
[fake_joint_driver_node-4] [INFO] [1601439784.834661037] [fake_joint_trajectory_controller]: Controller state will be published at 50Hz.
[fake_joint_driver_node-4] [INFO] [1601439784.837846828] [fake_joint_trajectory_controller]: Action status changes will be monitored at 20Hz.
[fake_joint_driver_node-4] [INFO] [1601439784.893462479] [fake_joint_driver]: Successfully configured all controllers
[fake_joint_driver_node-4] [INFO] [1601439784.893674200] [fake_joint_driver]: Successfully activated all controllers
[rviz2-1] [INFO] [1601439787.259165522] [rviz2]: Stereo is NOT SUPPORTED
[... 44 lines cut ...]
[fake_joint_driver_node-4] [ERROR] [1601439790.630016890] [fake_joint_trajectory_controller]: Received trajectory with non zero time start time (1601439790.629841) that ends on the past (1601439790.629841)
[fake_joint_driver_node-4] [ERROR] [1601439790.664044523] [fake_joint_trajectory_controller]: Received trajectory with non zero time start time (1601439790.663892) that ends on the past (1601439790.663892)
[fake_joint_driver_node-4] [ERROR] [1601439790.697946243] [fake_joint_trajectory_controller]: Received trajectory with non zero time start time (1601439790.697820) that ends on the past (1601439790.697820)
[fake_joint_driver_node-4] [ERROR] [1601439790.731972108] [fake_joint_trajectory_controller]: Received trajectory with non zero time start time (1601439790.731843) that ends on the past (1601439790.731843)
[fake_joint_driver_node-4] [ERROR] [1601439790.765946214] [fake_joint_trajectory_controller]: Received trajectory with non zero time start time (1601439790.765839) that ends on the past (1601439790.765839)
[fake_joint_driver_node-4] [ERROR] [1601439790.799933074] [fake_joint_trajectory_controller]: Received trajectory with non zero time start time (1601439790.799822) that ends on the past (1601439790.799822)
[fake_joint_driver_node-4] [ERROR] [1601439790.841919394] [fake_joint_trajectory_controller]: Received trajectory with non zero time start time (1601439790.841805) that ends on the past (1601439790.841805)
[fake_joint_driver_node-4] [ERROR] [1601439790.867977104] [fake_joint_trajectory_controller]: Received trajectory with non zero time start time (1601439790.867840) that ends on the past (1601439790.867840)
[fake_joint_driver_node-4] [ERROR] [1601439790.902102221] [fake_joint_trajectory_controller]: Received trajectory with non zero time start time (1601439790.901853) that ends on the past (1601439790.901853)
[fake_joint_driver_node-4] [ERROR] [1601439790.936594193] [fake_joint_trajectory_controller]: Received trajectory with non zero time start time (1601439790.936119) that ends on the past (1601439790.936119)
[fake_joint_driver_node-4] [ERROR] [1601439790.970274027] [fake_joint_trajectory_controller]: Received trajectory with non zero time start time (1601439790.969956) that ends on the past (1601439790.969956)
[fake_joint_driver_node-4] [ERROR] [1601439791.003985435] [fake_joint_trajectory_controller]: Received trajectory with non zero time start time (1601439791.003832) that ends on the past (1601439791.003832)
[fake_joint_driver_node-4] [ERROR] [1601439791.037934439] [fake_joint_trajectory_controller]: Received trajectory with non zero time start time (1601439791.037820) that ends on the past (1601439791.037820)
[fake_joint_driver_node-4] [ERROR] [1601439791.072354688] [fake_joint_trajectory_controller]: Received trajectory with non zero time start time (1601439791.071962) that ends on the past (1601439791.071962)
[fake_joint_driver_node-4] [ERROR] [1601439791.105971157] [fake_joint_trajectory_controller]: Received trajectory with non zero time start time (1601439791.105794) that ends on the past (1601439791.105794)
[fake_joint_driver_node-4] [ERROR] [1601439791.140164245] [fake_joint_trajectory_controller]: Received trajectory with non zero time start time (1601439791.139943) that ends on the past (1601439791.139943)
[fake_joint_driver_node-4] [ERROR] [1601439791.174565239] [fake_joint_trajectory_controller]: Received trajectory with non zero time start time (1601439791.174112) that ends on the past (1601439791.174112)
[fake_joint_driver_node-4] [ERROR] [1601439791.221048629] [fake_joint_trajectory_controller]: Received trajectory with non zero time start time (1601439791.220843) that ends on the past (1601439791.220843)
[fake_joint_driver_node-4] [ERROR] [1601439791.242457924] [fake_joint_trajectory_controller]: Received trajectory with non zero time start time (1601439791.241978) that ends on the past (1601439791.241978)
[fake_joint_driver_node-4] [ERROR] [1601439791.276656136] [fake_joint_trajectory_controller]: Received trajectory with non zero time start time (1601439791.276062) that ends on the past (1601439791.276062)
[fake_joint_driver_node-4] [ERROR] [1601439791.329361830] [fake_joint_trajectory_controller]: Received trajectory with non zero time start time (1601439791.329035) that ends on the past (1601439791.329 [... cut ...]
[fake_joint_driver_node-4] [ERROR] [1601439791.344549165] [fake_joint_trajectory_controller]: Received trajectory with non zero time start time (1601439791.344282) that ends on the past (1601439791.344282)
[fake_joint_driver_node-4] [ERROR] [1601439791.388239407] [fake_joint_trajectory_controller]: Received trajectory with non zero time start time (1601439791.387997) that ends on the past (1601439791.387997)
[fake_joint_driver_node-4] [ERROR] [1601439791.412059746] [fake_joint_trajectory_controller]: Received trajectory with non zero time start time (1601439791.411829) that ends on the past (1601439791.411829)
```

henningkayser, 2020-10-01:
@sachinkum0009 This looks like yet another breaking change in `ros2_controllers` that we didn't consider. Could you try checking out https://github.com/ros-controls/ros2_controllers/commit/e3509e13196e8a35fb7d9391708d6e277a1f3204 and report if this fixes your problem?

sachinkum0009, 2020-10-03:
@henningkayser Thank you. It fixed the above issue.

closed 2020-10-03 by sachinkum0009