moveit/moveit2 #295: boost system not included?

brookdpg, 2020-10-15:
Description

Overview of your issue here.

 Your environment
* ROS Distro: [foxy ]
* OS Version: e.g. Ubuntu 20.04
* Source or Binary build? source
HEAD detached at d95e4aecf

### Steps to reproduce
initial compiling of ws_moveit directory 

 Expected behaviour
all packages finish processing successfully

 Actual behaviour
build fails

Backtrace or Console output
[error.txt](https://github.com/ros-planning/moveit2/files/5381591/error.txt)

JafarAbdi, 2020-10-15:
Looking into the CMakeLists of the package it seems like it's linked correctly against Boost, can you check if the shared library exists on your system by running `find /usr/lib -name libboost_system.so*` or `find /usr/lib -name libboost_*.so*`

brookdpg, 2020-10-17:
running the search command gives me 

ops@ops-XPS-15-9500:~/ws_ros2$ sudo find /usr/lib -name libboost_*
/usr/lib/x86_64-linux-gnu/libboost_numpy38.so
/usr/lib/x86_64-linux-gnu/libboost_date_time.so
/usr/lib/x86_64-linux-gnu/libboost_filesystem.so.1.71.0
/usr/lib/x86_64-linux-gnu/libboost_stacktrace_addr2line.a
/usr/lib/x86_64-linux-gnu/libboost_chrono.a
/usr/lib/x86_64-linux-gnu/libboost_chrono.so.1.71.0
/usr/lib/x86_64-linux-gnu/libboost_stacktrace_noop.a
/usr/lib/x86_64-linux-gnu/libboost_thread.so
/usr/lib/x86_64-linux-gnu/libboost_random.so.1.71.0
/usr/lib/x86_64-linux-gnu/libboost_type_erasure.a
/usr/lib/x86_64-linux-gnu/libboost_math_tr1.so
/usr/lib/x86_64-linux-gnu/libboost_filesystem.a
/usr/lib/x86_64-linux-gnu/libboost_container.so
/usr/lib/x86_64-linux-gnu/libboost_math_c99l.a
/usr/lib/x86_64-linux-gnu/libboost_stacktrace_backtrace.so.1.71.0
/usr/lib/x86_64-linux-gnu/libboost_stacktrace_noop.so.1.71.0
/usr/lib/x86_64-linux-gnu/libboost_math_c99.so.1.71.0
/usr/lib/x86_64-linux-gnu/libboost_stacktrace_noop.so
/usr/lib/x86_64-linux-gnu/libboost_mpi.so.1.71.0
/usr/lib/x86_64-linux-gnu/libboost_graph_parallel.so.1.71.0
/usr/lib/x86_64-linux-gnu/libboost_math_c99f.a
/usr/lib/x86_64-linux-gnu/libboost_math_tr1l.so.1.71.0
/usr/lib/x86_64-linux-gnu/libboost_mpi_python38.so.1.71.0
/usr/lib/x86_64-linux-gnu/libboost_atomic.so.1.71.0
/usr/lib/x86_64-linux-gnu/libboost_graph.a
/usr/lib/x86_64-linux-gnu/cmake/boost_math_c99-1.71.0/libboost_math_c99-variant-static.cmake
/usr/lib/x86_64-linux-gnu/cmake/boost_math_c99-1.71.0/libboost_math_c99-variant-shared.cmake
/usr/lib/x86_64-linux-gnu/cmake/boost_coroutine-1.71.0/libboost_coroutine-variant-static.cmake
/usr/lib/x86_64-linux-gnu/cmake/boost_coroutine-1.71.0/libboost_coroutine-variant-shared.cmake
/usr/lib/x86_64-linux-gnu/cmake/boost_numpy-1.71.0/libboost_numpy-variant-shared-py3.8.cmake
/usr/lib/x86_64-linux-gnu/cmake/boost_numpy-1.71.0/libboost_numpy-variant-static-py3.8.cmake
/usr/lib/x86_64-linux-gnu/cmake/boost_unit_test_framework-1.71.0/libboost_unit_test_framework-variant-static.cmake
/usr/lib/x86_64-linux-gnu/cmake/boost_unit_test_framework-1.71.0/libboost_unit_test_framework-variant-shared.cmake
/usr/lib/x86_64-linux-gnu/cmake/boost_python-1.71.0/libboost_python-variant-shared-py3.8.cmake
/usr/lib/x86_64-linux-gnu/cmake/boost_python-1.71.0/libboost_python-variant-static-py3.8.cmake
/usr/lib/x86_64-linux-gnu/cmake/boost_thread-1.71.0/libboost_thread-variant-static.cmake
/usr/lib/x86_64-linux-gnu/cmake/boost_thread-1.71.0/libboost_thread-variant-shared.cmake
/usr/lib/x86_64-linux-gnu/cmake/boost_mpi-1.71.0/libboost_mpi-variant-static.cmake
/usr/lib/x86_64-linux-gnu/cmake/boost_mpi-1.71.0/libboost_mpi-variant-shared.cmake
/usr/lib/x86_64-linux-gnu/cmake/boost_locale-1.71.0/libboost_locale-variant-shared.cmake
/usr/lib/x86_64-linux-gnu/cmake/boost_locale-1.71.0/libboost_locale-variant-static.cmake
/usr/lib/x86_64-linux-gnu/cmake/boost_atomic-1.71.0/libboost_atomic-variant-shared.cmake
/usr/lib/x86_64-linux-gnu/cmake/boost_atomic-1.71.0/libboost_atomic-variant-static.cmake
/usr/lib/x86_64-linux-gnu/cmake/boost_context-1.71.0/libboost_context-variant-static.cmake
/usr/lib/x86_64-linux-gnu/cmake/boost_context-1.71.0/libboost_context-variant-shared.cmake
/usr/lib/x86_64-linux-gnu/cmake/boost_prg_exec_monitor-1.71.0/libboost_prg_exec_monitor-variant-static.cmake
/usr/lib/x86_64-linux-gnu/cmake/boost_prg_exec_monitor-1.71.0/libboost_prg_exec_monitor-variant-shared.cmake
/usr/lib/x86_64-linux-gnu/cmake/boost_math_tr1f-1.71.0/libboost_math_tr1f-variant-static.cmake
/usr/lib/x86_64-linux-gnu/cmake/boost_math_tr1f-1.71.0/libboost_math_tr1f-variant-shared.cmake
/usr/lib/x86_64-linux-gnu/cmake/boost_graph_parallel-1.71.0/libboost_graph_parallel-variant-static.cmake
/usr/lib/x86_64-linux-gnu/cmake/boost_graph_parallel-1.71.0/libboost_graph_parallel-variant-shared.cmake
/usr/lib/x86_64-linux-gnu/cmake/boost_wave-1.71.0/libboost_wave-variant-static.cmake
/usr/lib/x86_64-linux-gnu/cmake/boost_wave-1.71.0/libboost_wave-variant-shared.cmake
/usr/lib/x86_64-linux-gnu/cmake/boost_log_setup-1.71.0/libboost_log_setup-variant-shared.cmake
/usr/lib/x86_64-linux-gnu/cmake/boost_log_setup-1.71.0/libboost_log_setup-variant-static.cmake
/usr/lib/x86_64-linux-gnu/cmake/boost_stacktrace_noop-1.71.0/libboost_stacktrace_noop-variant-static.cmake
/usr/lib/x86_64-linux-gnu/cmake/boost_stacktrace_noop-1.71.0/libboost_stacktrace_noop-variant-shared.cmake
/usr/lib/x86_64-linux-gnu/cmake/boost_filesystem-1.71.0/libboost_filesystem-variant-static.cmake
/usr/lib/x86_64-linux-gnu/cmake/boost_filesystem-1.71.0/libboost_filesystem-variant-shared.cmake
/usr/lib/x86_64-linux-gnu/cmake/boost_timer-1.71.0/libboost_timer-variant-static.cmake
/usr/lib/x86_64-linux-gnu/cmake/boost_timer-1.71.0/libboost_timer-variant-shared.cmake
/usr/lib/x86_64-linux-gnu/cmake/boost_fiber-1.71.0/libboost_fiber-variant-shared.cmake
/usr/lib/x86_64-linux-gnu/cmake/boost_fiber-1.71.0/libboost_fiber-variant-static.cmake
/usr/lib/x86_64-linux-gnu/cmake/boost_type_erasure-1.71.0/libboost_type_erasure-variant-shared.cmake
/usr/lib/x86_64-linux-gnu/cmake/boost_type_erasure-1.71.0/libboost_type_erasure-variant-static.cmake
/usr/lib/x86_64-linux-gnu/cmake/boost_math_c99f-1.71.0/libboost_math_c99f-variant-static.cmake
/usr/lib/x86_64-linux-gnu/cmake/boost_math_c99f-1.71.0/libboost_math_c99f-variant-shared.cmake
[... 113 lines cut ...]
/usr/lib/x86_64-linux-gnu/libboost_regex.a
/usr/lib/x86_64-linux-gnu/libboost_system.so
[... cut ...]
/usr/lib/x86_64-linux-gnu/libboost_wserialization.so.1.71.0
/usr/lib/x86_64-linux-gnu/libboost_random.so
/usr/lib/x86_64-linux-gnu/libboost_regex.so
/usr/lib/x86_64-linux-gnu/libboost_thread.a
/usr/lib/x86_64-linux-gnu/libboost_program_options.so
/usr/lib/x86_64-linux-gnu/libboost_type_erasure.so
/usr/lib/x86_64-linux-gnu/libboost_locale.a
/usr/lib/x86_64-linux-gnu/libboost_log.so.1.71.0
/usr/lib/x86_64-linux-gnu/libboost_fiber.so.1.71.0
/usr/lib/x86_64-linux-gnu/libboost_math_tr1f.so.1.71.0
ops@ops-XPS-15-9500:~/ws_ros2$

JafarAbdi, 2020-10-19:
Both `Boost::filesystem` and `Boost::system` do exist, I'm not sure why the linker can't find the symbols, have you test the latest commit of the main branch .?

JafarAbdi, 2021-04-15:
This should be fixed in the latest main branch

closed 2021-04-15 by JafarAbdi