moveit/moveit2 #296: NO PLANNING LIBRARY LOADED in moveit 2 for this example

yangli-flexiv, 2020-10-19:
### Description

Overview of your issue here.

### Your environment
* ROS Distro: [Foxy]
* OS Version: e.g. Ubuntu 20.04
* Binary build for installing ROS 2, and source build for moveit 2
* For ROS 2 (https://index.ros.org/doc/ros2/Installation/Foxy/Linux-Development-Setup/)
* For moveit 2 (https://moveit.ros.org/install-moveit2/source/)

### Steps to reproduce
Run the command to start the moveit 2 example:

ros2 launch run_moveit_cpp run_moveit_cpp.launch.py

### Expected behaviour
I should be able to load the OMPL planner, and do motion planning on this example.

### Actual behaviour

In the Context section of MotionPlanning, it reports that "NO PLANNING LIBRARY LOADED" in red color. I can't do any operation on this provided example.

[image]

### Backtrace or Console output

Use [gist.github.com](gist.github.com) to copy-paste the console output or segfault backtrace using gdb.

jediofgever, 2020-10-22:
I also encounter this. But what even more interesting is how the robot is able to plan and execute in;
https://github.com/ros-planning/moveit2/blob/7dba4db9934d323953e10c6de5e33dad6b721646/moveit_demo_nodes/run_moveit_cpp/src/run_moveit_cpp.cpp#L58
here.

henningkayser, 2020-10-22:
@yangli-flexiv The motion planning panel can only be used with MoveGroup and not with MoveItCpp since that one doesn't provide the required capabilities. Please run the [MoveGroup demo](https://github.com/ros-planning/moveit2/tree/main/moveit_demo_nodes/run_move_group) if you need this feature

    ros2 launch run_move_group run_move_group.launch.py

 or check out the MoveGroupInterface example

    ros2 launch run_move_group run_move_group_interface.launch.py

I don't consider this a bug since it's apparently a misconception of these components. Maybe we should add a note to the MoveItCpp demo and reference MoveGroup to clarify this, though. Please close this issue if you consider it resolved.

ErenYeager0, 2021-01-19:
I encouter this problem,too. At the same time, there is a ERROR in the terminal:
`[ERROR] [run_moveit_cpp-4]: process has died [pid 3738, exit code -6, cmd '/home/siasun/ws_ros2/install/run_moveit_cpp/lib/run_moveit_cpp/run_moveit_cpp --ros-args -r __node:=run_moveit_cpp --params-file /home/siasun/ws_ros2/install/run_moveit_cpp/share/run_moveit_cpp/config/moveit_cpp.yaml --params-file /tmp/launch_params_hce4ecya --params-file /tmp/launch_params_rywkro66 --params-file /tmp/launch_params_tk9w4j85 --params-file /tmp/launch_params_qkj7c61t --params-file /tmp/launch_params_jkcsv0or'].`
How can I solve it? @henningkayser

JafarAbdi, 2021-09-28:
Should be fixed with https://github.com/ros-planning/moveit2/pull/461

closed 2021-09-28 by JafarAbdi