moveit/moveit2 #297: Error while installing Moveit2 from Source for Linux

kev278, 2020-10-22:
Followed the instructions o this page:
https://moveit.ros.org/install-moveit2/source/

1) In the Build Moveit section
cd $COLCON_WS
colcon build --event-handlers desktop_notification- status- --cmake-args -DCMAKE_BUILD_TYPE=Release

The above commands have to be executed.

2) After executing the command:
[image]

keval@keval-Nitro-AN515-42:~/ws_ros2$ colcon build --event-handlers desktop_notification- status- --cmake-args -DCMAKE_BUILD_TYPE=Release
Starting >>> geometric_shapes
Starting >>> moveit_msgs
Starting >>> srdfdom
Starting >>> moveit_common
Starting >>> hardware_interface
Starting >>> moveit_resources_panda_description
Starting >>> moveit_resources_fanuc_description
Starting >>> moveit_resources_pr2_description
Finished <<< moveit_common [6.42s]
Starting >>> warehouse_ros
Finished <<< moveit_resources_fanuc_description [6.40s]
Finished <<< moveit_resources_panda_description [6.41s]
Starting >>> moveit_resources_fanuc_moveit_config
Starting >>> moveit_resources_panda_moveit_config
Finished <<< moveit_resources_pr2_description [6.45s]
Finished <<< moveit_resources_fanuc_moveit_config [1.59s]
Finished <<< moveit_resources_panda_moveit_config [1.61s]
Starting >>> moveit_resources
Finished <<< moveit_resources [1.42s]
--- stderr: hardware_interface
In file included from /home/keval/ws_ros2/src/ros2_control/hardware_interface/src/robot_hardware.cpp:15:
/home/keval/ws_ros2/src/ros2_control/hardware_interface/include/hardware_interface/robot_hardware.hpp:22:10: fatal error: control_msgs/msg/dynamic_joint_state.hpp: No such file or directory
   22 | #include "control_msgs/msg/dynamic_joint_state.hpp"
      |          ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
compilation terminated.
make[2]: *** [CMakeFiles/hardware_interface.dir/build.make:115: CMakeFiles/hardware_interface.dir/src/robot_hardware.cpp.o] Error 1
make[2]: *** Waiting for unfinished jobs....
make[1]: *** [CMakeFiles/Makefile2:537: CMakeFiles/hardware_interface.dir/all] Error 2
make[1]: *** Waiting for unfinished jobs....
make: *** [Makefile:141: all] Error 2
---
Failed   <<< hardware_interface [35.0s, exited with code 2]
Aborted  <<< srdfdom [39.1s]
Aborted  <<< warehouse_ros [47.7s]
Aborted  <<< moveit_msgs [56.7s]
Aborted  <<< geometric_shapes [1min 59s]

Summary: 7 packages finished [2min 0s]
  1 package failed: hardware_interface
  4 packages aborted: geometric_shapes moveit_msgs srdfdom warehouse_ros
  3 packages had stderr output: geometric_shapes hardware_interface moveit_msgs
  27 packages not processed

I have added the terminal image. Please let me know what the issue is.

JafarAbdi, 2020-10-22:
Hi @kev278,

Did you run this command to install the dependecies `rosdep install -r --from-paths . --ignore-src --rosdistro foxy -y
`

Also, make sure that you're using the versions as shown [here](https://github.com/ros-planning/moveit2/blob/main/moveit2.repos#L24)

kev278, 2020-10-22:
Hello

Yes I did install rosdep and run the command you have mentioned. The problem is that there is no "control_msgs" folder in the include folder. As per "robot_hardware.hpp" there should be 2 folders in "ros2_control" but there is only 1 named "hardware_interface" I cloned the files as per the command but it seems that a folder is missing.

JafarAbdi, 2020-10-22:
We install control_msgs from the official ROS apt repo, so the file should be located in `/opt/ros/foxy/include/control_msgs/msg/dynamic_joint_state.hpp`

kev278, 2020-10-22:
oh alright, I will add the path in hpp file and then let you know after checking

JafarAbdi, 2020-10-22:
You shouldn't add it manually, can you check if that file does exist .?

kev278, 2020-10-22:
it actually does not exist in the include folder, how do I manually install it?

JafarAbdi, 2020-10-22:
It should be automatically installed when you run `rosdep install -r --from-paths . --ignore-src --rosdistro foxy -y` but you could install it manually by running `sudo apt install ros-foxy-control-msgs`

kev278, 2020-10-22:
[image]
so the previous problem was solved by manually downloading the control_msgs package. I tried redoing the whole installation but it does not install control_msgs, I have to manually install it.

And now this new error

sachinkum0009, 2020-10-24:
I also had this issue earlier.

Try these steps:- 
```bash
cd $COLCON_WS/src

git clone https://github.com/ros-planning/moveit2.git -b main

cd moveit2

git checkout d95e4aecf4ac7a0f8b43fe91742b86251b8762ca

vcs import < moveit2/moveit2.repos

rosdep install -r --from-paths . --ignore-src --rosdistro foxy -y

``` 

**checkout ros2_control to 2c3aad1cb8ad4d23925bceeb5c26e22d6cf859ef and ros2_controllers to d52902538a970e792ecc604416e22ef90a57f29e**

```bash
cd $COLCON_WS

colcon build --event-handlers desktop_notification- status- --cmake-args -DCMAKE_BUILD_TYPE=Release
```

Let me know if it works for you or not ^^

srujan-aris, 2020-10-26:
> I also had this issue earlier.
> 
> Try these steps:-
> 
> ```shell
> cd $COLCON_WS/src
> 
> git clone https://github.com/ros-planning/moveit2.git -b main
> 
> cd moveit2
> 
> git checkout d95e4aecf4ac7a0f8b43fe91742b86251b8762ca
> 
> vcs import < moveit2/moveit2.repos
> 
> rosdep install -r --from-paths . --ignore-src --rosdistro foxy -y
> ```
> 
> **checkout ros2_control to 2c3aad1cb8ad4d23925bceeb5c26e22d6cf859ef and ros2_controllers to d52902538a970e792ecc604416e22ef90a57f29e**
> 
> ```shell
> cd $COLCON_WS
> 
> colcon build --event-handlers desktop_notification- status- --cmake-args -DCMAKE_BUILD_TYPE=Release
> ```
> 
> Let me know if it works for you or not ^^

Worked perfectly for me. Thanks!

dayvsonleandro, 2020-10-28:
I am also having problems following the same steps.

[image]

``dayvson@dayvson:~/ws_moveit2$ colcon build --event-handlers desktop_notification- status- --cmake-args -DCMAKE_BUILD_TYPE=Release
Starting >>> geometric_shapes
Starting >>> moveit_msgs
Starting >>> srdfdom
Starting >>> moveit_common
Starting >>> hardware_interface
Starting >>> moveit_resources_panda_description
Starting >>> moveit_resources_fanuc_description
Starting >>> moveit_resources_pr2_description
Starting >>> warehouse_ros
--- stderr: moveit_common
CMake Error at CMakeLists.txt:4 (find_package):
  By not providing "Findament_cmake.cmake" in CMAKE_MODULE_PATH this project
  has asked CMake to find a package configuration file provided by
  "ament_cmake", but CMake did not find one.

  Could not find a package configuration file provided by "ament_cmake" with
  any of the following names:

    ament_cmakeConfig.cmake
    ament_cmake-config.cmake

  Add the installation prefix of "ament_cmake" to CMAKE_PREFIX_PATH or set
  "ament_cmake_DIR" to a directory containing one of the above files.  If
  "ament_cmake" provides a separate development package or SDK, be sure it
  has been installed.

---
Failed   <<< moveit_common [0.40s, exited with code 1]
Aborted  <<< srdfdom [1.35s]
Aborted  <<< moveit_resources_panda_description [1.46s]
Aborted  <<< moveit_resources_fanuc_description [1.46s]
Aborted  <<< moveit_msgs [1.49s]
Aborted  <<< hardware_interface [1.49s]
Aborted  <<< moveit_resources_pr2_description [1.48s]
Aborted  <<< warehouse_ros [1.44s]
Aborted  <<< geometric_shapes [1.54s]

Summary: 0 packages finished [2.63s]
  1 package failed: moveit_common
  8 packages aborted: geometric_shapes hardware_interface moveit_msgs moveit_resources_fanuc_description moveit_resources_panda_description moveit_resources_pr2_description srdfdom warehouse_ros
  9 packages had stderr output: geometric_shapes hardware_interface moveit_common moveit_msgs moveit_resources_fanuc_description moveit_resources_panda_description moveit_resources_pr2_description srdfdom warehouse_ros
  30 packages not processed

JafarAbdi, 2020-10-28:
> I am also having problems following the same steps.
>
> [image]
>
> ``dayvson@dayvson:~/ws_moveit2$ colcon build --event-handlers desktop_notification- status- --cmake-args -DCMAKE_BUILD_TYPE=Release [...]

@dayvsonleandro Please make sure to source the foxy setup.bash file `source /opt/ros/foxy/setup.bash`

dayvsonleandro, 2020-10-28:
Now i have this erro:

dayvson@dayvson:~/ws_moveit2$ colcon build --event-handlers desktop_notification- status- --cmake-args -DCMAKE_BUILD_TYPE=Release
Starting >>> geometric_shapes
Starting >>> moveit_msgs
Starting >>> srdfdom
Starting >>> hardware_interface
Starting >>> moveit_resources_panda_description
Starting >>> moveit_resources_fanuc_description
Starting >>> moveit_resources_pr2_description
Starting >>> warehouse_ros
Finished <<< moveit_resources_panda_description [0.59s]
Starting >>> moveit_resources_panda_moveit_config
Finished <<< moveit_resources_fanuc_description [0.62s]
Starting >>> moveit_resources_fanuc_moveit_config
Finished <<< moveit_resources_pr2_description [0.66s]
Finished <<< warehouse_ros [0.81s]
Starting >>> warehouse_ros_mongo
Finished <<< geometric_shapes [0.90s]
Finished <<< hardware_interface [0.93s]
--- stderr: srdfdom
CMake Error at /opt/ros/foxy/share/ament_cmake_gtest/cmake/ament_add_gtest_executable.cmake:50 (add_executable):
  Target "test_cpp" links to target "urdfdom_headers::urdfdom_headers" but
  the target was not found.  Perhaps a find_package() call is missing for an
  IMPORTED target, or an ALIAS target is missing?
Call Stack (most recent call first):
  /opt/ros/foxy/share/ament_cmake_gtest/cmake/ament_add_gtest_executable.cmake:37 (_ament_add_gtest_executable)
  /opt/ros/foxy/share/ament_cmake_gtest/cmake/ament_add_gtest.cmake:68 (ament_add_gtest_executable)
  CMakeLists.txt:79 (ament_add_gtest)

CMake Error at /opt/ros/foxy/share/ament_cmake_gtest/cmake/ament_add_gtest_executable.cmake:50 (add_executable):
  Target "test_cpp" links to target "urdfdom_headers::urdfdom_headers" but
  the target was not found.  Perhaps a find_package() call is missing for an
  IMPORTED target, or an ALIAS target is missing?
Call Stack (most recent call first):
  /opt/ros/foxy/share/ament_cmake_gtest/cmake/ament_add_gtest_executable.cmake:37 (_ament_add_gtest_executable)
  /opt/ros/foxy/share/ament_cmake_gtest/cmake/ament_add_gtest.cmake:68 (ament_add_gtest_executable)
  CMakeLists.txt:79 (ament_add_gtest)

CMake Error at /opt/ros/foxy/share/ament_cmake_gtest/cmake/ament_add_gtest_executable.cmake:50 (add_executable):
  Target "test_cpp" links to target "urdfdom_headers::urdfdom_headers" but
  the target was not found.  Perhaps a find_package() call is missing for an
  IMPORTED target, or an ALIAS target is missing?
Call Stack (most recent call first):
  /opt/ros/foxy/share/ament_cmake_gtest/cmake/ament_add_gtest_executable.cmake:37 (_ament_add_gtest_executable)
  /opt/ros/foxy/share/ament_cmake_gtest/cmake/ament_add_gtest.cmake:68 (ament_add_gtest_executable)
  CMakeLists.txt:79 (ament_add_gtest)

CMake Error at /opt/ros/foxy/share/ament_cmake_gtest/cmake/ament_add_gtest_executable.cmake:50 (add_executable):
  Target "test_cpp" links to target "urdfdom_headers::urdfdom_headers" but
  the target was not found.  Perhaps a find_package() call is missing for an
  IMPORTED target, or an ALIAS target is missing?
Call Stack (most recent call first):
  /opt/ros/foxy/share/ament_cmake_gtest/cmake/ament_add_gtest_executable.cmake:37 (_ament_add_gtest_executable)
  /opt/ros/foxy/share/ament_cmake_gtest/cmake/ament_add_gtest.cmake:68 (ament_add_gtest_executable)
  CMakeLists.txt:79 (ament_add_gtest)

CMake Error at /opt/ros/foxy/share/ament_cmake_gtest/cmake/ament_add_gtest_executable.cmake:50 (add_executable):
  Target "test_cpp" links to target "urdfdom_headers::urdfdom_headers" but
  the target was not found.  Perhaps a find_package() call is missing for an
  IMPORTED target, or an ALIAS target is missing?
Call Stack (most recent call first):
  /opt/ros/foxy/share/ament_cmake_gtest/cmake/ament_add_gtest_executable.cmake:37 (_ament_add_gtest_executable)
  /opt/ros/foxy/share/ament_cmake_gtest/cmake/ament_add_gtest.cmake:68 (ament_add_gtest_executable)
  CMakeLists.txt:79 (ament_add_gtest)

CMake Error at CMakeLists.txt:35 (add_library):
  Target "srdfdom" links to target "urdfdom_headers::urdfdom_headers" but the
  target was not found.  Perhaps a find_package() call is missing for an
  IMPORTED target, or an ALIAS target is missing?

CMake Error at CMakeLists.txt:35 (add_library):
  Target "srdfdom" links to target "urdfdom_headers::urdfdom_headers" but the
  target was not found.  Perhaps a find_package() call is missing for an
  IMPORTED target, or an ALIAS target is missing?

CMake Error at CMakeLists.txt:35 (add_library):
  Target "srdfdom" links to target "urdfdom_headers::urdfdom_headers" but the
  target was not found.  Perhaps a find_package() call is missing for an
  IMPORTED target, or an ALIAS target is missing?

CMake Error at CMakeLists.txt:35 (add_library):
  Target "srdfdom" links to target "urdfdom_headers::urdfdom_headers" but the
  target was not found.  Perhaps a find_package() call is missing for an
  IMPORTED target, or an ALIAS target is missing?

CMake Error at CMakeLists.txt:35 (add_library):
  Target "srdfdom" links to target "urdfdom_headers::urdfdom_headers" but the
  target was not found.  Perhaps a find_package() call is missing for an
  IMPORTED target, or an ALIAS target is missing?

CMake Generate step failed.  Build files cannot be regenerated correctly.
---
Failed   <<< srdfdom [0.94s, exited with code 1]
Aborted  <<< moveit_resources_panda_moveit_config [0.33s]
Aborted  <<< moveit_resources_fanuc_moveit_config [0.29s]
Aborted  <<< moveit_msgs [1.00s]
Aborted  <<< warehouse_ros_mongo [0.19s]

Summary: 6 packages finished [1.40s]
  1 package failed: srdfdom
  4 packages aborted: moveit_msgs moveit_resources_fanuc_moveit_config moveit_resources_panda_moveit_config warehouse_ros_mongo
  1 package had stderr output: srdfdom
  26 packages not processed

sachinkum0009, 2020-11-04:
Should we close this issue?

dqii, 2020-11-10:
I tried both ```source /opt/ros/foxy/setup.bash``` as suggested by @JafarAbdi and checking out different versions as suggested by @sachinkum0009 but still had errors. My last error was below.

```
di@di:~/ws_ros2$ colcon build --event-handlers desktop_notification- status- --cmake-args -DCMAKE_BUILD_TYPE=Release
Starting >>> urdfdom_headers
[0.594s] WARNING:colcon.colcon_ros.prefix_path.ament:The path '/home/di/ws_ros2/install/srdfdom' in the environment variable AMENT_PREFIX_PATH doesn't exist
[0.594s] WARNING:colcon.colcon_ros.prefix_path.ament:The path '/home/di/ws_ros2/install/moveit_common' in the environment variable AMENT_PREFIX_PATH doesn't exist
[0.605s] WARNING:colcon.colcon_ros.prefix_path.catkin:The path '/home/di/ws_ros2/install/srdfdom' in the environment variable CMAKE_PREFIX_PATH doesn't exist
[0.605s] WARNING:colcon.colcon_ros.prefix_path.catkin:The path '/home/di/ws_ros2/install/moveit_common' in the environment variable CMAKE_PREFIX_PATH doesn't exist
Starting >>> geometric_shapes
Starting >>> moveit_msgs
Starting >>> hardware_interface
Finished <<< urdfdom_headers [0.42s]
Starting >>> srdfdom
Finished <<< srdfdom [28.2s]
Starting >>> urdfdom
[Processing: geometric_shapes, hardware_interface, moveit_msgs, urdfdom]
Finished <<< hardware_interface [1min 7s]
Starting >>> moveit_resources_panda_description
Finished <<< moveit_resources_panda_description [4.71s]
Starting >>> moveit_resources_fanuc_description
Finished <<< urdfdom [42.9s]
Starting >>> moveit_resources_pr2_description
Finished <<< moveit_resources_fanuc_description [4.21s]
Starting >>> controller_interface
Finished <<< moveit_resources_pr2_description [4.44s]
Starting >>> moveit_resources_panda_moveit_config
Finished <<< moveit_resources_panda_moveit_config [4.13s]
Starting >>> moveit_resources_fanuc_moveit_config
Finished <<< moveit_resources_fanuc_moveit_config [4.23s]
Starting >>> test_robot_hardware
^CAborted  <<< test_robot_hardware [4.67s]
Aborted  <<< controller_interface [13.4s]
Aborted  <<< moveit_msgs [1min 29s]
Aborted  <<< geometric_shapes [1min 29s]

Summary: 9 packages finished [1min 29s]
  4 packages aborted: controller_interface geometric_shapes moveit_msgs test_robot_hardware
  3 packages had stderr output: controller_interface geometric_shapes moveit_msgs
  24 packages not processed
```

dqii, 2020-11-10:
@kev278 did this end up working for you?

mqjinwon, 2020-11-18:
> I also had this issue earlier.
>
> Try these steps:-
>
> ```shell
> cd $COLCON_WS/src
>
> git clone https://github.com/ros-planning/moveit2.git -b main
>
> cd moveit2
>
> git checkout [...]

@sachinkum0009  it works well! thank you so much

JafarAbdi, 2021-04-15:
Look like this got fixed

closed 2021-04-15 by JafarAbdi