moveit/moveit2 #300: Implement a concept of global and local planner

sjahr, 2020-10-27:
## Intro
Currently, MoveIt’s core strength is offline, global motion planning in a kind of “Sense-Plan-Act” based behavior. Most scenarios in which robots are used involve a dynamic environment with moving obstacles and target poses. To make MoveIt better utilizable in these scenarios I started working on a hybrid motion planning architecture as part of [MoveIt 2 roadmap](https://moveit.ros.org/documentation/contributing/roadmap/) Milestone 2: Real-time Support #277. The idea is to combine a slower global planner and a fast local planner to enable online motion planning/replanning and reactive planning/control.

To gain as much feedback and input as possible I want to share the progress of the ongoing work on this issue. Please leave a comment on everything that’s on your mind related to hybrid motion planning with MoveIt.

## Next Step
Approximately within the next two weeks, I plan to add a detailed architectural draft of the concept. If you have user stories on your mind, I would really appreciate to hear them in the comments. Thanks in advance!

gokul-gokz, 2020-11-12:
Hello @sjahr !
I am really interested in this architecture and I would love to contribute to this. So, please let me know is there any way we can collaborate. I have a question ,for the local planner, is it going to be the same set of planners available in global planners or is it going to be specific planners(Jacobian based) which are faster than the available global planners.

sjahr, 2020-11-16:
> Hello @sjahr !
> I am really interested in this architecture and I would love to contribute to this. So, please let me know is there any way we can collaborate. I have a question ,for the local [...]

Hi @gokul-gokz, 
nice to hear that you are interested in the architecture. We are planning to use specific planners (e.g. Jacobian or artificial potential field based) as local planners because the planners currently available in MoveIt are not necessarily fast enough or suitable for online motion planning. Furthermore, it should be possible to use the local planner for trajectory execution.
On the other hand, we want to make architecture as user-customizable as possible, so you should not be limited by the planners we had in mind. The goal is to design the architecture in a way that it is possible to use or combine every planner you want (similar to [MoveIt's current implementation](https://ros-planning.github.io/moveit_tutorials/doc/creating_moveit_plugins/plugin_tutorial.html)).

In case you are interested in contributing, it would be great if you could give a rough description of how or in which scenario you want to use hybrid planning. Additionally, I've started with the implementation of a prototype last week and plan to open a work-in-progress PR this week. I could add you as a reviewer once a first functioning version exists if you are interested.

Briancbn, 2020-11-16:
Hi @sjahr,  first of all, I am also very interested in this effort and thanks so much for making it happen.

> nice to hear that you are interested in the architecture. We are planning to use specific planners (e.g. Jacobian or artificial potential field based) as local planners because the planners currently available in MoveIt are not necessarily fast enough or suitable for online motion planning

I think you can also have a look at [tesseract_ros](https://github.com/ros-industrial-consortium/tesseract_ros#online-planning-example) where Trajopt is used. 

Another question that I have is that will the collision checking be done at realtime during execution as well, in this implementation?

sjahr, 2020-11-16:
Thanks for the hint @Briancbn. Yes, collision checking is planned to be done at real-time during the execution. We want to include a trajectory monitor in the hybrid planning manager for online collision checking of the global trajectory (assuming that during the execution unforeseen collisions could appear). Furthermore, one of the use-cases of the local planner could be to react at real-time to new detected collisions based on sensor data. How the system reacts to collision's will depend on customizable logic inside the planning manager.

davetcoleman, 2020-11-18:
Nice write-up @sjahr!

I realize you do not have a ton of time to implement this, so the following concerns may not be relevant at this stage of development. However I would encourage you to think about this as you working on this project:

- I think hybrid planning should be tightly integrated into the core MoveIt components, not an add-on thing on the side. This means the high levle interface should work from moveit_cpp (and maybe even move_group, but I think that should be deprecated personally).
- Don't be afraid to refactor core MoveIt components to support this. For example, the TrajectoryExecutionManager should maybe include a streaming mode. Or we refactor KinematicBase to work as a local planner.

AndyZe, 2020-11-19:
You should consider inheriting from a ros2_control JointTrajectoryController for the Local Planner. That will ensure the system is realtime safe. It also allows you to update trajectories quickly (maybe 100 Hz or so). Finally, it eliminates one extra component in the pipeline. I don't see any downsides to inheriting from a JointTrajectoryController.

See 'trajectory replacement`:  http://wiki.ros.org/joint_trajectory_controller#Trajectory_replacement

AndyZe, 2020-11-19:
Actually, I'm not 100% sure if you would want to inherit from a JointTrajectoryController or a ForwardCommandController. ForwardCommandController would let you hit the really high control rates.

sjahr, 2020-11-24:
Thanks a lot for the input! I was definitely thinking about including some kind of wrapper to make ros2_controllers available as local planners. I agree with you that often a local planner is basically a controller. I had a hard time distinguishing between both as the line between a local planner and a controller is blurry and one could even say that a controller is a kind of local planner. One good explanation I've heard is that a controller is an equation and a planner has an internal state. Please, correct me if you think that's wrong.
Regard the inheritance, I am not sure if that would work because I don't want to limit users to controllers as local planners.  It should be possible to utilize the planner to optimize the trajectory locally or even to locally sample new points for example to dodge an unexpected collision object or grasp something.
I will try to add a few examples of use cases, where a pure closed-loop controller is probably not sufficient as a local planner within the next days!

AndyZe, 2020-11-24:
OK, I'm glad you're thinking about these things.

Inheriting from the controller doesn't necessarily mean you have to implement the planner like a controller. Think of it more in coding terms -- it gives you access to the trajectory.

I do think this is wrong. Even a simple PID controller has a state (the error and the error integral):

`One good explanation I've heard is that a controller is an equation and a planner has an internal state.`

AndyZe, 2020-11-24:
I did something like that in ROS1 before:  https://drive.google.com/file/d/15Q2UxuUGHTMoC76gnOzB3C8KYzzEa9BG/view?usp=sharing

sjahr, 2020-12-04:
> I do think this is wrong. Even a simple PID controller has a state (the error and the error integral).

Good point, I've corrected the comment above, and thanks again for your hints.

tianshiz, 2020-12-15:
Hi guys, excited about the potentials here. One question is, how is **realtime** defined here?  Could I expect to run this on a RT kernel and get some kind of soft realtime expectation? I ask because I can see applications in flight hardware running ROS2+MoveIt if traditional realtime requirements could be met or approximated.

sjahr, 2020-12-17:
@tianshiz Yes, goal is that the architecture will be somehow soft real-time capable. For example by using ROS 2 and probably some kind of (soft) real-time capable Linux kernel.

closed 2021-04-22 by sjahr