moveit/moveit2 #305: Source build error @ colcon build

dqii, 2020-11-10:
I followed the instructions for installing ROS2 from source (which worked) and for MoveIt2, but I'm getting stuck at this command for MoveIt2.
```
colcon build --event-handlers desktop_notification- status- --cmake-args -DCMAKE_BUILD_TYPE=Release
```

This is the output

```
di@di:~/ws_ros2$ colcon build --event-handlers desktop_notification- status- --cmake-args -DCMAKE_BUILD_TYPE=Release

Starting >>> geometric_shapes
[0.599s] WARNING:colcon.colcon_ros.prefix_path.ament:The path '/home/di/ws_ros2/install/srdfdom' in the environment variable AMENT_PREFIX_PATH doesn't exist
[0.599s] WARNING:colcon.colcon_ros.prefix_path.ament:The path '/home/di/ws_ros2/install/moveit_common' in the environment variable AMENT_PREFIX_PATH doesn't exist
[0.612s] WARNING:colcon.colcon_ros.prefix_path.catkin:The path '/home/di/ws_ros2/install/srdfdom' in the environment variable CMAKE_PREFIX_PATH doesn't exist
[0.612s] WARNING:colcon.colcon_ros.prefix_path.catkin:The path '/home/di/ws_ros2/install/moveit_common' in the environment variable CMAKE_PREFIX_PATH doesn't exist
Starting >>> moveit_msgs
Starting >>> srdfdom
Starting >>> moveit_common
Finished <<< moveit_common [2.55s]
Starting >>> hardware_interface
--- stderr: moveit_msgs
CMake Error at CMakeLists.txt:22 (find_package):
  By not providing "Findobject_recognition_msgs.cmake" in CMAKE_MODULE_PATH
  this project has asked CMake to find a package configuration file provided
  by "object_recognition_msgs", but CMake did not find one.

  Could not find a package configuration file provided by
  "object_recognition_msgs" with any of the following names:

    object_recognition_msgsConfig.cmake
    object_recognition_msgs-config.cmake

  Add the installation prefix of "object_recognition_msgs" to
  CMAKE_PREFIX_PATH or set "object_recognition_msgs_DIR" to a directory
  containing one of the above files.  If "object_recognition_msgs" provides a
  separate development package or SDK, be sure it has been installed.

---
Failed   <<< moveit_msgs [4.38s, exited with code 1]
Aborted  <<< srdfdom [4.59s]
Aborted  <<< hardware_interface [2.41s]
Aborted  <<< geometric_shapes [5.33s]

Summary: 1 package finished [5.76s]
  1 package failed: moveit_msgs
  3 packages aborted: geometric_shapes hardware_interface srdfdom
  3 packages had stderr output: geometric_shapes hardware_interface moveit_msgs
  34 packages not processed
```

I tried removing the package and trying again but it still isn't working.

JafarAbdi, 2020-11-10:
Hi @dqii,

Please make sure to run the following command `source /opt/ros/foxy/setup.bash`

dqii, 2020-11-10:
Thanks for the reply @JafarAbdi !

I just tried that and ended up with a different error.
```
Starting >>> moveit_ros_occupancy_map_monitor
Starting >>> moveit_simple_controller_manager
Finished <<< moveit_simple_controller_manager [20.1s]
Finished <<< moveit_ros_occupancy_map_monitor [39.0s]
Starting >>> moveit_ros_planning
[Processing: moveit_ros_planning]
--- stderr: moveit_ros_planning
In file included from /home/di/ws_ros2/src/moveit2/moveit_ros/planning/planning_pipeline/src/planning_pipeline.cpp:37:
/home/di/ws_ros2/src/moveit2/moveit_ros/planning/planning_pipeline/include/moveit/planning_pipeline/planning_pipeline.h:41:10: fatal error: pluginlib/class_loader.hpp: No such file or directory
   41 | #include <pluginlib/class_loader.hpp>
      |          ^~~~~~~~~~~~~~~~~~~~~~~~~~~~
compilation terminated.
make[2]: *** [planning_pipeline/CMakeFiles/moveit_planning_pipeline.dir/build.make:63: planning_pipeline/CMakeFiles/moveit_planning_pipeline.dir/src/planning_pipeline.cpp.o] Error 1
make[1]: *** [CMakeFiles/Makefile2:514: planning_pipeline/CMakeFiles/moveit_planning_pipeline.dir/all] Error 2
make[1]: *** Waiting for unfinished jobs....
make: *** [Makefile:141: all] Error 2
---
Failed   <<< moveit_ros_planning [38.2s, exited with code 2]

Summary: 25 packages finished [10min 0s]
  1 package failed: moveit_ros_planning
  2 packages had stderr output: moveit_core moveit_ros_planning
  11 packages not processed
```

dqii, 2020-11-10:
I had changed the versions of ros2_control and ros2_controllers following @sachinkum0009's suggestion on https://github.com/ros-planning/moveit2/issues/297, I just reverted that, and it seems to cause a different error.

```
Starting >>> moveit_ros_planning
Finished <<< transmission_interface [11.9s]
Starting >>> moveit_simple_controller_manager
Finished <<< moveit_simple_controller_manager [2.11s]
--- stderr: joint_limits_interface
In file included from /home/di/ws_ros2/src/ros2_control/joint_limits_interface/test/joint_limits_urdf_test.cpp:19:
/home/di/ws_ros2/src/ros2_control/joint_limits_interface/include/joint_limits_interface/joint_limits_urdf.hpp:22:10: fatal error: urdf/urdfdom_compatibility.h: No such file or directory
   22 | #include <urdf/urdfdom_compatibility.h>
      |          ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
compilation terminated.
make[2]: *** [CMakeFiles/joint_limits_urdf_test.dir/build.make:63: CMakeFiles/joint_limits_urdf_test.dir/test/joint_limits_urdf_test.cpp.o] Error 1
make[1]: *** [CMakeFiles/Makefile2:159: CMakeFiles/joint_limits_urdf_test.dir/all] Error 2
make: *** [Makefile:141: all] Error 2
---
Failed   <<< joint_limits_interface [37.5s, exited with code 2]
Aborted  <<< controller_manager [1min 1s]
Aborted  <<< moveit_ros_planning [1min 11s]

Summary: 19 packages finished [1min 53s]
  1 package failed: joint_limits_interface
  2 packages aborted: controller_manager moveit_ros_planning
  2 packages had stderr output: joint_limits_interface moveit_ros_planning
  21 packages not processed
```

JafarAbdi, 2021-04-15:
I'll close this, please feel free to open an issue if you still can't build moveit2

closed 2021-04-15 by JafarAbdi