moveit/moveit2 #313: Migrate MoveItSetupAssistant to ROS 2

henningkayser, 2020-12-04:
Porting the MSA requires more effort than simply applying the migration steps. Since we are switching to Python launch files, I suggest the following steps for porting:

1. Apply basic package migration while dropping xml launch file templates and generators
2. Fixup config files based on new yaml format constraints (missing map support). 
3. Prepare a set of default launch scripts that are then maintained in the MoveIt Repo 

IMO MoveIt configs should be static by default and only contain the config files needed to describe the setup configuration. The provided launch scripts are not copied into each package, they are only used to initialize a setup by reading the config files from a target package. Customization of launch descriptions can be enabled by supporting launch file overrides or exporting/importing of factory functions. This would allow to keep a standard and reduce redundancy (and maintenance overhead) in launch files. Of course it's up to the devs to copy the whole launch architecture and modify it as needed.

Some items that came up with the discussion:
* Need for standardizing parameter usage for MoveIt components (proper declaration & documentation, deprecation handling)
* Option for keeping versions for the config files to notify users about changes and help with migration

DLu, 2021-10-11:
# MoveIt Setup Assistant Refactor Proposal

I propose that MSA be reimplemented using pluginlib. Most of the "widgets" would be moved to separate packages and implemented as plugins. The core `moveit_setup_assistant` package would only contain a few key elements.
 * List of available plugins
 * Set of shared data files
 * Functionality for loading some subset of the plugins
 * Framework to populate the GUI (including RViz display)
 * Intro and Outro widgets

The data files need to be shared at the high level because there are multiple configuration steps that operate on the same data file (i.e. the SRDF) and some configuration steps will need to load data from other configuration steps.

The overall control flow might be something like:
 * Show the Initial `SetupScreenWidget` to select the URDF to operate on OR load an existing `moveit_config` package (same as before)
 * With an eye toward #449, ask the user what they want to do.
    * Each of the available plugins could report a hierarchical string describing what that configuration step does. For example, the `VirtualJointsWidget` could report something like `["Configure the SRDF", "Update the Virtual Joints"]` and multiple operations would be grouped under the section "Configure the SRDF"
    * User would then select a subset of the options which would dictate which plugins get loaded.
 * The user would then walk through each of the chosen configuration steps.
    * There needs to be a mechanism for specifying dependencies, so that, for example, you can't specify robot poses until you configure the planning groups.
 * At the end, you are presented with the `ConfigurationFilesWidget` which presents the different shared files that need to be written to.

As much as possible, the GUI logic should be kept separately from the core data-management.

v4hn, 2021-10-17:
> I propose that MSA be reimplemented using pluginlib.

Reading this from you reminded me of slide 30 [from your 2017 ROSCon talk](https://roscon.ros.org/2017/presentations/ROSCon%202017%20Fundamentals%20of%20Local%20Planning.pdf). :joy:

I would love plugin support for MSA for users to add their own configuration options.
I don't see the merit in "outsourcing" all standard tabs though and would keep them local classes. But, granted, the difference between local class and plugin in the local package is minor.

As you said the defining data structures need to be defined outside the plugins anyway.

Asking what steps to perform later on would be just another technical hurdle for the user.
It's equally possible to always add all plugins/widgets to the list. Maybe invoke them lazily if the user clicks on them.

> As much as possible, the GUI logic should be kept separately from the core data-management.

:+1: It would be great to have cli programs for all functionality, if just to finally write reasonable CI tests for MSA!

gavanderhoorn, 2021-10-17:
As I don't see it mentioned yet: https://github.com/ros-planning/moveit/pull/1375.

henningkayser, 2021-11-02:
Hey @DLu I’m using ZenHub in GitHub, [click this link](https://app.zenhub.com/workspaces/moveit-61675936b391800012280f6d/board) to join my workspace and see other features available in GitHub or [download the ZenHub extension](https://www.zenhub.com/extension) and sign up with your GitHub account. 
  _Posted using [ZenHub](https://www.zenhub.com)_

cgarry-vs, 2022-05-11:
Are you aiming to have this go out with the Humble Hawksbill release?

JafarAbdi, 2022-05-12:
> Are you aiming to have this go out with the Humble Hawksbill release?

Yes, that's the plan

vatanaksoytezer, 2022-05-12:
We plan to attempt to merge feature MSA branch very soon after https://github.com/ros-planning/moveit2/pull/1240, and the plan is after some testing next week we will hopefully release Humble with MSA.

nlgilbert, 2022-06-16:
Any updates on getting MSA in ROS 2? Will we see it in any versions prior to Humble (Foxy or Galactic)?

DLu, 2022-06-16:
It is in the process of being merged into `main`: https://github.com/ros-planning/moveit2/pull/1254 and will be backported to old releases.

jure-no, 2022-06-20:
Is there any new information, when will we get MSA for humble? It would really help me a lot

jure-no, 2022-06-20:
Is this expected to be published in the next 14 days to one month?

vatanaksoytezer, 2022-06-20:
Don't want to make any promises but, yes hopefully it should be ready in that time interval

mikeferguson, 2022-06-27:
Branch is merged - kudos to @DLu and @vatanaksoytezer

jure-no, 2022-06-28:
Really? that is awesome. Where can i get MSA for moveit2 now?

vatanaksoytezer, 2022-06-28:
@jure-no  You can use the main branch to compile on Humble or Rolling by following the instructions on our webpage, please let us know if you see any issues.

@mikeferguson we can close the ticket, thanks for the heads up!

closed 2022-06-28 by vatanaksoytezer

mikeferguson, 2022-06-28:
@jure-no I will note, MOST things work with the MSA - but you'll see a few remaining bug tickets for certain more advanced features (and some active PRs) if you search "MSA" in our issue tracker. If you run into any issues, please post an issue (or comment on an existing one if that particular bug is blocking you).