moveit/moveit2 #314: Migrate Python bindings

henningkayser, 2020-12-04:
Tracking issue to discuss the need and capacity for porting the `py_binding_tools`, `moveit_commander`.

rhaschke, 2020-12-04:
It's on my todo list (unfortunately, since ages) to migrate python bindings to pybind11 and clean them up in Noetic.
I progressed a lot in MTC on that and hopefully can adapt main MoveIt more easily now.

maslovay, 2021-01-13:
@rhaschke any news with with migrating py_binding_tools, moveit_commander?

WyattAutomation, 2021-03-05:
+1 for this enhancement request; critical feature for several use cases I will need it for.  

Would be willing to help test/review if/when this enhancement nears completion.  Thank you for your effort on this, it's definitely appreciated.

felixvd, 2021-03-06:
Related: https://github.com/ros-planning/moveit/issues/900

lorepieri8, 2021-07-30:
+1 to see Python fully supported

tlpss, 2021-09-25:
@rhaschke has there been any update on migrating the python bindings? I might be willing to offer some help as I am really interested in the python bindings, but my experience with Moveit is limited.

henningkayser, 2022-06-02:
closing in favor of #1279

closed 2022-06-02 by henningkayser