moveit/moveit2 #325: Compatibility of tutorials with ROS2

gsion1, 2021-01-07:
Hi, 
I've found a lot of well explained tutorials for moveIt2 but most of them seems to target ROS1 and use _rospy_ module or the _roslaunch_ command. As a beginner I find that it's really hard to begin with moveIt2 and python scripts to define poses/goals etc. 
(Please excuse me if I'm wrong but I'm very new to ros and the differences between ROS1 and ROS2 are sometimes hard to understand, especially when some packages contain a "2" but work with ROS1.)

For example this [tutorial](https://ros-planning.github.io/moveit_tutorials/doc/move_group_python_interface/move_group_python_interface_tutorial.html) about Move Group is very complete but seems, at least with my ROS knowledge unusable with ROS2 without some modifications (and obviously replacing _roslaunch_ with _ros2 launch_ did not work). 
Moreover, the [python tutorial](https://github.com/ros-planning/moveit2/tree/main/moveit_demo_nodes/run_move_group) that I was able to run seems to have the ["interesting" part written in C](https://github.com/ros-planning/moveit2/blob/main/moveit_demo_nodes/run_move_group/src/run_move_group.cpp), which is frightening for beginners like me.

Does ROS2 tutorials exist and is there a separate place to find them ? I found nothing after hours of researches. 
If not, is it planned to port tutorials to ROS2 (and python)? The roadmap doesn't give information about documentation

Thanks a lot for your work on MoveIt !

tylerjw, 2021-01-07:
Welcome, @gsion1.  ROS2 specific tutorials are definitely on our radar and we are hoping to have them soon.  In the meantime, in addition to the move_group example, there is this for using it in c++: https://github.com/ros-planning/moveit2/tree/main/moveit_demo_nodes/run_moveit_cpp

> seems to have the "interesting" part written in C, which is frightening for beginners like me.

Right now the focus for us with MoveIt with ROS2 is to first support using it in C++.  From our perspective, many of the advantages of ROS2 are performance-related and therefore we should first focus on applications that are written in C++.  Long-term ROS1 will be replaced by ROS2 and therefore we'll need to support the python applications that MoveIt for ROS1 supported to continue to support people who Python has acceptable enough performance.

That being said, I'm just giving you my perspective on the priorities for MoveIt2.  If you'd like to be involved in helping as we begin to support python better.

gsion1, 2021-01-08:
Thanks for your answer, I understand that C++ is better for this application than python

 I really would like to get involved but as I said my programming skills are not sufficient to understand the C++ example and port it to python, at least without "an example"

closed 2021-01-08 by gsion1