moveit/moveit2 #328: Migration of moveit_setup_assistant and moveit_commander

maslovay, 2021-01-13:
Thank you for your work on MoveIt2. 

Please tell us when moveit_setup_assistant and moveit_commander will be in production. 
The code, that is now presented in git has errors on build.

```
CMake Error at CMakeLists.txt:23 (find_package):
  By not providing "Findcatkin.cmake" in CMAKE_MODULE_PATH this project has
  asked CMake to find a package configuration file provided by "catkin", but
  CMake did not find one.

  Could not find a package configuration file provided by "catkin" with any
  of the following names:

    catkinConfig.cmake
    catkin-config.cmake

  Add the installation prefix of "catkin" to CMAKE_PREFIX_PATH or set
  "catkin_DIR" to a directory containing one of the above files.  If "catkin"
  provides a separate development package or SDK, be sure it has been
  installed.
```

henningkayser, 2021-05-04:
Hi @maslovay, we've started making efforts to port MSA with milestone https://github.com/ros-planning/moveit2/milestone/3. Please feel welcome to get involved with the corresponding issues.

closed 2021-05-04 by henningkayser

HappySamuel, 2024-12-06:
Hi

What to use for python API of moveit2 on ROS2 Humble?

Best
Samuel