moveit/moveit2 #329: Rolling Ridley

tylerjw, 2021-01-13:
Create documentation on how to use MoveIt2 on Rolling Rodley from the source.

tylerjw, 2021-03-15:
These are the packages we need to get released before we build on rolling (or add as source dependencies):

```
~ rosdep install --from-paths src --ignore-src -y
ERROR: the following packages/stacks could not have their rosdep keys resolved
to system dependencies:
moveit_msgs: Cannot locate rosdep definition for [object_recognition_msgs]
moveit_ros_perception: Cannot locate rosdep definition for [object_recognition_msgs]
geometric_shapes: Cannot locate rosdep definition for [random_numbers]
joint_trajectory_controller: Cannot locate rosdep definition for [realtime_tools]
diff_drive_controller: Cannot locate rosdep definition for [realtime_tools]
moveit_ros_visualization: Cannot locate rosdep definition for [object_recognition_msgs]
moveit_servo: Cannot locate rosdep definition for [control_toolbox]
moveit_ros_occupancy_map_monitor: Cannot locate rosdep definition for [octomap]
moveit_planners_ompl: Cannot locate rosdep definition for [ompl]
moveit_core: Cannot locate rosdep definition for [random_numbers]
```

vatanaksoytezer, 2021-06-16:
Hey @Drojas251 moveit2 currently works on rolling and galactic. A CI has already been setup for both and currently passing.

There is a nice video that @tylerjw recorded here: https://www.youtube.com/watch?v=FgIK27jN4bk.  Differing from video all the fixes are merged into ros_control and moveit2, so you won't need to checkout these PRs. Current main branch compiles just fine.

Drojas251, 2021-06-16:
@vatanaksoytezer  Wonderful! Thank you!

vatanaksoytezer, 2021-07-14:
Closed by #494

closed 2021-07-14 by vatanaksoytezer