moveit/moveit2 #336: Servo test hang in CI

tylerjw, 2021-01-20:
### Description

When the tests in moveit_servo are running on CI in Travis sometimes they hang (and eventually timeout) waiting for the `client_servo_start_` service to be available.  This is probably caused by a race condition or deadlock of some sort where the servo_serer is hanging during initialization or its threads are never being scheduled and the test never makes it past waiting for servo to be ready to use.  This could also be somehow related to the number of threads on the Travis runner and we should figure out how to modify the tests so they are robust on machines with few threads.

Here is an example of the failure: https://travis-ci.com/github/ros-planning/moveit2/jobs/473193304  See lines 1600-1695.

rhaschke, 2021-01-20:
Linux doesn't have an explicit limit on the number of threads. Do you have evidence that Travis has such a limit?
If not, I don't think this is the culprit here.

tylerjw, 2021-01-20:
I think the issue is not a limit on the number of threads but the actual number of threads running and as a result, for some reason only some of the threads are executing causing some of them (the one that should be initializing servo (and therefore providing this service) or whatever it is blocked on) to not execute.

That being said I can't reproduce this failure locally sadly which makes it hard to debug.

tylerjw, 2021-02-11:
I'm closing this because it isn't happening anymore.

closed 2021-02-11 by tylerjw