moveit/moveit2 #355: Demo launch for run_move_group fails

chiatseen, 2021-02-03:
### Description

I have Moveit2 installed on Ubuntu 20.04 with ROS2 Foxy and attempted to launch run_move_group but failed. 

### Your environment
* ROS Distro: ROS2 Foxy
* OS Version: Ubuntu 20.04.1 LTS
* Source build
* Branch main at e108c35

### Steps to reproduce
1. Get fresh Ubuntu 20.04.1 LTS
2. Install ROS2 Foxy with build tools
3. Build moveit2 by the [guide](https://moveit.ros.org/install-moveit2/source/)
4. Launch the run_move_group [demo](https://github.com/ros-planning/moveit2/tree/main/moveit_demo_nodes/run_move_group)

### Expected behaviour
Running the launch file should launch panda, and running the interface launch file should show planning for the different scenarios described in [the cpp file](https://github.com/ros-planning/moveit2/blob/main/moveit_demo_nodes/run_move_group/src/run_move_group.cpp)

### Actual behaviour
Panda launches as shown in the image below
[image]

Running the run_move_group_interface file gives the error:
`[ERROR] [move_group_interface]: MoveGroupInterface:plan() failed or timeout reached`

### Backtrace or Console output

[output](https://gist.github.com/chiatseen/a8ce050836ee4d86a4ced5b53f2344ef)

JafarAbdi, 2021-02-03:
Hi @chiatseen,

Which branch of fake_joint are you using .?

chiatseen, 2021-02-03:
Hi @JafarAbdi ,

I am on branch foxy.

Thank you

closed 2021-03-30 by henningkayser