moveit/moveit2 #363: World MoveIt Day 2021 Tracks

tylerjw, 2021-02-09:
### Description

Tracks for World MoveIt day.  These will have separate Google Meet and Discord rooms with maintainers supporting groups of participants in working on various parts of MoveIt.  The goal is to allow users to self-select based on their background and interests into different tracks where they can contribute.

### Robotics Challenge: Sanding Demo

Introduction: Users will be shown demos of the latest features (hybrid planning, ompl constraints, etc) and given a new environment with a robot and object to be sanded.  Users will work together to use the new features to create a demo of generating plans for tracing the surface of the object with the end effector.
Background: Robotics experience using MoveIt doing path planning and excited about using MoveIt2 to solve robotics problems.
Maintainers:
Tasks:
- [ ] Assign maintainers to lead this track
- [ ] Prepare environment for demo
- [ ] Prepare readme with links to resources (tutorials, etc)

### Intro to Contributing

Introduction: Users will be given an introduction to ament_lint and shown instructions on how to run various linters and build/test moveit2.
Background: Experience with building and testing software and interest in becoming a contributor.
Maintainers: @tylerjw 
Tasks:
- [ ] Prepare readme with links to resources
- [ ] Merge PR enabling ROS2 linting

### Intro to ROS2 and MoveIt2

Introduction: We need issues to understand where the tutorials and demos are confusing to new users.  The goal of this group will be to do tutorials, create those issues, and begin conversations around how to make tutorials better.  This group will be invited to go through both the official ROS2 tutorials and the MoveIt2 tutorials.
Background: Users who are new to ROS2 and MoveIt and are excited about getting started.
Maintainers:
Tasks:
- [ ] Assign maintainers to lead this track
- [ ] Prepare readme with links to resources (tutorials, etc)

### Meta

This is our rough initial design for MoveIt day.  This is here so we can work together on the plan for World MoveIt day to make it really productive and good for new users.  We know there are many rough edges and usability issues with MoveIt2 and hope that this can help round some of those issues and lower the barrier to entry for new users.  To that point, we are planning on preparing an LXD container with instructions for using it to allow everyone to start using moveit2 quickly.

mamoll, 2021-02-10:
The sanding demo sounds challenging. It could be split into two main subproblems:
* Take a predefined sanding tool path, and turn it into a constraint function that can be used with the new constrained motion planning that @JeroenDM and @bostoncleek have been working on. This constraint function would take arbitrary joint states and project them onto a nearby join state s.t. the end effector is _somewhere_ along the tool path. Constrained interpolation allows you to move along the tool path for two valid join states. 
* Take a valid trajectory and execute it subject to force constraints: force in the surface normal direction should above some value, forces in all other directions should be below some level. No idea how easy this is. @AndyZe?

tylerjw, 2021-02-10:
The idea of the sanding problem was an attempt at creating a fun robotics project for those that have the correct background and interest.  It also seems like a great way to announce get people excited about the new features in moveit2.  This plan is mostly an attempt to make WMD useful and engaging for people of varying backgrounds and skillsets.  Please change/challenge any part of the plan itself too.

AndyZe, 2021-02-11:
Sanding is gonna be more of a ros_control problem, I think. There are some practical issues, like:

-how are forces going to be sensed, for people working remotely? Is the PickNik ur5e going to be used?
- if we would rather simulate, then be aware that Gazebo and Ignition aren't very capable in ROS2 yet.

I think sanding is the type of problem that's going to require one or two people hacking for a few days, rather than a group of 5 people hacking for one day. It takes focus.

AndyZe, 2021-02-11:
Here are some ideas that are more bite sized:

- get a robot working with MoveIt in Gazebo or Ignition
- Port tutorials
- we could have a list of extraneous features we want removed. It's easier to remove code and test it than write new code, I think
- Write integration tests for the tutorials. The test could simply check that the tutorial nodes don't die. It doesn't need to be fancy.

AndyZe, 2021-02-11:
A ros2_control help room would be great, to help people start using more hardware.

tylerjw, 2021-02-11:
@AndyZe thank you for the reply.  Is there one of the things you listed that you think you'd like to lead a group doing?  Sanding is a really hard problem and I'm not sure it is possible to completely do it, just that it would be a fun way to get people to start hacking using the new features in planning.

AndyZe, 2021-02-11:
I'd be interested in doing a ros2_control help room. Hopefully with @destogl and @livanov93

mamoll, 2021-02-11:
If you change sanding to, say, spray painting, then the first part of what I wrote above (finding a continuous point path for a given tool path) would still be useful on its own.

felixvd, 2021-02-12:
I like the idea of topic rooms / shared goals rather than one big "ok let's all develop something" because it's better not to feel lost as a newcomer, but we will need to be careful with topic selection and making sure you can join it at any time (and from any time zone). Am I getting right that this is meant to group people into roughly "Applications", "Contributing" and "ROS2"?

tylerjw, 2021-02-12:
The painting application would get around trying to do force control.  I think the idea here is that there is a handful of separate rooms lead by different people that people could select based on background and interests.  In the past when we'd have people come into our office it was a challenge to accomplish a change that'd get merged that day.  For the "intro to contributing" group, I'm hoping to prepare a guide and have it be fixing linting issues so we can turn on linters used on the ros2 project itself on moveit2.

Thank you for pointing out that we need to make this work across timezones.  For these tracks, we should publish when each of us plans on being where so people know when they can find us if they want to talk with us directly during this.  For the work, I am hoping much of it can be somewhat self-directed and based on some work prepared by whoever is leading each group.  However, by trying to have multiple people work on similar tasks we can hopefully work together at some level.  At the very least we'll be able to do reviews easily because we should understand what the work is trying to do.

mlautman, 2021-02-25:
I am interested to work on porting the tutorials during WMD. If anyone else wants to team up on this, I wouldn't mind having a coding buddy.

felixvd, 2021-03-01:
I can't get over the fact that we don't have [multiple planner support](https://github.com/ros-planning/moveit/issues/1257). If @henningkayser can make time, I'd like to discuss what to do about it. Maybe we could have a call the day before WMD, since most of my workday is during the EU night? I'll see about spending WMD on this, in that case.

bostoncleek, 2021-03-10:
Here is the [WIP PR #337](https://github.com/ros-planning/moveit2/pull/377) for the cartesian path planning track. I look forward to others joining me!

tylerjw, 2021-03-26:
closing as we did this

closed 2021-03-26 by tylerjw