moveit/moveit2 #365: Fake joint driver package missing

kekeblom, 2021-02-10:
### Description

I was running into some issues when setting up moveit2 for a custom simulation environment and was seeing some issues. As a first step, I wanted to validate my setup by running the `run_moveit_cpp` example.

However, the fake controller package doesn't seem to be available for moveit2. Where can I get that package? If I could install that package and test the example, I would be able to further narrow down my original issue.

### Your environment
* ROS Distro: foxy
* OS Version: Ubuntu 20.04
* Source or Binary build? Source
* If source, which branch? `master`, `bbe859ee6c2bd35c9ba01f666ff9e9a8a4915e1f`

### Expected behaviour

Launch the moveit example within RViz.

### Actual behaviour

Crashes when launching. See output below.

### Backtrace or Console output

```
[INFO] [launch]: All log files can be found below /home/user/.ros/log/2021-02-10-17-17-56-331691-newton-124541                                                                                                                                 
[INFO] [launch]: Default logging verbosity is set to INFO                                                                                                                                                                                     
Task exception was never retrieved                                                                                                                                                                                                            
future: <Task finished name='Task-2' coro=<LaunchService._process_one_event() done, defined at /opt/ros/foxy/lib/python3.8/site-packages/launch/launch_service.py:271> exception=PackageNotFoundError("package 'fake_joint_driver' not found
Traceback (most recent call last):                                                                                                                                                                                                            
  File "/opt/ros/foxy/lib/python3.8/site-packages/ament_index_python/packages.py", line 49, in get_package_prefix                                                                                                                             
    content, package_prefix = get_resource('packages', package_name)                                                                                                                                                                          
  File "/opt/ros/foxy/lib/python3.8/site-packages/ament_index_python/resources.py", line 48, in get_resource                                                                                                                                  
    raise LookupError(                                                                                                                                                                                                                        
LookupError: Could not find the resource 'fake_joint_driver' of type 'packages'                                                                                                                                                               
                                                                                                                                                                                                                                              
During handling of the above exception, another exception occurred:                                                                                                                                                                           
                                                                                                                                                                                                                                              
Traceback (most recent call last):                                                                                                                                                                                                            
  File "/opt/ros/foxy/lib/python3.8/site-packages/launch/launch_service.py", line 273, in _process_one_event                                                                                                                                  
    await self.__process_event(next_event)                                                                                                                                                                                                    
  File "/opt/ros/foxy/lib/python3.8/site-packages/launch/launch_service.py", line 293, in __process_event                                                                                                                                     
    visit_all_entities_and_collect_futures(entity, self.__context))                                                                                                                                                                           
  File "/opt/ros/foxy/lib/python3.8/site-packages/launch/utilities/visit_all_entities_and_collect_futures_impl.py", line 45, in visit_all_entities_and_collect_futures                                                                        
    futures_to_return += visit_all_entities_and_collect_futures(sub_entity, context)                                                                                                                                                          
  File "/opt/ros/foxy/lib/python3.8/site-packages/launch/utilities/visit_all_entities_and_collect_futures_impl.py", line 45, in visit_all_entities_and_collect_futures                                                                        
    futures_to_return += visit_all_entities_and_collect_futures(sub_entity, context)                                                                                                                                                          
  File "/opt/ros/foxy/lib/python3.8/site-packages/launch/utilities/visit_all_entities_and_collect_futures_impl.py", line 45, in visit_all_entities_and_collect_futures                                                                        
    futures_to_return += visit_all_entities_and_collect_futures(sub_entity, context)                                                                                                                                                          
  [Previous line repeated 1 more time]                                                                                                                                                                                                        
  File "/opt/ros/foxy/lib/python3.8/site-packages/launch/utilities/visit_all_entities_and_collect_futures_impl.py", line 38, in visit_all_entities_and_collect_futures                                                                        
    sub_entities = entity.visit(context)                                                                                                                                                                                                      
  File "/opt/ros/foxy/lib/python3.8/site-packages/launch/action.py", line 108, in visit                                                                                                                                                       
    return self.execute(context)                                                                                                                                                                                                              
  File "/opt/ros/foxy/lib/python3.8/site-packages/launch_ros/actions/node.py", line 426, in execute                                                                                                                                           
    ret = super().execute(context)                                                                                                                                                                                                            
  File "/opt/ros/foxy/lib/python3.8/site-packages/launch/actions/execute_process.py", line 775, in execute                                                                                                                                    
    self.__expand_substitutions(context)
[... cut ...]
[ERROR] [run_moveit_cpp-4]: process has died [pid 124549, exit code -15, cmd '/home/user/Hack/ws/install/run_moveit_cpp/lib/run_moveit_cpp/run_moveit_cpp --ros-args -r __node:=run_moveit_cpp --params-file /home/user/Hack/ws/install/run_moveit_cpp/share/run_moveit_cpp/config/moveit_cpp.yaml --params-file /tmp/launch_params_i095kdmq --params-file /tmp/launch_params_fple0eus --params-file /tmp/launch_params_8m5hw3yk --params-file /tmp/launch_params_0un0ydro --params-file /tmp/launch_params_0c4gwkmq'].
```

kekeblom, 2021-02-10:
Realized, I missed the vcs step of the source build instructions. Reporting back once I have those dependencies installed.

kekeblom, 2021-02-10:
Now that I installed the missing packages, it works, but the robot is initialized in and invalid state. It seems the `fake_joint_driver` is not respecting the `start_position` parameter. Anyone else seing this?

Here is a picture of the state:
[image]

JafarAbdi, 2021-04-15:
Closing since we no longer depend on the fake joint driver

closed 2021-04-15 by JafarAbdi