moveit/moveit2 #366: cannot build moveit2 and moveit_servo

tianshiz, 2021-02-11:
### Description

Overview of your issue here.

### Your environment
* ROS Distro: Noetic and Foxy
* OS Version: 20.04
* Source or Binary build? Binary
* If binary, which release version? 

### Steps to reproduce
Followed https://github.com/ros-planning/moveit2/blob/main/moveit_ros/moveit_servo/doc/running_the_demos.md

### Expected behaviour
All packages built

### Actual behaviour
Many pacakges fail, I was able to bypass some issues by installing some foxy message deps:
sudo apt install -y ros-foxy-object-recognition-msgs ros-foxy-control-msgs \
ros-foxy-octomap-msgs 

### Backtrace or Console output

latest error:
```
home/micra/ros2_ws/src/ros2_controllers/diff_drive_controller/src/diff_drive_controller.cpp:441:34:   required from here
/usr/include/c++/9/ext/new_allocator.h:145:20: error: no matching function for call to ‘realtime_tools::RealtimePublisher<tf2_msgs::msg::TFMessage_<std::allocator<void> > >::RealtimePublisher(std::shared_ptr<rclcpp_lifecycle::LifecyclePublisher<tf2_msgs::msg::TFMessage_<std::allocator<void> > > >&)’
  145 |  noexcept(noexcept(::new((void *)__p)
      |                    ^~~~~~~~~~~~~~~~~~
  146 |        _Up(std::forward<_Args>(__args)...)))
      |        ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~

```

tylerjw, 2021-02-15:
This seems unrelated.  Please provide more details.  realtime_tools or ros2_controllers are not part of the moveit repo.

tianshiz, 2021-02-16:
Hi, I was trying to following the instruction here:
https://moveit.ros.org/install-moveit2/source/
which required me to first build moveit_servo:
https://github.com/ros-planning/moveit2/blob/main/moveit_ros/moveit_servo/doc/running_the_demos.md

realtime_tools and ros2_controllers I suspect are the dependencies I had to get to build moveit:
`wget https://raw.githubusercontent.com/ros-planning/moveit2/main/moveit2.repos
vcs import < moveit2.repos
rosdep install -r --from-paths . --ignore-src --rosdistro foxy -y`

When I try to colcon build, I get all sort of issues, not from moveit src, but the deps that were sourced into the same workplace. Sorry is this the wrong place to ask about this problem?

`colcon build --event-handlers desktop_notification- status- --cmake-args -DCMAKE_BUILD_TYPE=Release
Starting >>> geometric_shapes
Starting >>> moveit_msgs
Starting >>> srdfdom
Starting >>> moveit_common
Starting >>> hardware_interface
Starting >>> warehouse_ros
Starting >>> moveit_resources_panda_description
Starting >>> moveit_resources_fanuc_description
Starting >>> moveit_resources_pr2_description
Finished <<< moveit_resources_fanuc_description [0.88s]
Starting >>> moveit_resources_fanuc_moveit_config
Finished <<< moveit_common [0.92s]
Finished <<< moveit_resources_panda_description [0.92s]
Starting >>> moveit_resources_panda_moveit_config
Finished <<< moveit_resources_pr2_description [0.94s]
Finished <<< moveit_resources_panda_moveit_config [0.57s]
Finished <<< moveit_resources_fanuc_moveit_config [0.66s]
Starting >>> moveit_resources
Finished <<< moveit_resources [1.11s]
Finished <<< srdfdom [11.7s]
Finished <<< moveit_msgs [13.0s]
Finished <<< hardware_interface [19.0s]
Starting >>> controller_interface
Starting >>> test_robot_hardware
Starting >>> transmission_interface
Finished <<< warehouse_ros [22.4s]
Starting >>> warehouse_ros_mongo
Finished <<< controller_interface [9.24s]
Finished <<< transmission_interface [16.9s]
Finished <<< warehouse_ros_mongo [14.4s]
Finished <<< test_robot_hardware [18.9s]
Starting >>> controller_manager
Finished <<< controller_manager [16.0s]
Starting >>> joint_state_controller
Starting >>> joint_trajectory_controller
Starting >>> diff_drive_controller
Starting >>> ros2_control
Finished <<< ros2_control [0.97s]
--- stderr: geometric_shapes
CMake Warning at CMakeLists.txt:54 (add_library):
  Cannot generate a safe runtime search path for target geometric_shapes
  because there is a cycle in the constraint graph:

    dir 0 is [/opt/ros/foxy/lib]
      dir 2 must precede it due to runtime library [librandom_numbers.so]
    dir 1 is [/opt/ros/foxy/lib/x86_64-linux-gnu]
    dir 2 is [/opt/ros/noetic/lib]
      dir 0 must precede it due to runtime library [libresource_retriever.so]

  Some of these libraries may not be found correctly.

CMake Warning at /opt/ros/foxy/share/ament_cmake_gtest/cmake/ament_add_gtest_executable.cmake:50 (add_executable):
  Cannot generate a safe runtime search path for target
  test_bounding_cylinder because there is a cycle in the constraint graph:

    dir 0 is [/home/micra/ros2_ws/build/geometric_shapes]
    dir 1 is [/opt/ros/foxy/lib]
[... 164 lines cut ...]
/usr/bin/ld: CMakeFiles/test_body_operations.dir/test_body_operations.cpp.o: in function `Bodies_ConstructMarkerFromBodyMesh_Test::TestBody()':
test_body_operations.cpp:(.text+0x3417): undefined reference to `boost::filesystem::path::operator/=(boost::filesystem::path const&)'
collect2: error: ld returned 1 exit status
make[2]: *** [test/CMakeFiles/test_body_operations.dir/build.make:158: test/test_body_operations] Error 1
make[1]: *** [CMakeFiles/Makefile2:480: test/CMakeFiles/test_body_operations.dir/all] Error 2
/usr/bin/ld: CMakeFiles/test_point_inclusion.dir/test_point_inclusion.cpp.o: in function `MeshPointContainment_Pr2Forearm_Test::TestBody()':
test_point_inclusion.cpp:(.text+0x7108): undefined reference to `boost::filesystem::path::operator/=(boost::filesystem::path const&)'
/usr/bin/ld: CMakeFiles/test_point_inclusion.dir/test_point_inclusion.cpp.o: in function `MeshPointContainment_Basic_Test::TestBody()':
test_point_inclusion.cpp:(.text+0x17213): undefined reference to `boost::filesystem::path::operator/=(boost::filesystem::path const&)'
collect2: error: ld returned 1 exit status
make[2]: *** [test/CMakeFiles/test_point_inclusion.dir/build.make:158: test/test_point_inclusion] Error 1
make[1]: *** [CMakeFiles/Makefile2:277: test/CMakeFiles/test_point_inclusion.dir/all] Error 2
make: *** [Makefile:141: all] Error 2
---
Failed   <<< geometric_shapes [1min 15s, exited with code 2]
Aborted  <<< diff_drive_controller [25.6s]
Aborted  <<< joint_state_controller [26.8s]
Aborted  <<< joint_trajectory_controller [29.3s]

Summary: 17 packages finished [1min 24s]
  1 package failed: geometric_shapes
  3 packages aborted: diff_drive_controller joint_state_controller joint_trajectory_controller
  1 package had stderr output: geometric_shapes
  25 packages not processed
`

tylerjw, 2021-02-16:
Looking through your output I'm confused how this is happening.  I just tried doing this and it is working locally for me.  Have you sourced the foxy install in opt first before running `colcon build`?

tianshiz, 2021-02-16:
sorry i was trying to reproduce my issue and forgot to source. ive updated again. Now it's missing geometric_shapes. If I install the prebuilt binary for geometric_shapes, it allows me to bypass that issue and then im hit with another package build issue. Just seems never ending.

Maybe something wrong with my ROS2 system? I simply installed the ros2 binary for foxy. I tried the caller/listener demo and it works in ROS2.

tylerjw, 2021-02-16:
Clean the workspace and try again.

```
# remove everything from the workspace you previously built moveit in
cd $COLCON_WS
rm -rf build install log src

# start new subshell and source foxy
bash
source /opt/ros/foxy/setup.bash

# install colcon mixin
colcon mixin add default https://raw.githubusercontent.com/colcon/colcon-mixin-repository/master/index.yaml
colcon mixin update default

# download the repos file and re-download the source dependencies
wget https://raw.githubusercontent.com/ros-planning/moveit2/main/moveit2.repos
mkdir src
vcs import src --input moveit2.repos

# install binary dependencies
rosdep update
rosdep install -r --from-paths src --ignore-src --rosdistro foxy -y

# build
colcon build --mixin release
```

tianshiz, 2021-02-17:
thanks that helped. I seem to be having some issues due to the ROS1 install on my machine.  I did not source noetic at all however:

> --- stderr: moveit_servo                                      
In file included from /opt/ros/noetic/include/octomap/octomap.h:36,
                 from /home/micra/ros2_ws/install/moveit_ros_occupancy_map_monitor/include/moveit/occupancy_map_monitor/occupancy_map.h:39,
                 from /home/micra/ros2_ws/install/moveit_ros_occupancy_map_monitor/include/moveit/occupancy_map_monitor/occupancy_map_monitor.h:47,
                 from /home/micra/ros2_ws/install/moveit_ros_planning/include/moveit/planning_scene_monitor/planning_scene_monitor.h:45,
                 from /home/micra/ros2_ws/src/moveit2/moveit_ros/moveit_servo/include/moveit_servo/servo_calcs.h:50,
                 from /home/micra/ros2_ws/src/moveit2/moveit_ros/moveit_servo/src/servo_calcs.cpp:45:
/opt/ros/noetic/include/octomap/ScanGraph.h: In constructor ‘octomap::ScanNode::ScanNode(octomap::Pointcloud*, octomap::pose6d, unsigned int)’:
/opt/ros/noetic/include/octomap/ScanGraph.h:57:41: warning: implicitly-declared ‘octomath::Pose6D::Pose6D(const octomath::Pose6D&)’ is deprecated [-Wdeprecated-copy]
   57 |       : scan(_scan), pose(_pose), id(_id) {}

is sourcing foxy not enough to isolate the build from ROS1?

tylerjw, 2021-02-17:
No, the foxy source does not remove anything from your environment.  So, if you have previously sourced a ros1 workspace that will also be in your environment.  I would make sure you are not sourcing noetic in your .bashrc or .zshrc.

tianshiz, 2021-02-18:
thanks for your help! it was hiding in ~/.profile for me

closed 2021-02-18 by tianshiz