moveit/moveit2 #367: segfault in moveit_servo test

tylerjw, 2021-02-15:
### Description

I'm not sure when this started but this is from a fresh build from the `main` (commit `20c7d3348`) branch on `foxy`.  The src directory of my workspace matches what is in the `moveit2.repos` file.  This is not the only failure in `moveit_servo` but I'm documenting it here as I just found it.

I built the repo using this command:
```
colcon build --mixin ccache rel-with-deb-info compile-commands --symlink-install --packages-up-to moveit_servo
```

```
$ colcon test --packages-select moveit_servo --ctest-args -R test_servo_integration.test --event-handlers console_direct+
Starting >>> moveit_servo
UpdateCTestConfiguration  from :/home/ubuntu/code/ws_moveit2/build/moveit_servo/CTestConfiguration.ini
Parse Config file:/home/ubuntu/code/ws_moveit2/build/moveit_servo/CTestConfiguration.ini
   Site: focal
   Build name: (empty)
 Add coverage exclude regular expressions.
SetCTestConfiguration:CMakeCommand:/usr/bin/cmake
Create new tag: 20210215-2109 - Experimental
UpdateCTestConfiguration  from :/home/ubuntu/code/ws_moveit2/build/moveit_servo/CTestConfiguration.ini
Parse Config file:/home/ubuntu/code/ws_moveit2/build/moveit_servo/CTestConfiguration.ini
Test project /home/ubuntu/code/ws_moveit2/build/moveit_servo
Constructing a list of tests
Done constructing a list of tests
Updating test list for fixtures
Added 0 tests to meet fixture requirements
Checking test dependency graph...
Checking test dependency graph end
test 4
    Start 4: test_launch_test_servo_integration.test.py

4: Test command: /usr/bin/python3 "-u" "/opt/ros/foxy/share/ament_cmake_test/cmake/run_test.py" "/home/ubuntu/code/ws_moveit2/build/moveit_servo/test_results/moveit_servo/test_launch_test_servo_integration.test.py.xunit.xml" "--package-name" "moveit_servo" "--output-file" "/home/ubuntu/code/ws_moveit2/build/moveit_servo/ros_test/CHANGEME.txt" "--command" "ros2" "test" "/home/ubuntu/code/ws_moveit2/src/moveit2/moveit_ros/moveit_servo/test/launch/test_servo_integration.test.py" "test_binary_dir:=/home/ubuntu/code/ws_moveit2/build/moveit_servo" "--junit-xml=/home/ubuntu/code/ws_moveit2/build/moveit_servo/test_results/moveit_servo/test_launch_test_servo_integration.test.py.xunit.xml" "--package-name=moveit_servo"
4: Test timeout computed to be: 60
4: -- run_test.py: invoking following command in '/home/ubuntu/code/ws_moveit2/build/moveit_servo':
4:  - ros2 test /home/ubuntu/code/ws_moveit2/src/moveit2/moveit_ros/moveit_servo/test/launch/test_servo_integration.test.py test_binary_dir:=/home/ubuntu/code/ws_moveit2/build/moveit_servo --junit-xml=/home/ubuntu/code/ws_moveit2/build/moveit_servo/test_results/moveit_servo/test_launch_test_servo_integration.test.py.xunit.xml --package-name=moveit_servo
4: Running with ROS_DOMAIN_ID 7            
4: ROS_DOMAIN_ID 7
4: [INFO] [launch]: All log files can be found below /home/ubuntu/.ros/log/2021-02-15-21-09-31-353361-focal-10611
4: [INFO] [launch]: Default logging verbosity is set to INFO
4: test_gtest_run_complete (moveit_servo.TestGTestProcessActive) ... [INFO] [fake_joint_driver_node-1]: process started with pid [10625]
4: [INFO] [component_container-2]: process started with pid [10627]
4: [INFO] [test_servo_integration-3]: process started with pid [10629]
4: [test_servo_integration-3] [==========] Running 3 tests from 1 test case.
4: [test_servo_integration-3] [----------] Global test environment set-up.
4: [test_servo_integration-3] [----------] 3 tests from ServoFixture
4: [test_servo_integration-3] [ RUN      ] ServoFixture.StartStopTest
4: [test_servo_integration-3] [INFO] [1613423371.535181335] [moveit_servo.servo_launch_test_common.hpp]: client_servo_start_ service not available, waiting again...
4: [fake_joint_driver_node-1] Parsing robot urdf xml string.
4: [fake_joint_driver_node-1] [INFO] [1613423371.540126111] [controller_manager]: Loading controller 'fake_joint_state_controller'
4: [fake_joint_driver_node-1] 
4: [fake_joint_driver_node-1] [INFO] [1613423371.544766798] [controller_manager]: Loading controller 'fake_joint_trajectory_controller'
4: [fake_joint_driver_node-1] 
4: [fake_joint_driver_node-1] [INFO] [1613423371.551486934] [fake_joint_trajectory_controller]: Controller state will be published at 50Hz.
4: [fake_joint_driver_node-1] [INFO] [1613423371.551817232] [fake_joint_trajectory_controller]: Action status changes will be monitored at 20Hz.
4: [fake_joint_driver_node-1] [INFO] [1613423371.553633053] [fake_joint_driver]: Successfully configured all controllers
4: [fake_joint_driver_node-1] [INFO] [1613423371.553672260] [fake_joint_driver]: Successfully activated all controllers
4: [component_container-2] [INFO] [1613423371.763682754] [test_servo_integration_container]: Load Library: /opt/ros/foxy/lib/librobot_state_publisher_node.so
4: [component_container-2] [INFO] [1613423371.766637596] [test_servo_integration_container]: Found class: rclcpp_components::NodeFactoryTemplate<robot_state_publisher::RobotStatePublisher>
4: [component_container-2] [INFO] [1613423371.766664745] [test_servo_integration_container]: Instantiate class: rclcpp_components::NodeFactoryTemplate<robot_state_publisher::RobotStatePublisher>
4: [component_container-2] Parsing robot urdf xml string.
4: [component_container-2] Link panda_link1 had 1 children
4: [component_container-2] Link panda_link2 had 1 children
4: [component_container-2] Link panda_link3 had 1 children
4: [component_container-2] Link panda_link4 had 1 children
4: [component_container-2] Link panda_link5 had 1 children
4: [component_container-2] Link panda_link6 had 1 children
4: [component_container-2] Link panda_link7 had 1 children
4: [component_container-2] Link panda_link8 had 1 children
4: [INFO] [launch_ros.actions.load_composable_nodes]: Loaded node '/robot_state_publisher' in container '/test_servo_integration_container'
4: [component_container-2] Link panda_hand had 2 children
[... 98 lines cut ...]
4: 
4: ----------------------------------------------------------------------
4: Ran 1 test in 0.000s
4: 
4: FAILED (failures=1)
4: -- run_test.py: return code 1             
4: -- run_test.py: verify result file '/home/ubuntu/code/ws_moveit2/build/moveit_servo/test_results/moveit_servo/test_launch_test_servo_integration.test.py.xunit.xml'
1/1 Test #4: test_launch_test_servo_integration.test.py ...***Failed   46.06 sec

0% tests passed, 1 tests failed out of 1

Total Test time (real) =  46.06 sec

The following tests FAILED:
          4 - test_launch_test_servo_integration.test.py (Failed)
Errors while running CTest
--- stderr: moveit_servo
Errors while running CTest
---
Finished <<< moveit_servo [46.2s]       [ with test failures ]

Summary: 1 package finished [46.5s]
  1 package had stderr output: moveit_servo
  1 package had test failures: moveit_servo
```

tylerjw, 2021-02-15:
You can see the same failure in my PR for refactoring the parameter loading in servo: https://travis-ci.com/github/ros-planning/moveit2/jobs/483290313#L1174

tylerjw, 2021-02-15:
Here is potentially more context (this came from running in valgrind):

```
4: [component_container-2] ==11739== Process terminating with default action of signal 11 (SIGSEGV)
4: [component_container-2] ==11739==  Access not within mapped region at address 0x18
4: [component_container-2] ==11739==    at 0x49FE049: rclcpp::TimerBase::cancel() (in /opt/ros/foxy/lib/librclcpp.so)
4: [component_container-2] ==11739==    by 0xC188E04: ~CollisionCheck (collision_check.h:74)
4: [component_container-2] ==11739==    by 0xC188E04: operator() (unique_ptr.h:81)
4: [component_container-2] ==11739==    by 0xC188E04: operator() (unique_ptr.h:75)
4: [component_container-2] ==11739==    by 0xC188E04: ~unique_ptr (unique_ptr.h:292)
4: [component_container-2] ==11739==    by 0xC188E04: moveit_servo::Servo::~Servo() (servo.cpp:85)
4: [component_container-2] ==11739==    by 0xBF6C2F9: operator() (unique_ptr.h:81)
4: [component_container-2] ==11739==    by 0xBF6C2F9: operator() (unique_ptr.h:75)
4: [component_container-2] ==11739==    by 0xBF6C2F9: reset (unique_ptr.h:402)
4: [component_container-2] ==11739==    by 0xBF6C2F9: moveit_servo::ServoServer::reset() (servo_server.cpp:121)
4: [component_container-2] ==11739==    by 0xBF6C44C: moveit_servo::ServoServer::stopCB(std::shared_ptr<std_srvs::srv::Trigger_Request_<std::allocator<void> > >, std::shared_ptr<std_srvs::srv::Trigger_Response_<std::allocator<void> > >) (servo_server.cpp:140)
4: [component_container-2] ==11739==    by 0xBF7244C: __invoke_impl<void, void (moveit_servo::ServoServer::*&)(std::shared_ptr<std_srvs::srv::Trigger_Request_<std::allocator<void> > >, std::shared_ptr<std_srvs::srv::Trigger_Response_<std::allocator<void> > >), moveit_servo::ServoServer*&, std::shared_ptr<std_srvs::srv::Trigger_Request_<std::allocator<void> > >, std::shared_ptr<std_srvs::srv::Trigger_Response_<std::allocator<void> > > > (invoke.h:73)
4: [component_container-2] ==11739==    by 0xBF7244C: __invoke<void (moveit_servo::ServoServer::*&)(std::shared_ptr<std_srvs::srv::Trigger_Request_<std::allocator<void> > >, std::shared_ptr<std_srvs::srv::Trigger_Response_<std::allocator<void> > >), moveit_servo::ServoServer*&, std::shared_ptr<std_srvs::srv::Trigger_Request_<std::allocator<void> > >, std::shared_ptr<std_srvs::srv::Trigger_Response_<std::allocator<void> > > > (invoke.h:95)
4: [component_container-2] ==11739==    by 0xBF7244C: __call<void, std::shared_ptr<std_srvs::srv::Trigger_Request_<std::allocator<void> > >&&, std::shared_ptr<std_srvs::srv::Trigger_Response_<std::allocator<void> > >&&, 0, 1, 2> (functional:400)
4: [component_container-2] ==11739==    by 0xBF7244C: operator()<std::shared_ptr<std_srvs::srv::Trigger_Request_<std::allocator<void> > >, std::shared_ptr<std_srvs::srv::Trigger_Response_<std::allocator<void> > > > (functional:484)
4: [component_container-2] ==11739==    by 0xBF7244C: std::_Function_handler<void (std::shared_ptr<std_srvs::srv::Trigger_Request_<std::allocator<void> > >, std::shared_ptr<std_srvs::srv::Trigger_Response_<std::allocator<void> > >), std::_Bind<void (moveit_servo::ServoServer::*(moveit_servo::ServoServer*, std::_Placeholder<1>, std::_Placeholder<2>))(std::shared_ptr<std_srvs::srv::Trigger_Request_<std::allocator<void> > >, std::shared_ptr<std_srvs::srv::Trigger_Response_<std::allocator<void> > >)> >::_M_invoke(std::_Any_data const&, std::shared_ptr<std_srvs::srv::Trigger_Request_<std::allocator<void> > >&&, std::shared_ptr<std_srvs::srv::Trigger_Response_<std::allocator<void> > >&&) (std_function.h:300)
```

Looking at the destructor I believe there is a bug with how we are dealing with the timer_ unique_ptr (it might not exist yet and the destructor always calls `cancel` on it).  However, fixing that does not fix this... The memory error in this one is something inside that cancel call.

tylerjw, 2021-03-09:
I marked this as a `good first issue` as it seems that it is fairly isolated and reproducible.  To fix this someone will need to dig into how we are using these ros2 features to find the bug.  I'll be happy to help anyone that wants to work on this get started.

Raghav-04, 2021-03-17:
Hi sir, I want to work on this issue please this to me.

tylerjw, 2021-03-17:
@Raghav-04 I tried recently and wasn't able to reproduce it.  If you are able to reproduce it just post here.

tylerjw, 2021-03-18:
I'm closing this because I can't reproduce it, feel free to re-open if it shows back up.

closed 2021-03-18 by tylerjw

Raghav-04, 2021-03-18:
I also tried but unable to locate the package.

Raghav-04, 2021-03-18:
using moveit repo 2 in ubuntu 20.04. Is the same conf. u have?

yangtfu, 2021-03-23:
> I'm closing this because I can't reproduce it, feel free to re-open if it shows back up.

call the service /servo_server/start_servo twice will raise a error which i think is related.