moveit/moveit2 #370: Use of motion_planning_rviz_plugin crashes rviz2

livanov93, 2021-02-22:
### Description

Overview of your issue here.

### Your environment
* ROS Distro: Foxy
* OS Version: e.g. Ubuntu 20.04
* Source build (main branch)

### Steps to reproduce
Test was on real hardware. To reproduce, user should build [Universal_Robots_ROS2_Driver](https://github.com/PickNikRobotics/Universal_Robots_ROS2_Driver/tree/use_moveit) on use_moveit branch and use instructions to start the robot control.

### Expected behaviour
Should be able to plan and execute unlimited times.

### Actual behaviour
While using Plan and Execute button in motion_planning_rviz_plugin  interactive marker pose on the end effector suddenly refuses to move (it is blocked) and on the next click the whole rviz2 crashes.

### Console output
[output](https://gist.github.com/livanov93/9562ec1ce9335ed2bfe60c42eab306eb)

JafarAbdi, 2021-02-24:
Hi  @livanov93,

Can you share the stack-trace using GDB .?

jrgnicho, 2021-04-28:
I'm trying to build the UR driver repo as described in the link (use_moveit branch) and get build errors, most of them look as follows:
```
src/Universal_Robots_ROS2_Driver/ur_controllers/src/force_torque_sensor_controller.cpp:103:36: error: ‘get_node’ was not declared in this scope
  103 |   wrench_state_msg_.header.stamp = get_node()->get_clock()->now();
```

Besides that, Is there any way to reproduce the error without an actual robot?  that'll make things easier

livanov93, 2021-05-17:
> I'm trying to build the UR driver repo as described in the link (use_moveit branch) and get build errors, most of them look as follows:
>
> ``` [...]

It is not possible to reporoduce it anymore, especially with above mentioned instructions. UR ros2 driver has been actively developed so a lot of changes have happened. Closing the issue for now until the problem is reproduced on the updated stack.

closed 2021-05-17 by livanov93